dtcooper/raspotify #282: Raspotify stuck skipping tracks

crchiriboga, 2020-07-13:
When connecting to Raspotify, Spotify App on android keeps skipping tracks continuously.

DanielDocekalFHTW, 2020-07-13:
Same, it played one song, then it just skips in a loop without ever playing anything. As soon as I tell spotify to play it on my PC/Phone it stops skipping and music is playing from said devices

Tijs-B, 2020-07-13:
Same here, on Spotify for Mac and the Spotify Android app.

mix3d, 2020-07-13:
How dare spotify change something on their API.... lol.

Nick-Irza, 2020-07-13:
Also having this issue... Hopefully it's a Spotify and not a raspotify problem!

ManuelSchneid3r, 2020-07-13:
Any workarounds?

migazo, 2020-07-13:
I have same problem, from the CLI, I get the following errors:
... WARN  librespot_core::audio_key] error audio key 0 2
... ERROR librespot_core::channel] channel error: 2 1
... ERROR librespot_playback::player] Unable to load encrypted file.

This then leads the raspsotify to load a new song from the queue, which fails similarly again.

peterdk, 2020-07-13:
Ah, same issue. Thought it was my compiled version of librespot. I checked with latest dev version of librespot, same issue.

RobinDevogele, 2020-07-13:
Same issue here. Skipping all song till going back to one single "working" song .

mpdehnel, 2020-07-13:
Yup, same thing for me! Really annoying. Can't even get it to play a single 'working' song :-(

CryptoPenguin15, 2020-07-13:
Ditto.

0v3rCl0kEd, 2020-07-13:
Same problem for me, started about 2 hours ago.

mix3d, 2020-07-13:
Also working again for me. How fun! Guess we can mark this resolved.

crchiriboga, 2020-07-13:
Same here... Suddenly started working.

mpdehnel, 2020-07-13:
Any idea what caused this?

FistonBozzo, 2020-07-13:
Same problem. Loop.

manu9n, 2020-07-13:
Now it works great👍
Thanks for this great project!

RobinDevogele, 2020-07-16:
Happening again :(

kirenida, 2020-07-23:
I think this may be solved in librespot issue [497](https://github.com/librespot-org/librespot/issues/497), which also apparently solves the problem of not being able to play tracks from the latest Android Spotify app.

I have experienced both issues - the track skipping and inability to play songs from the app.

dtcooper, 2020-07-26:
Should be fixed with the upstream version of librespot now available. Please update and see if it works, open a new issue or comment here for me to re-open if it's not fixed.

closed 2020-07-26 by dtcooper

JasonLG1979, 2020-08-17:
Nope. It's still a problem.

```
Aug 17 17:32:49 raspberrypi librespot[1156]: [2020-08-17T16:32:49Z INFO  librespot_playback::player] Loading <Panama - 2015 Remaster> with Spotify URI <spotify:track:05RgAMGypEvqhNs5hPCbMS>
Aug 17 17:32:49 raspberrypi librespot[1156]: [2020-08-17T16:32:49Z ERROR librespot_core::channel] channel error: 2 0
Aug 17 17:32:49 raspberrypi librespot[1156]: [2020-08-17T16:32:49Z ERROR librespot_playback::player] Unable to load encrypted file.
```

Just about every other song in the Spotify '80s Hard Rock playlist errors out.

JasonLG1979, 2020-08-17:
I managed to get librespot-java running on my Pi Zero after installing a 3rd party version of Java 11 (Java 11 in the repos does not support armv6) and it doesn't seem to have this issue. But given the jvm overhead it has other issues on the Zero. It's a resource hog and slow, to the point that it has trouble keeping the decoder feed. There is mention here that this and similar issues could easily fixed in librespot.

https://github.com/librespot-org/librespot/issues/519#issuecomment-674564875

JasonLG1979, 2022-08-11:
> 2 years later, I seem to be having the same problem described here...

NecroBumping a closed issue with a "me too" isn't helpful. Do you have any the ng useful like logs?

TobiasGrothmann, 2022-08-11:
@JasonLG1979  I'm happy to open a new issue, if you would like me to.

I have two raspberry pis, both running raspotify. I haven't changed anything (not even connected) on either of them for a while.
Both started to:
* skip every song after a maximum of 6 seconds
* as far as I can hear they do not output any audio
* I cannot stop/pause the playback
* "playing" starts as soon as I select them as the output device
The only feedback I currently have is the spotify app just skipping from song to song.

The system log looks like this:
```
Aug 11 17:34:05 esszimmer systemd[1]: Starting Raspotify...
Aug 11 17:34:05 esszimmer systemd[1]: Started Raspotify.
Aug 11 17:34:06 esszimmer librespot[462]: [2022-08-11T16:34:06Z INFO  librespot] librespot (raspotify v0.16.0) 0adb851 (2020-12-15). Built on 2020-12-30. Build ID: KLdkAvgt
Aug 11 17:34:06 esszimmer librespot[462]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 19, kind: Other, message: "No such device" }', src/main.rs:440:81
Aug 11 17:34:06 esszimmer librespot[462]: stack backtrace:
Aug 11 17:34:06 esszimmer librespot[462]:    0:   0x8f2278 - std::backtrace_rs::backtrace::libunwind::trace::h9c6af82ecb947420
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/../../backtrace/src/backtrace/libunwind.rs:96
Aug 11 17:34:06 esszimmer librespot[462]:    1:   0x8f2278 - std::backtrace_rs::backtrace::trace_unsynchronized::h80561bf66e637b12
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/../../backtrace/src/backtrace/mod.rs:66
Aug 11 17:34:06 esszimmer librespot[462]:    2:   0x8f2278 - std::sys_common::backtrace::_print_fmt::h8904fddc87bf6b5e
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/sys_common/backtrace.rs:79
Aug 11 17:34:06 esszimmer librespot[462]:    3:   0x8f2278 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hc883ceb7a13219f2
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/sys_common/backtrace.rs:58
Aug 11 17:34:06 esszimmer librespot[462]:    4:   0x91a290 - core::fmt::write::ha75209c8e86bef45
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/core/src/fmt/mod.rs:1080
Aug 11 17:34:06 esszimmer librespot[462]:    5:   0x8eae18 - std::io::Write::write_fmt::h4cce8be81a41c537
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/io/mod.rs:1516
Aug 11 17:34:06 esszimmer librespot[462]:    6:   0x8f51c0 - std::sys_common::backtrace::_print::hdc9e28891c09ab0b
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/sys_common/backtrace.rs:61
Aug 11 17:34:06 esszimmer librespot[462]:    7:   0x8f51c0 - std::sys_common::backtrace::print::h85895381379b4dc1
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/sys_common/backtrace.rs:48
Aug 11 17:34:06 esszimmer librespot[462]:    8:   0x8f51c0 - std::panicking::default_hook::{{closure}}::h6b4de484f2428118
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/panicking.rs:208
Aug 11 17:34:06 esszimmer librespot[462]:    9:   0x8f4dbc - std::panicking::default_hook::h3966366a200b03f1
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/panicking.rs:227
Aug 11 17:34:06 esszimmer librespot[462]:   10:   0x8f5910 - std::panicking::rust_panic_with_hook::h068b5368b0949ced
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/panicking.rs:577
Aug 11 17:34:06 esszimmer librespot[462]:   11:   0x8f54c4 - std::panicking::begin_panic_handler::{{closure}}::hde463ebfc6f96538
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/panicking.rs:484
Aug 11 17:34:06 esszimmer librespot[462]:   12:   0x8f27f8 - std::sys_common::backtrace::__rust_end_short_backtrace::hb945924c40726c97
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/sys_common/backtrace.rs:153
Aug 11 17:34:06 esszimmer librespot[462]:   13:   0x8f547c - rust_begin_unwind
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/panicking.rs:483
Aug 11 17:34:06 esszimmer librespot[462]:   14:   0x917960 - core::panicking::panic_fmt::h27b19d6e6dedd33e
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/core/src/panicking.rs:85
Aug 11 17:34:06 esszimmer librespot[462]:   15:   0x9177b8 - core::option::expect_none_failed::h0ec77a052e666b69
Aug 11 17:34:06 esszimmer librespot[462]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/core/src/option.rs:1234
[... cut ...]
Aug 11 17:43:57 esszimmer librespot[1253]: [2022-08-11T16:43:57Z ERROR librespot_core::channel] channel error: 2 0
Aug 11 17:43:57 esszimmer librespot[1253]: [2022-08-11T16:43:57Z ERROR librespot_playback::player] Unable to load encrypted file.
Aug 11 17:43:57 esszimmer librespot[1253]: [2022-08-11T16:43:57Z WARN  librespot_playback::player] Unable to load <SpotifyId { id: 328231358127420963662014606153691530037, audio_type: Track }>
...
```

I did a little `sudo apt update && sudo apt upgrade` + `reboot` which didn't change anything.

JasonLG1979, 2022-08-11:
Librespot 0.16 is WAY out of date. You would 1st need to update to the current Raspotify version before you can start a new issue. Keep in mind that Raspotify is on compatible with Debian Stable (11, bullseye) based distros and no longer supports Pi v1's or Pi Zero v1's.

JasonLG1979, 2022-08-11:
What stands out the most in your logs is the AP. That AP has been blacklisted in the latest librespot release because it is known to cause the very issue you're having.