dtcooper/raspotify #285: Output to icecast

djbarnat, 2020-07-25:
Hi can anyone suggest how one might get the output of raspotify to be directed to icecast?

artmg, 2020-07-25:
I'm not sure about sending to _icecast_, but I use **snapcast** to send my audio to multiple devices around the house. You can find more about the project at https://github.com/badaix/snapcast and see how to use librespot (as installed by **raspotify**) here https://github.com/badaix/snapcast/blob/master/doc/player_setup.md#spotify.

djbarnat, 2020-07-26:
Hi and thank you for you reply it is appreciated. I am familiar with snapcast and use it to good effect on all my main stereos in the house to sync audio. But it does require use of the snapcast client which my internet radio devices do not support. They do support a custom http stream though hence the hope of using raspotify with icecast. I have been playing with alsa loopback yesterday to try and do this but librespot seems to have an issue with setting the sample rate on intialisation of the alsa loopback device. I think the simplest solution may be to connect physical loop back cable from the output of my soundcard and then back into the input in conjunction with an icecast source like darkice or ices2 but I was hoping for a purely software  solution.

djbarnat, 2020-07-26:
Ok so oddly about 10 mins after the above post i have managed to get this working with alsa loopback and for some reason the complaints from librespot about not being able to set samplerate have disaapeared

this is how i skinned the cat;

load the alsa loop back module
`sudo modprobe snd-aloop
`

Confirm loop back hw addresses
`aplay -l`

`**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: CODEC [USB AUDIO  CODEC], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Loopback [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 2: Loopback [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
`
in my case the input used is hw:2,0,0 

Launch librespot pointing it at hw:2,0,0

`/usr/bin/librespot --name raspotify  --backend alsa --device hw:2,0,0  --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --username <your username> --password <yourpassword> &

install ffmeg
`sudo apt install ffmpeg`

Using ffmpeg capture the loopback "mirror" stream on hw:2,1,0  - stream with ffmpeg to icecast

`ffmpeg -ac 2 -f alsa -i hw:2,1,0 -acodec libmp3lame -ac 1 -content_type audio/mpeg -f mp3 icecast://source:passwordforicecast@192.168.1.103:8000/spotcon
`

configure my radio with a custom radio station that points at http://192.168.1.103:8000/spotcon and i am getting spotify connect stream playing.

Might help someone attemtptong the same hopefully.

dtcooper, 2020-07-26:
Not going to implement since this would go against the "easy to use" goal of the project, but do feel free to follow the example by @djbarnat and let us know how it goes!

closed 2020-07-26 by dtcooper