dtcooper/raspotify #289: Librespot error: option 'device' given more than once

Sanderei, 2020-08-05:
I changed my audio device in raspotify, and it simply wont start again because it says that the option device is given more than once. The options line is the only uncommented line in my config file, and there's one device command in it.

crowjake, 2020-08-08:
I had the same issue and it'd be fixed when I pointed spotify to a different output and then came back to the pi.

JasonLG1979, 2020-10-20:
I'd have to see your ```/etc/asound.conf```, ```/etc/default/raspotify``` and raspotify logs (```journalctl -u raspotify.service -b```) but I'm guessing something's not configured correctly.

Sanderei, 2020-11-07:
It was fine, but now it is doing it again after a fresh install on DietPi.
`ctl.eq {
              type equal
              controls "/root/.alsaequal.bin"
}

pcm.plugequal {
               type equal
               slave.pcm "plughw:1,0"
               controls "/root/.alsaequal.bin"
}

pcm.!default {
               type plug
               slave.pcm plugequal
               slave.channels 2
}

ctl.!default {
               type hw
               card 1
}`
/etc/asound.conf ^

In the logs it looks like it can't access the audio system, and it's too much to write it all here.
I don't know to be honest how I can get these logs imported here.

JasonLG1979, 2020-11-08:
I have not used the `equal` plugin before so I can't tell you if your `asound.conf` is correct or not but if your sound card does not have hardware mixing capabilities (most consumer cards do not) Only one app can use your sound card at a time since you're going `plug > equal > hw` and not doing any software mixing. To enable multiple apps to use your sound card simultaneously you'll need to throw a `dmix` in front of the hardware. I do not know how well `equal` plays with `dmix`?  

Here is a cheat sheet to get most of the info you'd need to configure audio on a Debian based system (it should work for about any distro really for the most part).

`aplay -l` (lowercase L) gets you the names and indexes of your sound cards.
For the `card` field in the `hw` plugin you can use either the name or index.

So for example if this is part of my output:
```
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 [HifiBerry DAC HiFi pcm5102a-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

```
I can use either `0` of `sndrpihifiberry` for `card`. Names are better than indexes as indexes can change.

`aplay -Dhw:<card>,<device #> --dump-hw-params /path/to/some/wav` will get you the capabilities of you sound card.

So with the above example:

`aplay -Dhw:0,0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav`

Outputs:
```
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: [8000 192000]
...
```
That tells me that the above card is 2 channel and supports formats S16_LE, S24_LE, and S32_LE, at up to a sampling rate of 192000.

JasonLG1979, 2020-11-08:
Provided your above card index is correct and your card supports 16 bit 44.1k something like this might work:

```
pcm.plugdmixer {
    type plug
    slave.pcm {
        type dmix
        ipc_key_add_uid false
        ipc_perm 0666
        ipc_key 1024
        slave {
            pcm "hw:1,0"
            channels 2
            rate 44100
            format S16_LE
            period_size 0
            buffer_size 0
            buffer_time 0
            period_time 250000
            periods 4
        }
        bindings {
            0 0
            1 1
        }
    }
}

pcm.equal {
    type equal
    slave.pcm plugdmixer
    controls "/root/.alsaequal.bin"
}

pcm.!default {
    type plug
    slave.pcm equal
}

ctl.eq {
    type equal
    controls "/root/.alsaequal.bin"
}

ctl.!default {
    type hw
    card 1
}
```

closed 2021-11-24 by github-actions[bot]