dtcooper/raspotify #291: Cant force audio through the 3.5mm headphone jack

Shayner2005, 2020-08-16:
I've set raspi-config to force audio through the headphone jack. All other audio is working through the jack. I've tried adding the hdmi_ignore to config.txt as well as the other troubleshooting steps found in the issues.
Thanks

This is my raspotify conf:
 /etc/default/raspotify -- Arguments/configuration for librespot

# Device name on Spotify Connect
#DEVICE_NAME="raspotify"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
#BITRATE="320"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info. Make sure whatever arguments you specify
# aren't already covered by other variables in this file. (See the daemon's
# config at `/lib/systemd/system/raspotify.service` for more technical details.)
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#--device hw:0
#OPTIONS="--username <-------------> --password <-------->"

# Uncomment to use a cache for downloaded audio files. Cache is disabled by
# default. It's best to leave this as-is if you want to use it, since
# permissions are properly set on the directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

# By default, the volume normalization is enabled, add alternative volume
# arguments here if you'd like, but these should be fine.
#VOLUME_ARGS="--enable-volume-normalisation --linear-volume --initial-volume=100"

# Backend could be set to pipe here, but it's for very advanced use cases of
# librespot, so you shouldn't need to change this under normal circumstances.
#BACKEND_ARGS="--backend alsa"

romariolopezc, 2020-08-28:
What adapter are you using?

Shayner2005, 2020-08-28:
Headphones

michaelfeil, 2020-09-01:
Your request is very ill formatted. 

Questions you should answer: 
Which device are you using? (Raspberry Pi X)
Have you a monitor connected over HDMI? 
If yes, is Raspotify paying over HDMI?
If yes, the answer is that you need to change from HDMI to the Standard Audio jack. 

Changing output device:
Formerly: 
#` OPTIONS= --device hw:0,1 `
change to
` OPTIONS= `--device hw:1` `

the number depends on your Playback Hardware Device. In my Case HDMI is 0, Headphones is 1, hifiberryaudio is 2

closed 2020-09-01 by Shayner2005