dtcooper/raspotify #293: Some songs cant be played

BrazilianArmy, 2020-08-18:
Hello!

I'm using raspotify at my old cubieboard 2, its work well but some songs dont play there, anyone here know something about it? I've seacher but i might be using wrong words and I didnt find nothing, my english is basic as it is not my primary language.
log:

`Aug 18 05:12:06 cubieboard2 librespot[1921]: [2020-08-18T05:12:06Z INFO  librespot_playback::player] Loading <Running Outta Love> with Spotify URI <spotify:track:1aSTn7F9RUijD31mAzjlQx>
Aug 18 05:12:06 cubieboard2 librespot[1921]: [2020-08-18T05:12:06Z ERROR librespot_core::channel] channel error: 2 0
Aug 18 05:12:06 cubieboard2 librespot[1921]: [2020-08-18T05:12:06Z ERROR librespot_playback::player] Unable to load encrypted file.
Aug 18 05:12:06 cubieboard2 librespot[1921]: [2020-08-18T05:12:06Z WARN  librespot_connect::spirc] The player has stopped unexpectedly.`

[image]

Thanks and sorry for my english

mk-360, 2020-08-19:
I have the same problem, Spotify Play only some songs and it stops unexpectedly. I can do a sudo systemctl restart raspotify but the album or playlist starts from the start and it stops again, usually on the same song. I'm using the latest version of raspbian (or Raspberry OS) with the last version of Raspotify from the raspotify repository under a Raspberry Pi 3 model B.
Here is my log:
ago 18 19:47:54 rpi-mk librespot[582]: [2020-08-18T23:47:54Z WARN  librespot_audio::fetch] Error from channel for data receiver for range 5045536 (+16384).
ago 18 19:49:02 rpi-mk librespot[582]: [2020-08-18T23:49:02Z INFO  librespot_playback::player] Loading <Hazard Profile, Pt. 1 - Live at the Baked Potato> with Spotify URI <spotify:track:5rGM9EVIwg57HV2wL13ZnL>
ago 18 19:49:03 rpi-mk librespot[582]: [2020-08-18T23:49:03Z INFO  librespot_playback::player] <Hazard Profile, Pt. 1 - Live at the Baked Potato> (389306 ms) loaded
ago 18 19:55:32 rpi-mk librespot[582]: [2020-08-18T23:55:32Z INFO  librespot_playback::player] Loading <Kings and Queens - Live at the Baked Potato> with Spotify URI <spotify:track:4iNAHHWfRL1XVmesLu6Kqw>
ago 18 19:55:32 rpi-mk librespot[582]: [2020-08-18T23:55:32Z ERROR librespot_core::channel] channel error: 2 0
ago 18 19:55:32 rpi-mk librespot[582]: [2020-08-18T23:55:32Z ERROR librespot_playback::player] Unable to load encrypted file.
ago 18 19:56:02 rpi-mk librespot[582]: [2020-08-18T23:56:02Z INFO  librespot_playback::player] Loading <Kings and Queens - Live at the Baked Potato> with Spotify URI <spotify:track:4iNAHHWfRL1XVmesLu6Kqw>
ago 18 19:56:03 rpi-mk librespot[582]: [2020-08-18T23:56:03Z ERROR librespot_core::channel] channel error: 2 0
ago 18 19:56:03 rpi-mk librespot[582]: [2020-08-18T23:56:03Z ERROR librespot_playback::player] Unable to load encrypted file.
ago 18 19:56:03 rpi-mk librespot[582]: [2020-08-18T23:56:03Z WARN  librespot_connect::spirc] The player has stopped unexpectedly.

natn, 2020-08-19:
I have the same problem with certain tracks.  It will fail consistently on some tracks and log messages similar to this.  Raspberry Pi 4

`18.08.20 20:59:31 (-0500)  spotify  [2020-08-19T01:59:31Z INFO  librespot_playback::player] Loading <Gendèr> with Spotify URI <spotify:track:73J0XZfBhvGgCloBnkXVCP>
18.08.20 20:59:34 (-0500)  spotify  [2020-08-19T01:59:34Z ERROR librespot_core::channel] channel error: 2 0
18.08.20 20:59:34 (-0500)  spotify  [2020-08-19T01:59:34Z ERROR librespot_playback::player] Unable to load encrypted file.
18.08.20 20:59:34 (-0500)  spotify  [2020-08-19T01:59:34Z WARN  librespot_connect::spirc] The player has stopped unexpectedly.`

JRansomed, 2020-08-19:
I believe it has something to do with the librespot that is included with raspotify.  I went to that project's site and followed their install/compile directions, then symlinked it where raspotify installed it's build of it.  It's been over a day since I've seen the issue now.  Previously, I was seeing it every time I listened.

mk-360, 2020-08-19:
Hmmmmm, I don't know if I have enough experience to do it, but I'll try...

JRansomed, 2020-08-19:
@mk-360 what are you running this on?  I'm running on a raspberry pi.  If you are running on a raspberry pi, I can write what I did to install/compile it.  It might be a day or so before I get the chance.

mk-360, 2020-08-19:
@JRansomed That would be great, I'm using a headless raspberry pi 3b+ with raspbian with all updates.
Thanks!

JRansomed, 2020-08-19:
@mk-360 That matches my setup.  I'll write something up and hopefully hit all the points I found.  If something is missed, we can work through and improve the documentation.  It really wasn't all that bad.  Just takes the pi some time to compile.

BrazilianArmy, 2020-08-19:
@natn 

If I restart the raspotify it will work again too but other files become unavailable.

@JRansomed Hey, thanks for the tips, i'll try compile it, i'm not a expertise on linux but i hope i can make it work as well. If you have any tips wich or where get the files to compile it will help a lot. I know i need the source code but i'm not sure if it only this.

If help you too, I'm using Cubieboard 2 with armbian

JRansomed, 2020-08-20:
@BrazilianArmy I just put a fresh microsd card in my pi.  Going through the motions from scratch and documenting it.  I'll cover creating the symlinks.  The reason I'm using symlinks is to make it easy to change to the binary and config we are building.  Likewise, easy to change back to the one dtcooper created.  I imagine these steps should work for the cubieboard 2.

Waiting on the longest part, the software compilation now.  Should be able to finish my notes soon and organize them into something readable, then post them.  : )

JRansomed, 2020-08-20:
Ok, this is going to be a lot to post.  It's not as bad as it looks.  : )

-------------------------------------------------------------------
Run these commands as your normal, logged in user.  Use sudo when a privileged
command is needed.

Install some prerequisites:
`sudo apt-get install build-essential libasound2-dev`

I needed a newer version of rust than my pi installed from apt. Per the
instructions at https://www.rust-lang.org/tools/install:
`curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh`

When prompted, proceed with the default by selecting `1`

When the install finishes, we need to update our paths for our currently logged in
session:
`source ~/.cargo/env`

Compile and install librespot (This step will take quite a while, 41 minutes for me)
`cargo install librespot`

If it completes successfully, you should see something similar to:
`    Finished release [optimized] target(s) in 41m 46s`
`  Installing /home/pi/.cargo/bin/librespot`
`   Installed package librespot v0.1.3 (executable librespot)`

If we ask the OS where librespot is installed, we should see two installs:
`whereis librespot`

Example output:
`librespot: /usr/bin/librespot /home/pi/.cargo/bin/librespot`

The new copy is in your home directory.  Before we make changes, lets shutdown
Raspotify:
`sudo systemctl stop raspotify`

Move old librespot and symlink to the new one.  I'm getting /usr/bin from my
whereis output.  My install had dtcooper's install in /usr/bin:
`cd /usr/bin/`
`sudo mv librespot librespot.old`
`sudo ln -sf ~/.cargo/bin/librespot /usr/bin/librespot`

The new librespot has changed some of it's option flags.  We will need to edit
our /etc/default/raspotify.  But first we'll copy and symlink.
`cd /etc/default`
`sudo cp raspotify raspotify.old`
`sudo mv raspotify raspotify.new`

Edit raspotify.new with your preferred editor.  You will need to sudo to edit
this file.  Raspotify by default uses --linear-volume.  This option changed in
the new version of librespot.  Uncomment the following line by removing the #
at the start of the line:
`#VOLUME_ARGS="--enable-volume-normalisation --linear-volume --initial-volume=100"`

Edit the line so that --linear-volume is now --mixer-linear-volume:
`VOLUME_ARGS="--enable-volume-normalisation --mixer-linear-volume --initial-volume=100"`

The default backend is incorrect with the new version.  At the bottom of the
file should be the following line.  Remove the # to uncomment it:
`#BACKEND_ARGS="--backend alsa"`

Make it look like the following (alsa no longer worked for me, at librespot's
github site, https://github.com/librespot-org/librespot, they indicate that
rodio is now the default.  This worked for my install):
`BACKEND_ARGS="--backend rodio"`

Finally symlink to our new config:
`sudo ln -sf raspotify.new raspotify`

Now start raspotify and test it.
`sudo systemctl start raspotify`

To check the logs:
`journalctl -u raspotify`

---------------------------------------------
To change everything back to the dtcooper's config and binary:
`sudo systemctl stop raspotify`
`sudo ln -sf /usr/bin/librespot.old /usr/bin/librespot`
`sudo ln -sf /etc/default/raspotify.old /etc/default/raspotify`
`sudo systemctl start raspotify`

mk-360, 2020-08-20:
@JRansomed Excellent! I've been playing a playlist for a couple of hours so far without any problem.
In the first attempt the instructions didn't work, cause I depend on ALSA for other things (a hifiberry like sound card, mopidy) but surfing a bit through the librespot documentation I found a solution. If someone else needs ALSA can use the install command described in the instructions in this manner:
cargo install librespot --no-default-features --features "alsa-backend"
All the rest worked perfectly :)
Thanks for your help!

BrazilianArmy, 2020-08-20:
@JRansomed Hey thanks you!

When i get home i'll reinstall my system and try it, yesterday i was trying to compile it by myself but got stuck at some erros, hope today with you explanation i can pass it.

Thanks!

JRansomed, 2020-08-20:
@mk-360 Awesome!  I'll edit the instructions to show the alsa-backend option.  Thanks for sharing!
@BrazilianArmy Hopefully these instructions will get you going.  If not, post back here.  Maybe we can figure it out.

I'm still hopeful that Raspotify's repo will get updated with the new version of librespot.  We should see it when it does.  It should overwrite the symlink.  ; )  Happy to have helped!

BrazilianArmy, 2020-08-23:
Hi @JRansomed after a while trying to figure out others issue while compiling i finally made it work, everything is edited but when i do the command: 
journalctl -u raspotify
i see a error there, i'm pretty sure its is easy to find out for those who already had a good experience with linux, but im stuck there since yesterday trying to figure out what 'im doing wrong.
By now, seems something about permission but i dont find a way by my self to correct this, could you help me?

Aug 23 13:26:59 cubieboard2 systemd[1290]: raspotify.service: Failed to execute command: Permission denied
Aug 23 13:26:59 cubieboard2 systemd[1290]: raspotify.service: Failed at step EXEC spawning /usr/bin/librespot: Permission denied

 raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2020-08-23 13:47:27 -03; 3s ago
  Process: 1807 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS (code=exited, status=203/EXEC)
  Process: 1806 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1805 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1807 (code=exited, status=203/EXEC)

Thanks!

Callenberg, 2020-09-06:
Hello @BrazilianArmy ,
I had the same annoying error (203/EXEC). 

First I tried from the CLI with my set-up:
`/usr/bin/librespot --name Player --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --initial-volume=5 --mixer-linear-volume --onevent /player/spotifyevent.sh`

That worked! The things that might stick out here is that I want a low start-up volume and I catch the start and stop in a script. Besides that I think it is a rather normal option set-up.

**Quick workaround**
The problem is as you suggest a permission problem, it can be found in the Raspotify systemd service definition, i.e. in the file /lib/systemd/system/raspotify.service. I just changed the user from raspotify to root in order to get any permission hinders out of the way.
`User=root`

Restarted the Raspotify service by:
` systemctl daemon-reload`
` systemctl restart raspotify.service`

Now librespot works with the Raspotify wrap. 

However, it is generally believed that you should not run programs as root. I understand the reasoning for that... well, in my case, I own and control the computer in question, it only streams Spotify, it is connected to a private network,... That said, yes there has to be a better solution than editing the raspotify.service file and using root.

Finally I want to thank @JRansomed and @mk-360  for the instructions above. It was really helpful. So far it has felt a little bit overwhelming to compile librespot myself, but now I am more confident about that. I actually think that I in the future will skip Raspotify and just compile from the source.

closed 2021-11-24 by github-actions[bot]