dtcooper/raspotify #296: Error: "couldn't parse packet from V4"

floodwayprintco, 2020-09-08:
Hi there,

I am receiving an error a LOT and not sure how to diagnose. I think this is causing my Raspotify player to lag a bit. 

I have Adguard running but it doesn't seem to be blocking Spotify at all.

Here is the error code, it's spamming like crazy.

`[2020-09-08T15:28:21Z WARN  libmdns::fsm] couldn't parse packet from V4(172.30.32.1:5353): type 47 is invalid`

Any help would be appreciated! I searched a bit but couldn't find anything.

JasonLG1979, 2020-10-19:
librespot has [mDNS](https://en.wikipedia.org/wiki/Multicast_DNS) (libmdns) built into it. mDNS uses port 5353. It looks like it may be a coexistence problem between librespot and Adguard. avahi-daemon has a config option that allows it to coexist on the same machine as other mDNS resolvers (```disallow-other-stacks=no```) I have no idea if Adguard has something similar?

Otherwise you have 2 possible options:

1. Since you say Spotify is unaffected you can change the log level to error [like this PR does](https://github.com/dtcooper/raspotify/pull/232). So it won't log it since it's a ```WARN``` level log message.

2. You can use your username and password to login with librespot and turn off mDNS with the ```--disable-discovery``` option.

floodwayprintco, 2020-10-20:
Thanks a ton for looking into this. I actually was having a ton of issues
with lag.

Since I couldn't figure this out, I moved Raspotify and Forked-Daapd onto a
dedicated Raspberry Pi and everything works normally. Wishing I could run
it all off the HomeAssistant Pi but it was relatively simple to
troubleshoot it by moving it to dedicated hardware.

JasonLG1979, 2020-10-20:
I took a look at [Raspotify and Forked-Daapd]( https://github.com/Ulrar/hassio-addons). It appears to be raspotify in name only, it's a completely different project even going so far as to patch librespot. If you're having issues with that add-on then here is not the place to file bug reports because any solutions presented here mat not be even applicable.

floodwayprintco, 2020-10-20:
Well thank you for clearing that up. The name is definitely confusing! But
your insight is appreciated nonetheless because I had never considered the
ports issue. Thanks again!

For the record again, this Raspotify project is working perfectly!

JasonLG1979, 2020-10-20:
> Raspotify project is working perfectly!

I like it. I run it on a headless Pi Zero W with a hifiberry DAC + ZERO hat connected to my home stereo as basically a Spotify appliance. It works pretty good considering everything cost less than $50 shipped and only takes maybe 10 minutes to setup.

closed 2021-11-24 by github-actions[bot]