dtcooper/raspotify #298: Problem with credentials

KSuondProject, 2020-09-22:
Hi all,
I've a strange problem with the raspotify service.
It give me this error but the credentials are correct because if I enter these on the website of spotify these works fine.
Some one can help me please?

raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-09-23 00:08:24 CEST; 3s ago
  Process: 962 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 963 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 964 (librespot)
   CGroup: /system.slice/raspotify.service
           └─964 /usr/bin/librespot --name Krokify --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --username <xxxxxxx> --password <xxxxxxx> --device hw:1,0 --oneve

set 23 00:08:23 raspberrypi systemd[1]: Starting Raspotify...
set 23 00:08:24 raspberrypi systemd[1]: Started Raspotify.
set 23 00:08:24 raspberrypi librespot[964]: [2020-09-22T22:08:24Z INFO  librespot] librespot (raspotify v0.14.0) 6aac8a6 (2020-07-24). Built on 2020-07-26. Build ID: wPNyqn0m
set 23 00:08:24 raspberrypi librespot[964]: [2020-09-22T22:08:24Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-b-2t91.ap.spotify.com:4070"
set 23 00:08:25 raspberrypi librespot[964]: thread 'main' panicked at **'Authentication failed with reason: BadCredentials', core/src/connection/mod.rs:121:21**
set 23 00:08:25 raspberrypi librespot[964]: stack backtrace:
lines 1-15/15 (END)

JasonLG1979, 2020-09-30:
```--username``` and ```--password`` are not your account creds. They are your device username and password. Go to your account overview and your username will be a number. Next you need to set your device password by going to "Set Device Password" and following the instructions.

There's really no reason to do all that unless you're trying to make so that no other account can use the device.

KSuondProject, 2020-09-30:
> `--username` and `--password` are not your account creds. They are your device username and password. Go to your account overview and your username will be a number. Next you need to set your device [...]

I've already did it, I've a device username that is a several numbers and a special password for this.
Is not a mail and password that I use to login with Android or pc.

I've do that because I've logged in Spotify with Facebook account and I can't manual log in with this method in raspotify. Or not?

Really thanks in advance.

JasonLG1979, 2020-09-30:
> I've already did it, I've a device username that is a several numbers and a special password for this.
> Is not a mail and password that I use to login with Android or pc.
>
> I've do that because I've [...]

You don't need to supply raspotify (librespot) any creds at all. You don't need to login. By default it's zeroconf meaning that it's discoverable and when you connect to it it handles authentication automatically.

The only reason to manually set a username and password would be if your raspotify device is on a some kind of non-private/shared network and you didn't want anyone else to be able to connect to it. In that case you'd also want to pass ```-- disable-discovery``` to also make it so that only you can see it.

KSuondProject, 2020-09-30:
> You don't need to supply raspotify (librespot) any creds at all. You don't need to login. By default it's zeroconf meaning that it's discoverable and when you connect to it it handles authentication [...]

So in OPTIONS I don't need to write --username and --password because it's do an auto login when connect my app right? 
If yes it would be beautiful!

JasonLG1979, 2020-09-30:
> So in OPTIONS I don't need to write --username and --password because it's do an auto login when connect my app right?
> If yes it would be beautiful!

Yes.

You shouldn't have to change anything really. It should work out of the box. The only things I usually charge are ```DEVICE_NAME```,  ```BITRATE```, and I remove ```--enable-volume-normalisation``` from ```VOLUME_ARGS```.

But all of those are my own personal preferences, they don't affect connectivity.

KSuondProject, 2020-10-01:
> Yes.
>
> You shouldn't have to change anything really. It should work out of the box. The only things I usually charge are `DEVICE_NAME`, `BITRATE`, and I remove `--enable-volume-normalisation` from [...]

It's fantastic, I need to reflash a new raspbian OS to test it from a clean new ambient btw I hope this only the things that I've do wrong.

So really really thanks for the help @JasonLG1979 !!

Have a nice day!!

closed 2020-10-01 by KSuondProject

JasonLG1979, 2020-10-01:
> It's fantastic, I need to reflash a new raspbian OS to test it from a clean new ambient btw I hope this only the things that I've do wrong.

If you have any other questions or problems I may be able to help. Depending on your setup configuring audio optimally can be a bit tricky.

KSuondProject, 2020-10-02:
> > It's fantastic, I need to reflash a new raspbian OS to test it from a clean new ambient btw I hope this only the things that I've do wrong.
> 
> If you have any other questions or problems I may be able to help. Depending on your setup configuring audio optimally can be a bit tricky.

Thanks for you help! Is really appreciate!
I've an raspberry pi zero w + pimoroni pirate audio + pimoroni usb extender. I would like to use the pidi-sporify code to use Spotify connect on it but also see the cover art on the 1.7" TFT display of the audio card.
The code is this: https://github.com/pimoroni/pidi-spotify and raspotify 
Can you help me with this setup? Thanks a lot!

JasonLG1979, 2020-10-02:
> Thanks for you help! Is really appreciate!
> I've an raspberry pi zero w + pimoroni pirate audio + pimoroni usb extender. I would like to use the pidi-sporify code to use Spotify connect on it but [...]

I don't have a pimoroni pirate audio, but I also run librespot on a pi zero w. I have a couple USB DACs and a HIFIBERRY DAC+ ZERO, it uses the same overlay, so I'm guessing it's in at least in the same family if not the same DAC chip? If the pimoroni pirate audio DAC portion is like the DAC+ Zero then it lacks hardware volume control.

If that is the case, if want volume control you have 2 choices:
1. Use librespot's builtin software volume.
2. Use the alsa softvol plugin.

I prefer the alsa softvol plugin because you can up the bit depth to 32 bit (not to be confused with oversampling) and get lossless volume control. Volume control and volume normalization in librespot is lossy, it is done in 16 bit so any volume or gain reduction throws away bits.

I also find that a bigger audio buffer seems to make it run a little smoother on a pi zero. The default buffer size is 0.5 sec with a crazy small period size. I prefer a 1 sec buffer with 4 periods.

[This gist](https://gist.github.com/JasonLG1979/f29cb0f14218a3b20db5d32c2cfae5ed) is the asound.conf I use on my pi zero. It allows you to tweak a lot of different parameters by just changing one line. It's sorta a ```hwplug``` on steroids. 

Just open ```/etc/asound.conf``` in nano and paste it in, save and close it. And then add ```--mixer alsa```  in the ```OPTIONS=``` section of ```/etc/default/raspotify``` and restart raspotify ```sudo systemctl restart raspotify``` And you're all set audio wise.

JasonLG1979, 2020-10-02:
> I would like to use the pidi-sporify code to use Spotify connect on it but also see the cover art on the 1.7" TFT display of the audio card.

It looks like it was designed to work with raspotify. I'd say follow the instructions.

KSuondProject, 2020-10-03:
> > I would like to use the pidi-sporify code to use Spotify connect on it but also see the cover art on the 1.7" TFT display of the audio card.
> 
> It looks like it was designed to work with raspotify. I'd say follow the instructions.

With a fresh install something works but I've a new probelm when I start to use the raspotify service:

**raspberrypi librespot[1214]: ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred**

It seems that the raspberry doesn't see my Pimoroni Pirate audio sound card?

JasonLG1979, 2020-10-03:
> raspberrypi librespot[1214]: ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred

No that's just a buffer underrun.

JasonLG1979, 2020-10-03:
That's one reason I prefer a larger buffer. The buffer period size that the (out of date) version of librespot that is packaged in raspotify is crazy low. That causes buffer underruns and higher CPU usage.

The buffer is 0.5 sec (22050 frames/samples) but the period size is like 225 (again frames/samples) that translates to almost 100 CPU interrupts per sec. I like a 1 sec buffer with 4 periods per buffer which translates to 4 CPU interrupts per sec. A much smoother experience.

Did you try that ```asound.conf``` in that gist I wrote? It should be a drop in improvement.

JasonLG1979, 2020-10-03:
Another thing that might help is to disable CPU frequency scaling on the pi zero with the ```force_turbo=1``` option in ```/boot/config.txt``` it will use a tiny bit more power and run a couple degrees warmer but unless electricity is crazy expensive in your area and/or you keep your pi zero in an oven it shouldn't cause any issues. And if all you're using your pi zero for is raspotify (as I do) you can disable a few services and/or completely uninstall some things to lower the idle CPU usage a tiny bit also.

KSuondProject, 2020-10-06:
> That's one reason I prefer a larger buffer. The buffer period size that the (out of date) version of librespot that is packaged in raspotify is crazy low. That causes buffer underruns and higher CPU [...]

Yes I've tried your asound.config but it doesn't work. I'll report to you the exact error ASAP.
In origin the asound.config is empty? Whe I had used nano to modify the file was empty, is correct?

> Another thing that might help is to disable CPU frequency scaling on the pi zero with the `force_turbo=1` option in `/boot/config.txt` it will use a tiny bit more power and run a couple degrees [...]

Thanks again for this tips! I'll give it a try of sure.
BTY it's better use a version of raspberry OS without desktop? 
Maybe is lighter and more performant? 
Whats the better setup?

Another big thanks in advance.

JasonLG1979, 2020-10-06:
> Yes I've tried your asound.config but it doesn't work. I'll report to you the exact error ASAP.
> In origin the asound.config is empty? Whe I had used nano to modify the file was empty, is correct?

Weird. The pimoroni pirate audio uses the same overlay (drivers) but maybe the DAC hardware is a little different. It might require a bit of tweaking?

> BTY it's better use a version of raspberry OS without desktop?

Well you'd at least want to set it to boot to console. If you did that there'd be no real difference beyond the unused gui related packages taking up more space on the SD card. Generally speaking you'd run it headless (no GUI) and administer it via ssh and never even ever actually connect a keyboard and monitor directly to the pi zero.

The short answer is yes use the lite version.

KSuondProject, 2020-10-07:
> Weird. The pimoroni pirate audio uses the same overlay (drivers) but maybe the DAC hardware is a little different. It might require a bit of tweaking?

When I have a little bit of free time I modify the OPTIONS in the raspotify file config then i post here the error.

> Well you'd at least want to set it to boot to console. If you did that there'd be no real difference beyond the unused gui related packages taking up more space on the SD card. Generally speaking [...]

Right, I try the lighter OS to have better performance. 
Now I use the raspberry with VNC to operate with my PC.
If there's no GUI I think that the performance will be better, or not?
BTW what are the things that I have to unistall to have more performance in the OS whit GUI?

JasonLG1979, 2020-10-07:
> If there's no GUI I think that the performance will be better, or not?

For sure. I would never run a desktop environment on a pi zero.

> BTW what are the things that I have to unistall to have more performance in the OS whit GUI?

I'd imagine It's about a lightweight desktop as you can get, I don't know? I don't run GUI's on any of my pi's never have. Except for RetroPi on a pi zero to play old Nintendo ROM's. I've always used my pi's as mini home servers or appliances, stuff like PiHole, Plex server, print/scanner server, file server, Bluetooth audio receiver and so on. To me that's where they shine.

KSuondProject, 2020-10-18:
> For sure. I would never run a desktop environment on a pi zero.
>
>
> I'd imagine It's about a lightweight desktop as you can get, I don't know? I don't run GUI's on any of my pi's never have. Except [...]

Hi Jason, 
I've reinstalled the Raspeberry OS without the GUI and i've setup all for the pidi-spotify. Can I ask you help for the code? Because the raspotify code works like a charm but when I try to use the code of Pimoroni all stops to work. If we are off topic can I ask to you to write me in some other platform (mail, facebook or sometinhg else) please?

JasonLG1979, 2020-10-18:
If raspotify works just fine then it's a pidi-spotify problem. I'd file a bug report there. I don't have the hardware I can't be of much help with that part.