dtcooper/raspotify #299: Raspotify connected but no sound

aholloway17, 2020-09-27:
I recently in stalled the easy version of Spotify connect. While I can connect there is not sound. I am able to play and hear music from files on the Pi 3 B just not through Raspotify. It may be of note I am using the new DAC2 HD. Below is from aplay -l and service config for Raspotify.

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 2: sndrpihifiberry [snd_rpi_hifiberry_dacplushd], device 0: HiFiBerry DAC+ HD HiFi pcm179x-hifi-0 [HiFiBerry DAC+ HD HiFi pcm179x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Description=Raspotify
After=network.target
[Service]
User=raspotify
Group=raspotify
Restart=always
RestartSec=10
PermissionsStartOnly=true
ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify ; /bin/chown raspotify:raspotify /var/cache/raspotify
Environment="DEVICE_NAME=raspotify (%H)"
Environment="BITRATE=320"
Environment="CACHE_ARGS=--disable-audio-cache"
Environment="VOLUME_ARGS=--enable-volume-normalisation --linear-volume --initial-volume=100"
Environment="BACKEND_ARGS=--backend alsa"
EnvironmentFile=-/etc/default/raspotify
ExecStart=/usr/bin/librespot --device hw:2,0 $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS

[Install]
WantedBy=multi-user.target

Any assistance would be gratefully received,

Cheers,

Allen

Callenberg, 2020-09-27:
Hello Allen,
One thing that seems wrong to me, after I just had a quick look, is that `--linear-volume` should be -`-mixer-linear-volume`
i.e. `VOLUME_ARGS="--enable-volume-normalisation --mixer-linear-volume --initial-volume=100"`

Since the new version of librespot has changed some of it's option flags.
Also check out this #293 

Hopefully you will now get one step closer...

JasonLG1979, 2020-10-02:
@Callenberg ```--mixer-linear-volume``` is not a valid arg for the version of librespot that is currently packaged with raspotify. In the issue you link to the OP replaced the packed version of librespot with a compiled version of librespot from the git dev branch.

librespot doesn't have a proper ```--help``` arg but if you call it without any args it will throw an error show you the available args and exit.

@aholloway17 Unless you've already changed it the problem is more than likely that the DAC you want to use is card 2. By default card 0 is the default which in your case is HDMI.

See https://github.com/dtcooper/raspotify#troubleshooting Except you'd want to change it to 2 not 1.

closed 2021-11-24 by github-actions[bot]