dtcooper/raspotify #300: Update librespot

JasonLG1979, 2020-09-30:
The current master (as of this comment) has an alsa backend fix that improves buffering behaviour and an improved volume implementation. Well worth the update.

skywinder, 2020-10-07:
@JasonLG1979 I ask here: https://github.com/dtcooper/raspotify/issues/277#issuecomment-683540674

Do you know how to update it?

looks like it just uses HEAD from main repo:

https://github.com/dtcooper/raspotify/blob/822c0d80616e85a45ee4ddf26d1550a812dc4c99/build.sh#L29

skywinder, 2020-10-07:
can you specify up-to-date  commit/branch/tag that we have to refer?

JasonLG1979, 2020-10-07:
The problem is that the binary used by raspotify was built on or about the 26th of July. 1.3 wasn't actually released until the 29th and there were a couple important last minute merges. So raspotify isn't even at the latest stable version yet...

@dtcooper?

joshuajaharwood, 2020-10-08:
Would also be nice to have the ability to choose different backends, rather than having to compile them in ourselves.

JasonLG1979, 2020-10-08:
> Would also be nice to have the ability to choose different backends, rather than having to compile them in ourselves.

I think the general idea is that you'd use raspotify on a headless pi which would more than likely not have pulseaudio or gstreamer or whatever installed by default. The alsa backend doesn't require any additional dependencies. And compiling in additional backends would make the binary larger.

Besides this issue is strictly about updating it. If you would like additional backends that is a completely different issue.

skywinder, 2020-10-08:
@JasonLG1979 Are you talking about built binary?  which installation method are you using?

JasonLG1979, 2020-10-08:
> @JasonLG1979 Are you talking about built binary? which installation method are you using?

For raspotify? I just used the install script.

JasonLG1979, 2020-10-08:
I did build the binary of the latest librespot a couple weeks ago. Once on my desktop (Ubuntu 12.04) and once on my pi zero just to see how much slower the pi zero would be to build it (Wow,lol!!!)  If I ever build it again for my pi zero I'll be crosscompling it for sure,lol!!! I did so to see the difference between the buffering behaviour. The current stable version is much better behaved. I didn't get any buffer underruns unlike the current version built in raspotify.

In the meantime I've just overridden pcm.default in asound.conf and manually specified my own global buffer parmas.

dtcooper, 2020-10-12:
I'll get on this today, folks. I'll post here when it's done so you can test (I'll do so on a Pi 3 + 4)

JasonLG1979, 2020-10-13:
There's going to be a breaking config change with volume control.

dtcooper, 2020-10-18:
Would you folks mind testing the new version, especially on buster (I have stretch on my pi).

Link here: [v0.15.0 `.deb` file](https://github.com/dtcooper/raspotify/releases/download/0.15.0/raspotify_0.15.0.librespot.v0.1.3_armhf.deb)

CC: @JasonLG1979 @joshuajaharwood @skywinder

JasonLG1979, 2020-10-19:
@dtcooper I just updated and it gives me the standard options to keep my current config (/etc/default/raspotify) but the unit file is not overwritten on update. I don't know if that's a packing option or not? If not you're going to have to slip the volume change and log level change (if you merge it) in there. That seemed to work pretty well with the config. I kept my current config and it still fixed up the volume args.

joshuajaharwood, 2020-10-19:
Failed to install on an RPi 3b+, Raspbian Buster, kernel 5.4.51.

```
sudo dpkg -i raspotify_0.15.0.librespot.v0.1.3_armhf.deb
Selecting previously unselected package raspotify.
(Reading database ... 147419 files and directories currently installed.)
Preparing to unpack raspotify_0.15.0.librespot.v0.1.3_armhf.deb ...
dpkg: error processing archive raspotify_0.15.0.librespot.v0.1.3_armhf.deb (--install):
 new raspotify package pre-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 raspotify_0.15.0.librespot.v0.1.3_armhf.deb
```

JasonLG1979, 2020-10-19:
@joshuajaharwood try apt:

```sudo apt install ./raspotify_0.15.0.librespot.v0.1.3_armhf.deb```

joshuajaharwood, 2020-10-19:
@JasonLG1979 Still doesn't like it.

```
sudo apt install ./raspotify_0.15.0.librespot.v0.1.3_armhf.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'raspotify' instead of './raspotify_0.15.0.librespot.v0.1.3_armhf.deb'
The following NEW packages will be installed:
  raspotify
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/2,458 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 /home/pi/Downloads/raspotify_0.15.0.librespot.v0.1.3_armhf.deb raspotify armhf 0.15.0~librespot.v0.1.3 [2,458 kB]
(Reading database ... 147419 files and directories currently installed.)
Preparing to unpack .../raspotify_0.15.0.librespot.v0.1.3_armhf.deb ...
dpkg: error processing archive /home/pi/Downloads/raspotify_0.15.0.librespot.v0.1.3_armhf.deb (--unpack):
 new raspotify package pre-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 /home/pi/Downloads/raspotify_0.15.0.librespot.v0.1.3_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

JasonLG1979, 2020-10-19:
> -bash: 2: command not found

Do you not have ```sed``` installed?

JasonLG1979, 2020-10-19:
```sed``` is a basic utility. It's an "essential package", it comes with even the most minimal Debian install.  If for some strange reason you've uninstalled it, reinstall it:

```sudo apt -y install sed```

dtcooper, 2020-10-19:
It may have been that the command failed?

D

joshuajaharwood, 2020-10-19:
@JasonLG1979 sed is installed, but is failing. Sorry, I should've formatted the last code block a bit better. It runs, but returns error code 2 which causes installation to fail.

@dtcooper Could it be because /etc/default/raspotify doesn't currently exist?

JasonLG1979, 2020-10-19:
> It may have been that the command failed?

```command not found``` generally means you fat fingered it or you don't have the program installed. I will purge and reinstall and see if I can duplicate it.

joshuajaharwood, 2020-10-19:
> 
> 
> > It may have been that the command failed?
> 
> `command not found` generally means you fat fingered it or you don't have the program installed. I will purge and reinstall and see if I can duplicate it.

@JasonLG1979 The 'command not found' is from me running $? to get the return code of the last command run.

JasonLG1979, 2020-10-19:
Putting ```exit 0``` at the end of the script should fix the issue then.

JasonLG1979, 2020-10-19:
The right way to do it though is to check if the files exist:
```
#!/bin/sh

# Fix changes in command line arguments for librespot v0.1.3

if [ -f /etc/default/raspotify ]; then
    sed -i 's/--linear-volume/--volume-ctrl=linear/' /etc/default/raspotify > /dev/null 2>&1
fi

if [ -f /etc/systemd/system/raspotify.service ]; then
    sed -i 's/--linear-volume/--volume-ctrl=linear/' /etc/systemd/system/raspotify.service > /dev/null 2>&1
fi

exit 0
```

JasonLG1979, 2020-10-19:
@dtcooper btw I'm running raspotify on a pi zero so you can add that to your "tested to work with" list. Although it's basically a Pi 1 but not everyone may know that?

joshuajaharwood, 2020-10-19:
Let me know if #309 is acceptable. Thanks!

dtcooper, 2020-10-27:
Hi again, am back! Doing well. Getting on this today. Left a comment on your PR @joshuajaharwood.

skywinder, 2020-11-21:
Up 😁

OberstVonGatow, 2020-12-08:
Is it now possible to get a librespot update? How can i check my librespot version? 
I saw on a librespot thread that podcast playback should be possible now.

dtcooper, 2020-12-08:
It is. I need to resolve the long-standing issue of librespot's volume control argument renaming but it's in my immediate queue. Thanks all!

dtcooper, 2021-11-11:
I have limited bandwidth to work on this, folks. Sincerest apologies, but circumstances outside my control with my professional leave almost no time to work on this.

PRs are welcome!

JasonLG1979, 2021-11-12:
@dtcooper I can help maintain Raspotify if no one else steps up. I'm pretty good with alsa and I'm a recent `librespot` contributor so I can get help get fixes upstream to librespot. Docker is my weakest point but nothing I don't some judicious use of Google can't solve.

I put in a [PR to update Raspotify](https://github.com/dtcooper/raspotify/pull/447) that seems work fine for me. Testing welcome.

The next thing I would do is clear out most of the 236 bug reports. I'm guessing that most are invalid because of age or vatios other reasons.

dtcooper, 2021-11-16:
@JasonLG1979 I've added you as a maintainer. Go ahead and merge your PRs. If you can attach the `.deb` artifacts here, we can test them!

JasonLG1979, 2021-11-16:
@dtcooper My plan is to wait for https://github.com/librespot-org/librespot/pull/875 to be merged as it represents some quality of life improvements as far as error handling and arg parsing (as in librespot will tell the user what they did wrong in plain English and exit cleanly as apposed to just exploding) and it introduces the `--quiet` flag. I would also like feedback from people about this https://github.com/dtcooper/raspotify/pull/449 (I know I didn't need to put in a PR but I want feedback before I merge it).

Here is a build of the hardened branch. To really test it you have to completely purge raspotify and delete the old systemd unit file before you install since the unit file is not overwritten on update.

https://drive.google.com/file/d/1WhOhuGUBdLsJmdEZBnrwEv1XeSH7sgpS/view?usp=sharing

JasonLG1979, 2021-11-16:
I'd like the next version to be 0.31.0 to align with Librespot's current version.

dtcooper, 2021-11-16:
@JasonLG1979 Stupid question (don't have time to dig into changes), does this change/rewrite/remove deprecate switches in `/etc/defaults/raspotify`? I worry that an upgrade will break existing installations.

JasonLG1979, 2021-11-16:
> @JasonLG1979 Stupid question (don't have time to dig into changes), does this rewrite/remove deprecated command-line arguments to `librespot` in `/etc/defaults/raspotify`? I worry that an upgrade will break existing installations.

Yes that was taken care of in https://github.com/dtcooper/raspotify/commit/10a7c9abcec6258304177a93434419d7f2d30edd

https://github.com/dtcooper/raspotify/blob/master/raspotify/DEBIAN/postinst#L28-L51

JasonLG1979, 2021-11-16:
In any event if it does fail for any reason the user will be given a nice hint in their systemd logs explaining exactly what failed and why. That's mostly what https://github.com/librespot-org/librespot/pull/875 is about. Making librespot less explodey when presented with malformed/invalid args and such and more helpful by presenting useful error messages for example:

```rust
                error!("Invalid `--{}` / `-{}`: {}", FORMAT, FORMAT_SHORT, format);
                println!(
                    "Valid `--{}` / `-{}` values: F64, F32, S32, S24, S24_3, S16",
                    FORMAT, FORMAT_SHORT
                );
                println!("Default: {:?}", AudioFormat::default());
                exit(1);
```

Currently if you pass a bad value to an option librespot will crash and puke a stack trace which is not very handy...

JasonLG1979, 2021-11-16:
That PR is not for the faint at heart. I completely rewrote main.rs.

JasonLG1979, 2021-11-17:
That .deb I posted runs like a charm on my Pi Zero v1.3 with a HiFiBerry DAC+ ZERO.

[image]

JasonLG1979, 2021-11-17:
There you go @dtcooper I issued a new release I just need you to do your thing with the repo.

closed 2021-11-17 by JasonLG1979