dtcooper/raspotify #301: It just stopped working...

spietrowski, 2020-09-30:
I did a full reinstall of raspotify, but i can't hear anything on my pi.
youtube and even the spotify webplayer is working, so it is not an hardware error.
spotify can connect and looks like it is playing on the pi.

any suggestings?

baybled, 2020-10-01:
can you check alsamixer volume?

spietrowski, 2020-10-02:
I just did, it's at 99%
[image]

JasonLG1979, 2020-10-02:
What does your ```/etc/default/raspotify``` look like and what does ```systemctl status raspotify``` say while it's supposed to be playing? You also mention spotify webplayer are you running it on a Pi with a DE? It would be helpful if provided a little detail about your system.

spietrowski, 2020-10-02:
My settings are looking good. I just added my username and password and changed the device name. 

The status is saying:
`pi@mirrorpi2:~ $ systemctl status raspotify
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset
   Active: active (running) since Wed 2020-09-30 23:26:20 CEST; 2 days ago
  Process: 789 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exi
  Process: 793 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify 
 Main PID: 795 (librespot)
    Tasks: 5 (limit: 2065)
   CGroup: /system.slice/raspotify.service
           └─795 /usr/bin/librespot --name Badezimmerspiegel2.0 --backend alsa -

Okt 03 01:03:02 mirrorpi2 librespot[795]: [2020-10-02T23:03:02Z ERROR librespot_
Okt 03 01:03:02 mirrorpi2 librespot[795]: [2020-10-02T23:03:02Z ERROR librespot_
Okt 03 01:03:07 mirrorpi2 librespot[795]: ALSA lib confmisc.c:1281:(snd_func_ref
Okt 03 01:03:07 mirrorpi2 librespot[795]: ALSA lib conf.c:4568:(_snd_config_eval
Okt 03 01:03:07 mirrorpi2 librespot[795]: ALSA lib confmisc.c:251:(snd_func_gete
Okt 03 01:03:07 mirrorpi2 librespot[795]: ALSA lib conf.c:4568:(_snd_config_eval
Okt 03 01:03:07 mirrorpi2 librespot[795]: ALSA lib conf.c:5036:(snd_config_expan
Okt 03 01:03:07 mirrorpi2 librespot[795]: ALSA lib pcm.c:2565:(snd_pcm_open_noup
Okt 03 01:03:07 mirrorpi2 librespot[795]: [2020-10-02T23:03:07Z ERROR librespot_
Okt 03 01:03:07 mirrorpi2 librespot[795]: [2020-10-02T23:03:07Z ERROR librespot_
lines 1-20/20 (END)`

I ran the web player on the pi's desktop browser

JasonLG1979, 2020-10-02:
Your logs are chopped off. You're getting errors but I can't read them.

spietrowski, 2020-10-03:
ah crap! you are right...
sorry it was late yesterday...
`pi@mirrorpi2:~ $ systemctl status raspotify
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-10-03 20:15:29 CEST; 4min 28s ago
  Process: 751 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 752 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 753 (librespot)
    Tasks: 4 (limit: 2065)
   CGroup: /system.slice/raspotify.service
           └─753 /usr/bin/librespot --name Badezimmerspiegel2.0 --backend alsa --bitrate 160 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --username %MYUSERNAME% --password %MYPASSWORD%

Okt 03 20:19:52 mirrorpi2 librespot[753]: [2020-10-03T18:19:52Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Okt 03 20:19:52 mirrorpi2 librespot[753]: [2020-10-03T18:19:52Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Okt 03 20:19:57 mirrorpi2 librespot[753]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.pcm.card'
Okt 03 20:19:57 mirrorpi2 librespot[753]: ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Okt 03 20:19:57 mirrorpi2 librespot[753]: ALSA lib confmisc.c:251:(snd_func_getenv) error evaluating default
Okt 03 20:19:57 mirrorpi2 librespot[753]: ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_getenv returned error: No such file or directory
Okt 03 20:19:57 mirrorpi2 librespot[753]: ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
Okt 03 20:19:57 mirrorpi2 librespot[753]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM default
Okt 03 20:19:57 mirrorpi2 librespot[753]: [2020-10-03T18:19:57Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Okt 03 20:19:57 mirrorpi2 librespot[753]: [2020-10-03T18:19:57Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
`

After seeing this... I am so confused on the "no such file or directory" and "Alsa error" error messages...

btw, thanks for your help!

JasonLG1979, 2020-10-03:
> After seeing this... I am so confused on the "no such file or directory" and "Alsa error" error messages...

librespot is trying to use ```defaults.pcm.card``` but whatever you have set to be your default sound card does not exist, or is already being used by another application. Without software mixing your sound card can only be used by one application at a time. If your system is running pulseaudio you may need to tell librespot to use pulse as your device as pulseaudio maybe perpetually monopolizing your sound card.

JasonLG1979, 2020-10-03:
I generally never run a full fat desktop install on any model of Raspberry Pi (as I am not a masochist) I run raspotify on a headless Pi Zero W configured specifically to just be a Spotify Connect device. Properly configured it works rather well. I get sub 40 MB total system RAM usage and about 20% CPU usage during regular play back with spikes up to 60% or 70% for a couple seconds while songs are preloaded. I have no experience with pulseaudio on a Raspberry Pi. It's of no value to me. I can duplicate whatever functionality it provides that I need with plain old ALSA.

ssombra, 2020-10-03:
> I generally never run a full fat desktop install on any model of Raspberry Pi (as I am not a masochist) I run raspotify on a headless Pi Zero W configured specifically to just be a Spotify Connect [...]

Do you use the Zero W with an audio hat?

JasonLG1979, 2020-10-03:
> Do you use the Zero W with an audio hat?

Yes, I have a HIFIBERRY DAC+ ZERO and a couple cheap USB DACs, a JDS Labs OLDAC and a HiFimeDIY Tiny USB DAC. They all work just fine. The JDS Labs OLDAC sounds the best ofc but the HIFIBERRY DAC+ ZERO is serviceable. I like the HIFIBERRY DAC+ ZERO because it's a "neater" solution and I can use the JDS Labs OLDAC on my computer. i've got my eye on a allo miniboss though.

spietrowski, 2020-10-03:
> librespot is trying to use `defaults.pcm.card` but whatever you have set to be your default sound card does not exist, or is already being used by another application. Without software mixing your [...]

The headphone jack is my output device... It was working for weeks and than stopped...

spietrowski, 2020-10-03:
> I generally never run a full fat desktop install on any model of Raspberry Pi (as I am not a masochist) I run raspotify on a headless Pi Zero W configured specifically to just be a Spotify Connect [...]

I wouldn't run a full desktop on a zero either. 
But I do not see any problems by running the lite desktop on a pi 3.
Tbh, I don't know how running a desktop on a pi has anything to do with being a masochist... 

Since my pi is also running magic mirror2, I switched to the lite desktop.
And yes raspotify is working with the magic mirror and no I did not change anything.

JasonLG1979, 2020-10-04:
Calm down killer,lol!!!

What's the output of ```aplay -l``` and ```aplay -L``` and while playing other audio does ```speaker-test -c 2``` work?

JasonLG1979, 2020-10-04:
> Tbh, I don't know how running a desktop on a pi has anything to do with being a masochist...

It's painfully slow compared to my other computers.

spietrowski, 2020-10-04:
> Calm down killer,lol!!!
> 
> What's the output of `aplay -l` and `aplay -L` and while playing other audio does `speaker-test -c 2` work?

:D i'm calm, i just angry about this stupid problem...

The Outputs are:
`pi@mirrorpi2:~ $ aplay -l
**** Liste der Hardware-Geräte (PLAYBACK) ****
Karte 0: b1 [bcm2835 HDMI 1], Gerät 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Sub-Geräte: 4/4
  Sub-Gerät #0: subdevice #0
  Sub-Gerät #1: subdevice #1
  Sub-Gerät #2: subdevice #2
  Sub-Gerät #3: subdevice #3
Karte 1: Headphones [bcm2835 Headphones], Gerät 0: bcm2835 Headphones [bcm2835 Headphones]
  Sub-Geräte: 3/4
  Sub-Gerät #0: subdevice #0
  Sub-Gerät #1: subdevice #1
  Sub-Gerät #2: subdevice #2
  Sub-Gerät #3: subdevice #3
pi@mirrorpi2:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
output
sysdefault:CARD=b1
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Default Audio Device
dmix:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Direct sample mixing device
dsnoop:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Direct sample snooping device
hw:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Direct hardware device without any conversions
plughw:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Hardware device with all software conversions
sysdefault:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
dmix:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample mixing device
dsnoop:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample snooping device
hw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct hardware device without any conversions
plughw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Hardware device with all software conversions
`

The speakertest ist working fine....

> It's painfully slow compared to my other computers. Edit: Even my Pi3B+.
Yeah but isn't that obvious? lol

spietrowski, 2020-10-04:
I connected a USB Soundcard right now and....

a) Speakertest is working
b) web radio is working
c) spotify is not working

output is as followed:

pi@mirrorpi2:~ $ aplay -l
**** Liste der Hardware-Geräte (PLAYBACK) ****
Karte 0: Headphones [bcm2835 Headphones], Gerät 0: bcm2835 Headphones [bcm2835 Headphones]
  Sub-Geräte: 8/8
  Sub-Gerät #0: subdevice #0
  Sub-Gerät #1: subdevice #1
  Sub-Gerät #2: subdevice #2
  Sub-Gerät #3: subdevice #3
  Sub-Gerät #4: subdevice #4
  Sub-Gerät #5: subdevice #5
  Sub-Gerät #6: subdevice #6
  Sub-Gerät #7: subdevice #7
Karte 1: Device [USB Audio Device], Gerät 0: USB Audio [USB Audio]
  Sub-Geräte: 0/1
  Sub-Gerät #0: subdevice #0
pi@mirrorpi2:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
output
sysdefault:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
dmix:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample mixing device
dsnoop:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample snooping device
hw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct hardware device without any conversions
plughw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Hardware device with all software conversions
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions
pi@mirrorpi2:~ $ systemctl status raspotify
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-10-04 14:36:07 CEST; 38min ago
  Process: 700 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 701 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 702 (librespot)
    Tasks: 4 (limit: 2065)
   CGroup: /system.slice/raspotify.service
           └─702 /usr/bin/librespot --name Badezimmerspiegel2.0 --backend alsa --bitrate 160 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100 --username kepan0 --password Balb0815#*

Okt 04 14:58:37 mirrorpi2 librespot[702]: [2020-10-04T12:58:37Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Okt 04 14:58:37 mirrorpi2 librespot[702]: [2020-10-04T12:58:37Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Okt 04 14:58:42 mirrorpi2 librespot[702]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.pcm.card'
Okt 04 14:58:42 mirrorpi2 librespot[702]: ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Okt 04 14:58:42 mirrorpi2 librespot[702]: ALSA lib confmisc.c:251:(snd_func_getenv) error evaluating default
Okt 04 14:58:42 mirrorpi2 librespot[702]: ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_getenv returned error: No such file or directory
Okt 04 14:58:42 mirrorpi2 librespot[702]: ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
Okt 04 14:58:42 mirrorpi2 librespot[702]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM default
Okt 04 14:58:42 mirrorpi2 librespot[702]: [2020-10-04T12:58:42Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Okt 04 14:58:42 mirrorpi2 librespot[702]: [2020-10-04T12:58:42Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed

JasonLG1979, 2020-10-04:
> The speakertest ist working fine....

That means your problem is something else is already using your sound card when ```speaker-test``` is trying to use it. something else is directly monopolizing your sound card. That something could be another app or pulseaudio.

If you have pulseaudio installed. Most desktop apps are "pulseaudio aware" meaning they will try to go though pulseaudio so they'll work just fine. ```speaker-test``` and ```librespot``` (compiled with the alsa backend) are not pulseaudio aware. They try to talk directly with alsa.  They fail because pulseaudio perpetually monopolizes your sound card.

If you do not have pulseaudio installed make sure no other app directly uses the hardware output bypassing dmix (software mixing). If an app directly uses the hardware output no other app can use it. That access is 1st come 1st served.

JasonLG1979, 2020-10-04:
Have you edited ```/etc/asound.conf```?

JasonLG1979, 2020-10-04:
If you've overridden ```default``` in /etc/asound.conf to point directly to hardware that would explain things.

spietrowski, 2020-10-04:
> In short something is misconfigured.

meh....
I did not install pulseaudio, so i'm trying to figure out with app could also use my soundcard...

> Have you edited /etc/asound.conf?

not that i am aware of...
it is just a blank file

spietrowski, 2020-10-04:
Well... I found the Error...
I installed the [rpi-audio-receiver](https://github.com/nicokaiser/rpi-audio-receiver) (i wanted bluetooth support on my pi) and somehow this screwd up my raspotfiy installation.
to be honest: i'm not sure if i selected the option to install raspotify... 

I removed the folder and that did the trick!

Thank you @JasonLG1979 for all the help!

closed 2020-10-04 by spietrowski

JasonLG1979, 2020-10-04:
> (i wanted bluetooth support on my pi) and somehow this screwd up my raspotfiy installation.

```bluealsa-aplay``` (what's used used to pipe the bluetooth audio to your sound card) will monopolize your sound card if not configured correctly. If you want detailed instructions on how to add bluetooth receiver capabilities to your raspberry pi that will not block other apps look [here]( https://github.com/JasonLG1979/PiZero-Bluetooth-Audio-Receiver). I wrote it with a pi zero in mind but it should work just fine on a Pi 3 just fine.

JasonLG1979, 2020-10-04:
If you want really fine grain control of your sound card you can also try [this asound.conf](https://gist.github.com/JasonLG1979/f29cb0f14218a3b20db5d32c2cfae5ed). And install ```libasound2-plugins``` (```sudo apt install -y --no-install-recommends libasound2-plugins```)

With it you can change:
```
    # CARD
    # DEV
    # SUBDEV
    # SAMPLE_RATE
    # FORMAT
    # SAMPLE_RATE_CONVERTER
    # BUFFER_PERIODS
    # BUFFER_PERIOD_TIME
    # VOL_MIN_DB
    # VOL_MAX_DB
    # VOL_RESOLUTION
    # VOL_NAME
```

For the default output I'd change:
(this is the best setting for a hifiberry dac zero on a pi zero)
```slave.pcm "hqstereo20:FORMAT=S32_LE,BUFFER_PERIOD_TIME=250000,VOL_MIN_DB=-48.0,VOL_NAME=PCM"```

To:
```slave.pcm "hqstereo20"```

Or maybe:
```slave.pcm "hqstereo20:SAMPLE_RATE=48000,SAMPLE_RATE_CONVERTER=speexrate_medium"```

For your DAC you'd have to play with it and see what works best.