dtcooper/raspotify #302: Mono Output

firewalkwithmetotwinpeaks, 2020-10-04:
Hello,

I'm trying to configure raspotify to have a mono output.

I'm using a HIFIBERRY mini amp, I found some configurations, but I'm having a hard time to find how to use them in rapsotify.

Best regards

JasonLG1979, 2020-10-07:
You're going to have to be more specific. Mono can mean a few different things on a 2 channel amp.

I'm going to assume you're not trying to bridge the amp.

The other 2 options are you're only going to use 1 of the 2 channels or you want dual mono. Which is it?

firewalkwithmetotwinpeaks, 2020-10-11:
Hello,
I am sorry for the vagueness of my request.
My goal is to use only one channel of the soundcard. Not dual mono.

Best regards,

JasonLG1979, 2020-10-11:
You'll want to use the Route alsa plugin.

Untested but something like this might do the trick:

```
pcm.!default {
    type route
    slave.pcm {
        type hw
        card 0
        channels 2
    }
    ttable.0.0 0.5
    ttable.0.1 0.5
    ttable.1.0 0.5
    ttable.1.1 0.5
}
```

This is for dual mono but I don't think it would hurt anything if one channel doesn't have a speaker connected to it.

JasonLG1979, 2020-10-11:
You're giving up half your power only using one channel. You could use both channels with one speaker if it were dual voice coil or coaxial.

JasonLG1979, 2020-10-11:
You're also going to have to use a pretty efficient speaker with only 3 x 2 watts. The specs don't say if 3 watts in the max or continuous so I'm going to assume it's the max so you're looking at maybe 2 watts clean. You'll need a speaker in the high 80's as far as SPL at least.

firewalkwithmetotwinpeaks, 2020-10-14:
Thank you very much
One more question, which file should I modify with these settings?

JasonLG1979, 2020-10-14:
1. Open ```/etc/asound.conf``` with nano:

   ```sudo nano /etc/asound.conf```

   *By default the file does not exist if there is anything in the file delete it.*

2. Paste this into it:

   ```
   pcm.!default {
       type route
       slave.pcm {
           type hw
           card 0
           channels 2
       }
       ttable.0.0 0.5
       ttable.0.1 0.5
       ttable.1.0 0.5
       ttable.1.1 0.5
   }
   ```

3. Exit and save:

   <kbd>Ctrl</kbd>+<kbd>x</kbd> <kbd>y</kbd> <kbd>Enter</kbd>

4. Restart raspotify:

   ```sudo systemctl restart raspotify```

firewalkwithmetotwinpeaks, 2020-10-14:
Thanks ! It works
I had created the right file, but the instructions for ALSA must have been wrong.

JasonLG1979, 2020-10-17:
@firewalkwithmetotwinpeaks if this solution works for you you can close the issue.

closed 2021-11-24 by github-actions[bot]