dtcooper/raspotify #306: Two way bluetooth (audio + controls) possible?

ssshake, 2020-10-13:
Hi I was skimming the issues for bluetooth related posts and they all seemed pretty involved and only address the audio streaming portion of it.

I have bluetooth book shelf speakers which include a remote. When I use Spotify directly from my phone, over bluetooth, to these speakers, I am able to pause and skip tracks using the remote. 

I'm unclear from what I'm reading in the existing issues whether this is implied functionality. Or unsupported.

If I manage to get my pi connected to my bluetooth speakers, will I then be able to use the remote controls?

Thanks for any advice. I haven't had to manually configure anything with pulse audio or alsa in like a decade. And I remember not having a fun time with it back then. 

I'm not sure if a simple a2dp device has two way control out of the box. 

I'm not sure if a spotify client can even dictate these things, or if "next track" commands must come from the initiators (my phone)

JasonLG1979, 2020-10-15:
If you're going Spotify app/phone > bluetooth > speaker already I'm not sure what the benefit of adding raspotify/librespot to the chain would be unless you're using a DAC with the pi and going in a line level/aux input of your speaker maybe? The builtin Pi audio is garbage. Bluetooth from your phone would probably sound better in that case without a DAC. Adding raspotify/librespot to the chain if the audio is coming in via bluetooth is just adding an extra link in the chain for no reason.

As far as I know librespot can only be controlled by a Spotify app. You could use [spotifyd](https://github.com/Spotifyd/spotifyd) But even then you'd have a lot of glueing to do as far as dealing with handling bluetooth media controls.

ssshake, 2020-10-16:
It's not an extra link in the chain for no reason. And I don't why my questions are coming across that way. With a headless spotify client you need some device with some user interface to pick which songs you want to listen to. That could be a phone, it couple be a PC with a web browser, it could be anything. 

The point of using raspotify is so that I can move the heavy lifting from the phone to the pi. It's the same thing as when using your phone to control music played on a google home. You aren't streaming from your phone, the google home does the streaming. Your phone is nothing more than a high end remote at that point.

So the reason is identical. I want to be able to pick what music is played using the refined phone interface. But I don't want the phone doing any of the processing. I don't want to music to cut out when I walk out to the garage for a moment while the phones in my pocket.

I thought that was the entire point of this project. To have a dedicated computer streaming from spotify. The only additional step I'm doing is wanting to pair the pi to bluetooth speakers instead of wires speakers. And because it's bluetooth speakers I want to know if I can take advantage of the music controls. Since pretty much any bluetooth media program supports media controls.  

As for the audio on the pi being garbage I would figure that's a universal problem for any user of raspotify and has nothing to do with my specific query. So does that mean the sentiment is nobody should use this project because it'll sound like crap? If so that's a pretty compelling point.

My whole goal is for a dedicated spotify streamer which can play music through my main speakers. 

I didn't want to use the analog audio output on the pi. I wanted to use bluetooth to a speaker, just like how my phone would work. I suppose I'm hoping that the bluetooth audio isn't crap like the analog out. But I know nothing on that matter. I just assumed one is streaming bits while the other is converting to an analog signal. So to your point about an ADC would that even apply in a bluetooth scenario?

I ask about the media controls because for all I know it's easy. For example when I pair my laptop to my bluetooth speakers, and load up spotify in the web browser (not even the stand alone app), the media controls work. I can hit pause on my speaker's remote, and it will pause music on my laptop, in chrome.

So I didn't know if that's one of those things you just get for free with a standard bluetooth library or not since it seems widely supported.

ssshake, 2020-10-16:
I guess my whole post could have been simplified down to one question.

Is raspotify compliant with the bluetooth media control standards? 

Which is either A2DP or AVRCP. I'm not sure which exact protocol supplies the media control capabilities. 

I think it would look something like this. And since we know linux supports the protocol it would be a matter of if raspotify handles these events.

bluetooth --> linux --> pulse audio --> raspotify

JasonLG1979, 2020-10-16:
> Is raspotify compliant with the bluetooth media control standards?

No. I already answered you. No open source Spotify Connect client that I am aware of is. You'd have to glue the 2 together with some scripts or whatever. 

> bluetooth --> linux --> pulse audio --> raspotify

The version of librespot compiled with raspotify does not include the pulseaudio backend. Even then the audio backends are only concerned with playback.

closed 2020-10-16 by ssshake