dtcooper/raspotify #307: Raspotify losing connection

yoniw, 2020-10-18:
Hi,
I have installed Raspotify about 3 weeks ago and everything worked as expected.
Ever since 1-2 weeks ago, Raspotify device is unselected automatically from Spotify after about 2 minutes. Meaning I have to re-select Raspotify every time.

Is there any fix for this issue?
Thank you :)

JasonLG1979, 2020-10-19:
What does your network look like? I ask because I was having the same issue when raspotify was Double-NAT'd.

yoniw, 2020-10-19:
> What does your network look like? I ask because I was having the same issue when raspotify was Double-NAT'd.

Just a regular setup, a router and a wifi dongle.
It worked with the same network setup before.

JasonLG1979, 2020-10-19:
> It worked with the same network setup before.

Before what? What I mean is things don't stop working for no reason. What is the difference between then and now software wise if any?

Logs would also be helpful:

```journalctl -u raspotify.service -b```

As would a list of any changes you may have made to the default config if any.

You might also try a wired connection to see if that has any affect if possible.

yoniw, 2020-10-20:
Thank you for your help.

I meant it worked until two weeks ago and all of a sudden it started to disconnect after 1 song or less, without any changes to the software or network setup.
No changes made in the config.
Unfortunately I currently don't have a cable to try a wired connection.
 
I have uninstalled and installed raspotify, and it still happens. 
Here's the log. However the trace timestamps aren't correlated with the dis/connections:

`-- Logs begin at Mon 2020-10-19 16:17:01 BST, end at Tue 2020-10-20 11:58:05 BST. --
Oct 19 16:17:07 raspberrypi systemd[1]: Starting Raspotify...
Oct 19 16:17:07 raspberrypi systemd[1]: Started Raspotify.
Oct 19 16:17:07 raspberrypi librespot[383]: [2020-10-19T15:17:07Z INFO  librespot] librespot (raspotify v0.14.0) 6aac8a6 (2020-07-24). Built on 2020-07-26. Build ID: wPNyqn0m
Oct 19 16:17:09 raspberrypi librespot[383]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 19, kind: Other, message: "No such device" }', src/main.rs:422:22
Oct 19 16:17:09 raspberrypi librespot[383]: stack backtrace:
Oct 19 16:17:11 raspberrypi librespot[383]:    0:   0x8bd584 - backtrace::backtrace::libunwind::trace::had1ce870e7642f03
Oct 19 16:17:11 raspberrypi librespot[383]:                        at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/libunwind.rs:86
Oct 19 16:17:11 raspberrypi librespot[383]:    1:   0x8bd584 - backtrace::backtrace::trace_unsynchronized::he46c884781dd51a8
Oct 19 16:17:11 raspberrypi librespot[383]:                        at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/mod.rs:66
Oct 19 16:17:11 raspberrypi librespot[383]:    2:   0x8bd584 - std::sys_common::backtrace::_print_fmt::h621902e5e498623c
Oct 19 16:17:11 raspberrypi librespot[383]:                        at src/libstd/sys_common/backtrace.rs:78
Oct 19 16:17:11 raspberrypi librespot[383]:    3:   0x8bd584 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h39c43d9eea8febb9
Oct 19 16:17:11 raspberrypi librespot[383]:                        at src/libstd/sys_common/backtrace.rs:59
Oct 19 16:17:12 raspberrypi librespot[383]:    4:   0x8e4e88 - core::fmt::write::h1faa4c62c7847860
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libcore/fmt/mod.rs:1076
Oct 19 16:17:12 raspberrypi librespot[383]:    5:   0x8b6cd4 - std::io::Write::write_fmt::h1d02362bb0653180
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/io/mod.rs:1537
Oct 19 16:17:12 raspberrypi librespot[383]:    6:   0x8bfe68 - std::sys_common::backtrace::_print::hb6e6bf916a89634f
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/sys_common/backtrace.rs:62
Oct 19 16:17:12 raspberrypi librespot[383]:    7:   0x8bfe68 - std::sys_common::backtrace::print::h70b0c25b24fd7afb
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/sys_common/backtrace.rs:49
Oct 19 16:17:12 raspberrypi librespot[383]:    8:   0x8bfe68 - std::panicking::default_hook::{{closure}}::hc261daa3190f9e66
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/panicking.rs:198
Oct 19 16:17:12 raspberrypi librespot[383]:    9:   0x8bfb1c - std::panicking::default_hook::h5ce743d47ceb319b
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/panicking.rs:218
Oct 19 16:17:12 raspberrypi librespot[383]:   10:   0x8c0520 - std::panicking::rust_panic_with_hook::hb609da6fd4918712
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/panicking.rs:486
Oct 19 16:17:12 raspberrypi librespot[383]:   11:   0x8c013c - rust_begin_unwind
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/panicking.rs:388
Oct 19 16:17:12 raspberrypi librespot[383]:   12:   0x8e2560 - core::panicking::panic_fmt::h48562d60f069801a
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libcore/panicking.rs:101
Oct 19 16:17:12 raspberrypi librespot[383]:   13:   0x8e23b8 - core::option::expect_none_failed::hde7f522962a61940
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libcore/option.rs:1272
Oct 19 16:17:12 raspberrypi librespot[383]:   14:   0x4ae3a0 - librespot::main::h6424af6f8ac98335
Oct 19 16:17:12 raspberrypi librespot[383]:   15:   0x4b1538 - std::rt::lang_start::{{closure}}::h20fd6404e8f8e33a
Oct 19 16:17:12 raspberrypi librespot[383]:   16:   0x8c08e4 - std::rt::lang_start_internal::{{closure}}::he71ca971345dcdbf
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/rt.rs:52
Oct 19 16:17:12 raspberrypi librespot[383]:   17:   0x8c08e4 - std::panicking::try::do_call::h3ebcd0c1fd722d55
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/panicking.rs:297
Oct 19 16:17:12 raspberrypi librespot[383]:   18:   0x8c08e4 - std::panicking::try::he6a0b75af361a6ec
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/panicking.rs:274
Oct 19 16:17:12 raspberrypi librespot[383]:   19:   0x8c08e4 - std::panic::catch_unwind::h2fbf955b2ece72b7
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/panic.rs:394
Oct 19 16:17:12 raspberrypi librespot[383]:   20:   0x8c08e4 - std::rt::lang_start_internal::ha55d91cd07438bce
Oct 19 16:17:12 raspberrypi librespot[383]:                        at src/libstd/rt.rs:51
[... cut ...]
Oct 19 16:17:12 raspberrypi librespot[383]:                        at /build/glibc-FUvrFr/glibc-2.28/csu/libc-start.c:308
Oct 19 16:17:12 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=101/n/a
Oct 19 16:17:12 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Oct 19 16:17:22 raspberrypi systemd[1]: raspotify.service: Service RestartSec=10s expired, scheduling restart.
Oct 19 16:17:22 raspberrypi systemd[1]: raspotify.service: Scheduled restart job, restart counter is at 1.
Oct 19 16:17:22 raspberrypi systemd[1]: Stopped Raspotify.
`

yoniw, 2020-10-20:
Thank you for your help.

I meant that it worked until two weeks ago, and then all of a sudden it started to disconnect after one song or less.
No changes made to the software or network setup.
No changes made to the config.
Unfortunately at the moment I don't have a cable to try a wired connection.
Here's the log:

> pi@raspberrypi:~ $ journalctl -u raspotify.service -b
> -- Logs begin at Tue 2020-10-20 12:03:07 BST, end at Tue 2020-10-20 12:06:55 BST. --
> Oct 20 12:03:14 raspberrypi systemd[1]: Starting Raspotify...
> Oct 20 12:03:14 raspberrypi systemd[1]: Started Raspotify.
> Oct 20 12:03:14 raspberrypi librespot[393]: [2020-10-20T11:03:14Z INFO  librespot] librespot (raspotify v0.14.0) 6aac8a6 (2020-07-24). Built on 2020-07-26. Build ID: wPNyqn0m
> Oct 20 12:03:16 raspberrypi librespot[393]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 19, kind: Other, message: "No such device" }', src/main.rs:422:22
> Oct 20 12:03:16 raspberrypi librespot[393]: stack backtrace:
> Oct 20 12:03:18 raspberrypi librespot[393]:    0:   0x8f8584 - backtrace::backtrace::libunwind::trace::had1ce870e7642f03
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/libunwind.rs:86
> Oct 20 12:03:18 raspberrypi librespot[393]:    1:   0x8f8584 - backtrace::backtrace::trace_unsynchronized::he46c884781dd51a8
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/mod.rs:66
> Oct 20 12:03:18 raspberrypi librespot[393]:    2:   0x8f8584 - std::sys_common::backtrace::_print_fmt::h621902e5e498623c
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/sys_common/backtrace.rs:78
> Oct 20 12:03:18 raspberrypi librespot[393]:    3:   0x8f8584 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h39c43d9eea8febb9
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/sys_common/backtrace.rs:59
> Oct 20 12:03:18 raspberrypi librespot[393]:    4:   0x91fe88 - core::fmt::write::h1faa4c62c7847860
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libcore/fmt/mod.rs:1076
> Oct 20 12:03:18 raspberrypi librespot[393]:    5:   0x8f1cd4 - std::io::Write::write_fmt::h1d02362bb0653180
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/io/mod.rs:1537
> Oct 20 12:03:18 raspberrypi librespot[393]:    6:   0x8fae68 - std::sys_common::backtrace::_print::hb6e6bf916a89634f
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/sys_common/backtrace.rs:62
> Oct 20 12:03:18 raspberrypi librespot[393]:    7:   0x8fae68 - std::sys_common::backtrace::print::h70b0c25b24fd7afb
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/sys_common/backtrace.rs:49
> Oct 20 12:03:18 raspberrypi librespot[393]:    8:   0x8fae68 - std::panicking::default_hook::{{closure}}::hc261daa3190f9e66
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/panicking.rs:198
> Oct 20 12:03:18 raspberrypi librespot[393]:    9:   0x8fab1c - std::panicking::default_hook::h5ce743d47ceb319b
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/panicking.rs:218
> Oct 20 12:03:18 raspberrypi librespot[393]:   10:   0x8fb520 - std::panicking::rust_panic_with_hook::hb609da6fd4918712
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/panicking.rs:486
> Oct 20 12:03:18 raspberrypi librespot[393]:   11:   0x8fb13c - rust_begin_unwind
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/panicking.rs:388
> Oct 20 12:03:18 raspberrypi librespot[393]:   12:   0x91d560 - core::panicking::panic_fmt::h48562d60f069801a
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libcore/panicking.rs:101
> Oct 20 12:03:18 raspberrypi librespot[393]:   13:   0x91d3b8 - core::option::expect_none_failed::hde7f522962a61940
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libcore/option.rs:1272
> Oct 20 12:03:18 raspberrypi librespot[393]:   14:   0x4e93a0 - librespot::main::h6424af6f8ac98335
> Oct 20 12:03:18 raspberrypi librespot[393]:   15:   0x4ec538 - std::rt::lang_start::{{closure}}::h20fd6404e8f8e33a
> Oct 20 12:03:18 raspberrypi librespot[393]:   16:   0x8fb8e4 - std::rt::lang_start_internal::{{closure}}::he71ca971345dcdbf
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/rt.rs:52
> Oct 20 12:03:18 raspberrypi librespot[393]:   17:   0x8fb8e4 - std::panicking::try::do_call::h3ebcd0c1fd722d55
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/panicking.rs:297
> Oct 20 12:03:18 raspberrypi librespot[393]:   18:   0x8fb8e4 - std::panicking::try::he6a0b75af361a6ec
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/panicking.rs:274
> Oct 20 12:03:18 raspberrypi librespot[393]:   19:   0x8fb8e4 - std::panic::catch_unwind::h2fbf955b2ece72b7
> Oct 20 12:03:18 raspberrypi librespot[393]:                        at src/libstd/panic.rs:394
> Oct 20 12:03:18 raspberrypi librespot[393]:   20:   0x8fb8e4 - std::rt::lang_start_internal::ha55d91cd07438bce
[... cut ...]
> Oct 20 12:06:55 raspberrypi librespot[779]: [2020-10-20T11:06:55Z INFO  librespot_playback::player] Loading <Heartbreaker - 1990 Remaster> with Spotify URI <spotify:track:6WE7jSshLCuVKoCmobVKVf>
> Oct 20 12:06:55 raspberrypi librespot[779]: [2020-10-20T11:06:55Z INFO  librespot_playback::player] <Heartbreaker - 1990 Remaster> (254133 ms) loaded
> lines 18-71/71 (END)

It stopped working after playing 'Rock and Roll - Remaster' and after playing some of 'Heartbreaker - 1990 Remaster', but nothing special is printed to the log after the disconnection.

Maybe something have changed in Spotify's side? Authentication tokens?

yoniw, 2020-10-20:
Update - now I see:

> sudo iptables -I OUTPUT -p tcp --dport 4070 -j ACCEPT
> Oct 20 12:35:08 raspberrypi librespot[779]: [2020-10-20T11:35:08Z INFO  librespot_playback::player] <D'yer Mak'er - Remaster> (262748 ms) loaded
> Oct 20 12:38:55 raspberrypi librespot[779]: [2020-10-20T11:38:55Z ERROR librespot_playback::player] Unable to load audio item.
> Oct 20 12:38:55 raspberrypi librespot[779]: [2020-10-20T11:38:55Z ERROR librespot_core::session] Os { code: 110, kind: TimedOut, message: "Connection timed out" }
>

JasonLG1979, 2020-10-20:
> Oct 20 12:03:16 raspberrypi librespot[393]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 19, kind: Other, message: "No such device" }', src/main.rs:422:22

This means that librespot is trying to access a sound card that doesn't exist.

> sudo iptables -I OUTPUT -p tcp --dport 4070 -j ACCEPT

I'm not sure how or why this is a part of your raspotify logs?

> Oct 20 12:38:55 raspberrypi librespot[779]: [2020-10-20T11:38:55Z ERROR librespot_core::session] Os { code: 110, kind: TimedOut, message: "Connection timed out" }

This is pretty self explanatory, the connection timed out. That could either be a problem with your connection or Spotify's problem. Librespot isn't the most network resilient program. It doesn't handle network hiccups very well sometimes. The 1st thing I would do is verify it's not a connection quality issue on your side. Check your speed and latency to see if it's reasonable and stable. If it's not talk to your ISP.

yoniw, 2020-10-20:
You're right.
I tried a hot spot from my mobile phone and everything works perfectly, so it must be my ISP.
Thanks once again.

JasonLG1979, 2020-10-20:
You're welcome. If you think you have the problem sorted you can close the issue.

closed 2020-10-20 by yoniw