dtcooper/raspotify #308: Feaure request: Nix packaging

moritzschaefer, 2020-10-18:
Since many people use NixOS on the Raspberry PI, it would awesome to have this package being packaged as a nix package. 
I'll try to make it work, however am happy happy for any support

JasonLG1979, 2020-10-19:
99% of raspotify is packaging. All it is is a compiled librespot binary, a systemd unit file and a config file packaged up in a deb. It's basically just librespot packaged for Raspberry Pi OS/Raspbian.

Making a librespot Nix package would be a different project IMHO. Nix-otify if you will.

JasonLG1979, 2020-10-19:
Do they not have a package conversion tool for Nix? Something like alien that converts deb's to rpm's but for Nix packages?

JasonLG1979, 2020-10-19:
Well, I guess there's already a [librespot Nix package](https://github.com/NixOS/nixpkgs/tree/master/pkgs/applications/audio/librespot). All you'd need to add would be the config and service file and you'd have the Nix equivalent of raspotify.

moritzschaefer, 2020-10-19:
I also found spotifyd, which comes packaged already (https://github.com/NixOS/nixpkgs/blob/master/pkgs/applications/audio/spotifyd/default.nix)
Spotifyd seems to basically do the same thing raspotify does, so its also a working solution for me. 

Thank you :)

closed 2020-10-19 by moritzschaefer