dtcooper/raspotify #310: Add a wiki

JasonLG1979, 2020-10-19:
It would be nice to add a wiki with basic troubleshooting steps, a deeper dive into audio configurations (systemwide and librespot) and whatnot.

JasonLG1979, 2020-10-30:
In the mean time this ```/etc/asound.conf``` has the instructions and means to configure the default ALSA audio device on a Raspberry Pi:

```
# /etc/asound.conf

###############################################################################

pcm.hqstereo20 {
    @args [
        SAMPLE_RATE FORMAT BUFFER_PERIODS
        BUFFER_PERIOD_TIME VOL_MIN_DB
        VOL_MAX_DB VOL_RESOLUTION VOL_NAME
    ]
    # Sampling rate in Hz:
    # 44100, 48000, 88200, 96000...
    # Defaults to 44.1 kHz (CD Quality).
    @args.SAMPLE_RATE {
        type integer
        default 44100
    }
    # Format:
    # S16_LE, S24_LE, S24_3LE, S32_LE...
    # Defaults to S16_LE (CD Quality).
    @args.FORMAT {
        type string
        default S16_LE
    }
    # Periods per buffer.
    @args.BUFFER_PERIODS {
        type integer
        default 4
    }
    # Period size in time.
    # Defaults to 125ms (0.125 sec).
    # BUFFER_PERIODS * BUFFER_PERIOD_TIME = buffer time/size
    @args.BUFFER_PERIOD_TIME {
        type integer
        default 125000
    }
    # Minimal dB value of the software volume control.
    @args.VOL_MIN_DB {
        type real
        default -51.0
    }
    # Maximal dB value of the software volume control.
    @args.VOL_MAX_DB {
        type real
        default 0.0
    }
    # How many steps between min and max volume.
    @args.VOL_RESOLUTION {
        type integer
        default 256
    }
    # The name of the software volume control.
    # If your card does not have hardware volume control
    # naming it PCM will cause most apps that use alsa volume
    # to use the software volume control.
    @args.VOL_NAME {
        type string
        default Softvol
    }
    type plug
    slave {
        channels 2
        rate $SAMPLE_RATE
        format $FORMAT
        pcm {
            type softvol
            min_dB $VOL_MIN_DB
[... 128 lines cut ...]
    # BUFFER_PERIOD_TIME: default: 125000
    # VOL_MIN_DB: default: -51.0
    # VOL_MAX_DB: default: 0.0
    # VOL_RESOLUTION: default: 256
    # VOL_NAME: default: Softvol

    # Example:
    # hifiberry dac+ zero on a pi zero.
    # slave.pcm "hqstereo20:FORMAT=S32_LE,BUFFER_PERIOD_TIME=250000,VOL_MIN_DB=-48.0,VOL_NAME=PCM"

    slave.pcm "hqstereo20"
}

###############################################################################

ctl.!default {
    type hw
    card {
        @func refer
        name defaults.ctl.card
    }
}
```

Following the steps in the above ``/etc/asound.conf``` (correctly) should solve and answer about 99.9% of all sound issues and questions for basic stereo sound cards.

JasonLG1979, 2020-10-30:
Explanation of my example hifiberry dac+ zero on a pi zero settings:
  
I use raspotify on a pi zero with a hifiberry dac+ zero hat.

The hifiberry dac+ zero hat supports up to S32_LE at 192k but lacks hardware volume control.

Librespot outputs S16_LE at 44.1k. Any volume reduction throws away bits, so for higher quality volume control it's better to use a hardware control. But since the hifiberry dac+ zero hat lacks hardware volume control we can fake it with the ```softvol``` ALSA plugin and up the bit depth to 32bit (not to be confused with upsampling) which allows us to lower the volume by 48dB before we have to start throwing bits away. (32 bit ALSA software volume control is actually more like 24bit as 8 bits are used for saturation). If we name the control ```PCM``` programs will use it as the default "hardware" volume control.

For the Pi Zero I've also increased the buffer to 1 sec as it seems to preform better.

dtcooper, 2020-12-30:
Done.

closed 2020-12-30 by dtcooper