dtcooper/raspotify #312: transfer audio from RASPOTIFY to build FM transmiter

szala1981, 2020-10-21:
I am using Raspotify on Raspberry Pi 3+

I have a built-in radio in the shower cabin and I came up with the idea to transmit sound from Raspotify through the built-in FM transmitter to radio FM, but I do not know how?

JasonLG1979, 2020-10-23:
What built in FM Transmitter? No version of Raspberry Pi has a builtin FM transmitter.

JasonLG1979, 2020-10-23:
You can send audio to the pipe backend but from there you're on your own.

RunningUtes, 2021-10-09:
I'm using PiFMAdv to transmit audio to any radio. It allows for a pipe function to the standard input. Below would send a starwars.wav to 87.6 MHz on a regular radio using just a 30cm wire to to GPIO4 pin.

```
pi_fm_adv --freq 87.6 --audio starwars.wav
```

This works when using sox for instance:

```
sox -t mp3 http://www.linuxvoice.com/episodes/lv_s02e01.mp3 -t wav -  | ./pi_fm_a --freq 87.6
```

What I'd like to do is send input from raspotify to PiFMAdv, but I can't get the piping to work.

Any tips?

JasonLG1979, 2021-11-17:
This very likely violates Spotify's ToS.

closed 2021-11-17 by JasonLG1979