dtcooper/raspotify #314: Raspotify - not connecting / not playing sound

farhandsome, 2020-10-23:
### Hi all, one issue I'm having is Raspotify not playing any sound when it does connect. Here is what I pulled from the log, it looks like this is where the issue arises:

Oct 23 12:01:00 raspberrypi librespot[3057]: [2020-10-23T17:01:00Z INFO  librespot_playback::player] Loading <Night Cruising> with Spotify URI <spotify:track:1McVhmCk8l07LIIdlIpqTF>
Oct 23 12:01:00 raspberrypi librespot[3057]: [2020-10-23T17:01:00Z INFO  librespot_playback::player] <Night Cruising> (188766 ms) loaded
Oct 23 12:01:00 raspberrypi librespot[3057]: ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
Oct 23 12:01:00 raspberrypi librespot[3057]: [2020-10-23T17:01:00Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Oct 23 12:01:00 raspberrypi librespot[3057]: [2020-10-23T17:01:00Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Oct 23 12:01:00 raspberrypi librespot[3057]: ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
Oct 23 12:01:00 raspberrypi librespot[3057]: [2020-10-23T17:01:00Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Oct 23 12:01:00 raspberrypi librespot[3057]: [2020-10-23T17:01:00Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Oct 23 12:01:00 raspberrypi librespot[3057]: ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
Oct 23 12:01:00 raspberrypi librespot[3057]: [2020-10-23T17:01:00Z ERROR librespot_playback::audio_backend::alsa] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENOENT: No such file or directory'
Oct 23 12:01:00 raspberrypi librespot[3057]: [2020-10-23T17:01:00Z ERROR librespot_playback::player] Could not start audio: Alsa error: PCM open failed
Oct 23 12:01:00 raspberrypi librespot[3057]: thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value', playback/src/audio_backend/alsa.rs:103:19

Anyone have any idea what this could be?

JasonLG1979, 2020-10-23:
A bunch of ALSA errors, that tells me that something is misconfigured or not configured at all.

1. What version Pi are you running on?

2. Will raspotify have to coexist with other audio players, will anything else be making noise? If so what?

3. What device are you trying to send audio to?

4. What is the output of ```aplay -l``` (lowercase L) ?

JasonLG1979, 2020-10-24:
@dtcooper This is were a wiki would come in handy.

farhandsome, 2020-10-24:
Hi @JasonLG1979,

Thanks for the reply. 

1. Running on an RPI4 Buster, fully updated.
2. Not at the moment, it is the only thing playing. 
3. Here’s where the issue probably lies. RPi’s audio config is abysmal. I have a USB speaker and trying to get all of the applications to default to that has been a pain since I’m also running the Pi thru a HDMI monitor. In the end, I created a .asoundrc file, changed the boot/config file (commented out hdmi force hot plug). This seemed to register my USB device finally. 
4. The only playback device (card 0) is now my USB device which is good. I can play music through Bluetooth and Airplay, just not Raspotify. 
5. I’ll check when I’m home, but I don’t have an .asound.conf file, just .asoundrc. Could that be it?

JasonLG1979, 2020-10-24:
> 5\. I’ll check when I’m home, but I don’t have an .asound.conf file, just .asoundrc. Could that be it?

Probably. ```asoundrc``` is basically a per user ```asound.conf```. If you want things to be systemwide regardless of user you should use ```asound.conf```. ```asoundrc``` would have no affect on system services like raspotify.

JasonLG1979, 2020-10-24:
> 1. Running on an RPI4 Buster, fully updated.

Headless or with a desktop?

> 2\. Not at the moment, it is the only thing playing.

Basically what I'm asking is if your Pi is going to be a dedicated Spotify connect device? If raspotify/librespot is the absolutely going to be the only thing outputting sound? If so we can optimize the configuration for raspotify/librespot and we don't have to care about sharing the audio output.

> 4\. The only playback device (card 0) is now my USB device which is good. I can play music through Bluetooth and Airplay, just not Raspotify.

Are you testing those services as a user or as system daemons?

farhandsome, 2020-10-24:
@JasonLG1979 

1. Desktop. Currently connected to an HDMI monitor. 

2. Ah, right. No, it won’t. I also have set up the Alexa AVS module. They won’t be running simultaneously, but Spotify isn’t the only sound emitting app on the Pi. 

3. So far, I’ve only tested them as the user pi. Pretty new to all this, could you provide some syntax on how to test it as the system daemons?

JasonLG1979, 2020-10-24:
> 2\. Ah, right. No, it won’t. I also have set up the Alexa AVS module. They won’t be running simultaneously, but Spotify isn’t the only sound emitting app on the Pi.

Will the output just be speech?

JasonLG1979, 2020-10-24:
> 3\. So far, I’ve only tested them as the user pi. Pretty new to all this, could you provide some syntax on how to test it as the system daemons?

That would explain why they work and raspotify doesn't. If you don't plan on using them there's no need to test them. But if you are it's going to be a bit of a dance to configure them so they gracefully yield to each other. Otherwise if you try to use more than one at a time you may get audio glitches caused by CPU spikes. I may not be a problem for a Pi 4 though, I don't know? The only Pi I have to play with at the moment is a Pi Zero but the Pi 4 is exponentially more powerful.

JasonLG1979, 2020-10-24:
> could you provide some syntax on how to test it as the system daemons

Untested but I imagine you could just test them as root.

```sudo su```

Then:

```<random service>```

farhandsome, 2020-10-24:
@JasonLG1979 I’ll give it a shot. May be worth a fresh install of Raspotify looks like

farhandsome, 2020-10-24:
@JasonLG1979 the Alexa output will just be speech, correct. I would love to have them both running at the same time since this project is for a Magic Mirror, but I’m not sure how feasible that is with all these alsa issues.

JasonLG1979, 2020-10-24:
> @JasonLG1979 the Alexa output will just be speech, correct. I would love to have them both running at the same time since this project is for a Magic Mirror, but I’m not sure how feasible that is with all these alsa issues.

It shouldn't be hard. If the output is just speech then it really doesn't matter much about the sound quality of the speech. Meaning  we can optimize the output for raspotify. Speech is pretty narrow bandwidth compared to music. Resampling it to what raspotify puts out (16bit 44.1 k) even if needed isn't going to affect it in an audible way.

If the Alexa bit has hooks we can even make it so raspotify is temporarily muted while Alexa speaks by giving raspotify a separate software volume control. Basically Alexa fires a hook script that mutes raspotify before it speaks that mutes raspotify and then when it's done speaking it fire another script that unmutes raspotify.

farhandsome, 2020-10-24:
@JasonLG1979 

That would be awesome. The Alexa module does have hooks, I know that because I’m trying to use an addressable LED strip to indicate its “listening” and “speaking” states. 

However, the Alexa files are all .cpp and I’m pretty new to this so I don’t know how to integrate python scripts into those files. 

I also use an audio-reactive application to sync up the LED strips when music is playing from the Pi. 

My ideal state is having the LED visualization while music is playing, but if the wake word model of Alexa is activated, it mutes Spotify and switches to Alexa AVS as well as switches the LED strip to the Alexa visual indicator.

JasonLG1979, 2020-10-24:
> That would be awesome. The Alexa module does have hooks, I know that because I’m trying to use an addressable LED strip to indicate its “listening” and “speaking” states.
>
> However, the Alexa files [...]

I was hoping to keep it simple with a bash script since all we'd be doing is muting and unmuting volume. But if you can hook a python script you can just use subprocess to fire off the commands.

> My ideal state is having the LED visualization while music is playing, but if the wake word model of Alexa is activated, it mutes Spotify and switches to Alexa AVS as well as switches the LED strip to the Alexa visual indicator.

That should be possible. I can help with the ALSA plumbing but I don't use alexa and I don't have any of the hardware, so the rest you may ultimately be on your own, sorry.

JasonLG1979, 2020-10-24:
I would start with a fresh install and then we can step though getting a good ALSA config. Then you can install alexa and raspotify.

JasonLG1979, 2020-10-25:
OK this will get you 2 separate inputs and volumes (one for spotify and one for alexa) and a master volume and default input.

Simplified (not showing all the resampling and software mixing bits) it looks like this:

[image]

Everything is resampled to 16 bit 44100 which is what raspotify/librespot outputs anyway so in the case of raspotify/librespot no resampling will actually be done. I don't know what your input looks like so we may have to tweak it if it's on the same device?

The device names are the same as the control names (spotify and alexa) so any config where you need to specify a device and/or a control they would be named the same. The master volume is named PCM because that's the convention for the name of the master volume control of a device. Most programs if you do not or can not specify a volume control will use the PCM control. Librespot does not keep track of outside changes to the volume if you use ALSA volume control so in your case until they fix that I would use the builtin software volume control and only use the "spotify" ALSA volume control for muting or dropping the volume when alexa speaks.

To get raspotify to use the spotify output add ```--device spotify``` to the ```OPTIONS``` section of ```/etc/default/raspotify```.

```
# /etc/asound.conf

# use aplay -l to determine the below.

# Change to the card number or name that you want to be the default control card.
# Default: 0
defaults.ctl.card 0

# Change to the card number or name that you want to be the default playback card.
# It should usually be the same as defaults.ctl.card.
# Default: 0
defaults.pcm.card 0

# Change to the device number that you want to be the default device on the default card.
# 0 or 1 is usually the correct device number.
# Default: 0
defaults.pcm.device 0

# Change to the subdevice number that you want to be the default subdevice on the default device.
# Should rarely need to be changed.
# Default: -1
defaults.pcm.subdevice -1

pcm.!default {
    type plug
    slave {
        channels 2
        rate 44100
        format S16_LE
        pcm {
            type softvol
            slave.pcm {
                type dmix
                ipc_key 1024
                ipc_perm 0660
                ipc_gid audio
                slowptr true
                hw_ptr_alignment roundup
                slave {
                    pcm {
                        type hw
                        card {
                            @func refer
                            name defaults.pcm.card
                        }
                        device {
                            @func refer
                            name defaults.pcm.device
                        }
                        subdevice {
                            @func refer
                            name defaults.pcm.subdevice
                        }
                    }
                    channels 2
                    period_size 0
                    buffer_size 0
                    buffer_time 0
                    period_time 250000
                    periods 4
                    rate 44100
                    format S16_LE
                }
                bindings {
                    0 0
                    1 1
                }
            }
            control {
                name PCM
                card {
                    @func refer
                    name defaults.ctl.card
                }
            }
        }
    }
}

pcm.spotify {
    type plug
    slave {
        channels 2
        rate 44100
        format S16_LE
        pcm {
            type softvol
            slave.pcm default
            control {
                name spotify
                card {
                    @func refer
                    name defaults.ctl.card
                }
            }
        }
    }
}

pcm.alexa {
    type plug
    slave {
        channels 2
        rate 44100
        format S16_LE
        pcm {
            type softvol
            slave.pcm default
            control {
                name alexa
                card {
                    @func refer
                    name defaults.ctl.card
                }
            }
        }
    }
}
```

JasonLG1979, 2020-10-25:
There will be about 1 sec of audio latency because of the buffer so anything used to control LED's may need to take that into account depending on how it's implemented to keep the audio and LED's in sync.

JasonLG1979, 2020-10-25:
You can test each output with:

```speaker-test -Dalexa -c2 -l1```

```speaker-test -Dspotify -c2 -l1```

It will be loud (the default volume I believe is 80% ?) so turn the volume down on whatever your sound card is connected to.

If whatever your sound card is connected to does not have a volume control you can use the ```-S``` arg to set the "scale" which is basically the volume. See ```speaker-test -h``` for details.

farhandsome, 2020-10-27:
@JasonLG1979 ,

This is awesome work. I've been out of town for the last few days, but I'll be back tomorrow so I'll definitely give this a test. Do I need to create this in a new file? (asound.conf). Because I have created an ".asoundrc" file that has a similar JSON-type look for the audio outputs. It only specifies system audio though, not specific application audios. I'm wondering if I should just delete that and use this?

farhandsome, 2020-10-27:
@JasonLG1979 ,

I also have a USB microphone (card 1, device 1) - is this something I should specify in asound.conf?

JasonLG1979, 2020-10-27:
> Do I need to create this in a new file? (asound.conf). Because I have created an ".asoundrc" file that has a similar JSON-type look for the audio outputs. It only specifies system audio though, not specific application audios. I'm wondering if I should just delete that and use this?

Yes create ```/etc/asound.conf``` and paste the contents. Delete ```.asoundrc```. ```asound.conf``` is system wide.

> I also have a USB microphone (card 1, device 1) - is this something I should specify in asound.conf?

If alexa is the only thing that will be using the mic and you can directly specify the mic somewhere in the alexa audio config you shouldn't have to worry about it. Just let alexa have exclusive access to the mic. Otherwise we can make the virtual ```alexa``` audio device asymmetrical and full duplex, meaning we point the playback to your sound card and the capture to your usb mic. Think of all this as a virtual mixing board with faders, batch bays, bus mixers, converters, inputs and outputs. If you can imagine a signal path we can make it happen.

JasonLG1979, 2020-10-27:
> It only specifies system audio though, not specific application audios

You don't have to route everything though program specific virtual devices. If not specified a program should use ```default```. ```default``` is specifically designed to do auto samplerate, format and channel count conversion and to software mix all streams. You could have as many applications playing audio simultaneously with any samplerate, format or channel count. The per app virtual devices just add a per app volume so you can adjust each apps volume independently at the system level. so you could use them to level match for example.

farhandsome, 2020-10-27:
@JasonLG1979 

                 
>bindings {
                  0 0
                  1 1
                }

Is this going to be different depending on which card/device the audio outputs are on?

JasonLG1979, 2020-10-27:
> Is this going to be different depending on which card/device the audio outputs are on?

That's not even needed most of the time. It's just saying channel 0 from dmix goes to channel 0 on the sound card and channel 1 goes to channel 1 on the sound card. Most stereo cards don't have a problem but some 5.1 and 7.1 drivers screw up channel mapping.

JasonLG1979, 2020-10-27:
You could switch it and the left channel would come out of the right speaker and the right channel would come out of the left speaker.

JasonLG1979, 2020-10-27:
If you have a 5.1 or 7.1 card we can spread out 2 channels to fill them. Basically left would get duplicated to all of the left channels and right to all of the right channels and then the center and sub channels would be mono (50% right + 50% left). there wouldn't be any surround sound effects but at least you'd be using all your channels. If you wanted to get fancy though you could probably measure the distance from each speaker to your listening position and calculate the needed delay to make sure everything hits you in phase at your listening position. But that's probably overkill.

farhandsome, 2020-10-28:
@JasonLG1979 

I tried both audio outputs and got this error message each time:

Playback device is spotify
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
ALSA lib pcm_params.c:2226:(snd1_pcm_hw_refine_slave) Slave PCM not usable
ALSA lib pcm_params.c:2226:(snd1_pcm_hw_refine_slave) Slave PCM not usable
Broken configuration for playback: no configurations available: No such file or directory
Setting of hwparams failed: No such file or directory

JasonLG1979, 2020-10-28:
Ok. You said your playback card is 0 so let's try to find the device number, and make sure your card supports 16bit 44.1k.

```aplay -Dhw:0,0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav```

Of if that doesn't work try this:

```aplay -Dhw:0,1 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav```

The output I want to see should look something like this:

```
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: [8000 192000]
PERIOD_TIME: (166 8192000]
PERIOD_SIZE: [32 65536]
PERIOD_BYTES: [256 524288]
PERIODS: [2 4096]
BUFFER_TIME: (333 16384000]
BUFFER_SIZE: [64 131072]
BUFFER_BYTES: [256 524288]
TICK_TIME: ALL
```

It will tell us what your sound card is capable of. It will also tell us the device number.

JasonLG1979, 2020-10-28:
It's most likely one of 3 things:

1. This section is not configured correctly:

```
# Change to the card number or name that you want to be the default control card.
# Default: 0
defaults.ctl.card 0

# Change to the card number or name that you want to be the default playback card.
# It should usually be the same as defaults.ctl.card.
# Default: 0
defaults.pcm.card 0

# Change to the device number that you want to be the default device on the default card.
# 0 or 1 is usually the correct device number.
# Default: 0
defaults.pcm.device 0

# Change to the subdevice number that you want to be the default subdevice on the default device.
# Should rarely need to be changed.
# Default: -1
defaults.pcm.subdevice -1
```

2. Your card doesn't support the ```S16_LE``` format.
3. Your card doesn't support a sampling rate of 44100.

Changing format and sampling rate isn't a huge deal though. But if you have to resample you'll want to install and enable a decent resampling lib. The default sounds like garbage and resampling is going to use a little more CPU.

farhandsome, 2020-10-28:
@JasonLG1979, looks like the rate is a static 48000 Hz? Maybe this can't be changed to 44100

JasonLG1979, 2020-10-28:
It must be one of those $5 USB DACs to only support ```S16_LE 48000```,lol!!!

farhandsome, 2020-10-28:
@JasonLG1979 I also get this error:

Oct 28 10:21:15 raspberrypi avahi-daemon[419]: Registering new address record for fe80::8893:73ba:7873:9636 on wlan0.*.
Oct 28 10:21:15 raspberrypi dhcpcd[615]: wlan0: deleting default route via fe80::366b:46ff:fe20:6d22
Oct 28 10:21:20 raspberrypi librespot[8542]: [2020-10-28T15:21:20Z WARN  libmdns::fsm] couldn't parse packet from V4(192.168.7.25:42096): ReservedBitsAreNonZ$
Oct 28 10:21:23 raspberrypi librespot[8542]: [2020-10-28T15:21:23Z WARN  libmdns::fsm] couldn't parse packet from V4(192.168.7.25:42096): ReservedBitsAreNonZ$
Oct 28 10:21:26 raspberrypi librespot[8542]: [2020-10-28T15:21:26Z WARN  libmdns::fsm] couldn't parse packet from V4(192.168.7.25:42096): ReservedBitsAreNonZ$
Oct 28 10:21:34 raspberrypi librespot[8542]: [2020-10-28T15:21:34Z WARN  libmdns::fsm] couldn't parse packet from V4(192.168.7.25:60758): ReservedBitsAreNonZ$
Oct 28 10:21:37 raspberrypi librespot[8542]: [2020-10-28T15:21:37Z WARN  libmdns::fsm] couldn't parse packet from V4(192.168.7.25:60758): ReservedBitsAreNonZ$
Oct 28 10:21:40 raspberrypi librespot[8542]: [2020-10-28T15:21:40Z WARN  libmdns::fsm] couldn't parse packet from V4(192.168.7.25:60758): ReservedBitsAreNonZ$
Oct 28 10:21:48 raspberrypi librespot[8542]: [2020-10-28T15:21:48Z WARN  libmdns::fsm] couldn't parse packet from V4(192.168.7.25:47961): ReservedBitsAreNonZ$
Oct 28 10:22:00 raspberrypi dhcpcd[615]: wlan0: Router Advertisement from fe80::366b:46ff:fe20:6d22
Oct 28 10:22:00 raspberrypi dhcpcd[615]: wlan0: adding address 2603:8080:c700:12b4:50d4:5fb9:857a:4354/64
Oct 28 10:22:00 raspberrypi dhcpcd[615]: wlan0: adding route to 2603:8080:c700:12b4::/64
Oct 28 10:22:00 raspberrypi dhcpcd[615]: wlan0: fe80::366b:46ff:fe20:6d22 is reachable again
Oct 28 10:22:00 raspberrypi dhcpcd[615]: wlan0: adding default route via fe80::366b:46ff:fe20:6d22
Oct 28 10:22:00 raspberrypi avahi-daemon[419]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::8893:73ba:7873:9636.
Oct 28 10:22:00 raspberrypi avahi-daemon[419]: Joining mDNS multicast group on interface wlan0.IPv6 with address 2603:8080:c700:12b4:50d4:5fb9:857a:4354.
Oct 28 10:22:00 raspberrypi avahi-daemon[419]: Registering new address record for 2603:8080:c700:12b4:50d4:5fb9:857a:4354 on wlan0.*.
Oct 28 10:22:00 raspberrypi avahi-daemon[419]: Withdrawing address record for fe80::8893:73ba:7873:9636 on wlan0.
Oct 28 10:25:05 raspberrypi dhcpcd[615]: wlan0: fe80::366b:46ff:fe20:6d22 is reachable again

And my Raspotify can't connect... this happens every so often. Is this something I need to add the username and password for in the config file?

farhandsome, 2020-10-28:
@JasonLG1979 but it looks like the asound.conf file is finally working!!! Appreciate it. My only issue now is the LAN thing.

JasonLG1979, 2020-10-28:
avahi-daemon and librespot aren't playing well together. If you don't need it for alexa then you can uninstall it.

farhandsome, 2020-10-28:
@JasonLG1979, I'll look into that if it keeps happening. For now, I'll close this. Thanks for all the help man, I really appreciate it!

closed 2020-10-28 by farhandsome

JasonLG1979, 2020-10-28:
No problem. Just let me know if you need anything else.

JasonLG1979, 2020-10-28:
For the LAN issue you might look [here](https://github.com/dtcooper/raspotify/issues/296) the OP seems to be getting similar errors.

farhandsome, 2020-10-28:
@JasonLG1979,

Hmm okay, I'll take a look at it.

One more thing - are there any hooks attached to Raspotify? Like I mentioned earlier, I'd like to add in a short python script to turn on some LED visualization lights when Raspotify starts to play. Similarly, turn them off when the player ends.

JasonLG1979, 2020-10-28:
Yep, Librespot has has hooks.
https://github.com/librespot-org/librespot/issues/185

farhandsome, 2020-10-30:
@JasonLG1979, got the visualization to work finally!

The issue is though - I'm using the "PLAYER_EVENT" environmental variable to trigger the lights on and off. However, when does "PLAYER_EVENT" = "stop"? If I pause the music, the variable becomes "paused". And even if I quit out of spotify on my phone, it doesn't change to "stop."

farhandsome, 2020-10-31:
@JasonLG1979 I'll play around with it... 

I'm still getting that error I mentioned earlier: 

Oct 28 10:21:15 raspberrypi avahi-daemon[419]: Registering new address record for fe80::8893:73ba:7873:9636 on wlan0..
Oct 28 10:21:15 raspberrypi dhcpcd[615]: wlan0: deleting default route via fe80::366b:46ff:fe20:6d22
Oct 28 10:21:20 raspberrypi librespot[8542]: [2020-10-28T15:21:20Z WARN libmdns::fsm] couldn't parse packet from V4(192.168.7.25:42096): ReservedBitsAreNonZ$
Oct 28 10:21:23 raspberrypi librespot[8542]: [2020-10-28T15:21:23Z WARN libmdns::fsm] couldn't parse packet from V4(192.168.7.25:42096): ReservedBitsAreNonZ$
Oct 28 10:21:26 raspberrypi librespot[8542]: [2020-10-28T15:21:26Z WARN libmdns::fsm] couldn't parse packet from V4(192.168.7.25:42096): ReservedBitsAreNonZ$
Oct 28 10:21:34 raspberrypi librespot[8542]: [2020-10-28T15:21:34Z WARN libmdns::fsm] couldn't parse packet from V4(192.168.7.25:60758): ReservedBitsAreNonZ$
Oct 28 10:21:37 raspberrypi librespot[8542]: [2020-10-28T15:21:37Z WARN libmdns::fsm] couldn't parse packet from V4(192.168.7.25:60758): ReservedBitsAreNonZ$
Oct 28 10:21:40 raspberrypi librespot[8542]: [2020-10-28T15:21:40Z WARN libmdns::fsm] couldn't parse packet from V4(192.168.7.25:60758): ReservedBitsAreNonZ$
Oct 28 10:21:48 raspberrypi librespot[8542]: [2020-10-28T15:21:48Z WARN libmdns::fsm] couldn't parse packet from V4(192.168.7.25:47961): ReservedBitsAreNonZ$
Oct 28 10:22:00 raspberrypi dhcpcd[615]: wlan0: Router Advertisement from fe80::366b:46ff:fe20:6d22
Oct 28 10:22:00 raspberrypi dhcpcd[615]: wlan0: adding address 2603:8080:c700:12b4:50d4:5fb9:857a:4354/64
Oct 28 10:22:00 raspberrypi dhcpcd[615]: wlan0: adding route to 2603:8080:c700:12b4::/64
Oct 28 10:22:00 raspberrypi dhcpcd[615]: wlan0: fe80::366b:46ff:fe20:6d22 is reachable again
Oct 28 10:22:00 raspberrypi dhcpcd[615]: wlan0: adding default route via fe80::366b:46ff:fe20:6d22
Oct 28 10:22:00 raspberrypi avahi-daemon[419]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::8893:73ba:7873:9636.
Oct 28 10:22:00 raspberrypi avahi-daemon[419]: Joining mDNS multicast group on interface wlan0.IPv6 with address 2603:8080:c700:12b4:50d4:5fb9:857a:4354.
Oct 28 10:22:00 raspberrypi avahi-daemon[419]: Registering new address record for 2603:8080:c700:12b4:50d4:5fb9:857a:4354 on wlan0..
Oct 28 10:22:00 raspberrypi avahi-daemon[419]: Withdrawing address record for fe80::8893:73ba:7873:9636 on wlan0.
Oct 28 10:25:05 raspberrypi dhcpcd[615]: wlan0: fe80::366b:46ff:fe20:6d22 is reachable again

What do I uninstall to get this to stop?

JasonLG1979, 2020-10-31:
You can try uninstalling ```avahi-daemon``` if anything you need depends on ```avahi-daemon``` add or change it to read ```disallow-other-stacks=no``` in ```avahi-daemon.conf```. I'm not sure if it's in there by default and you need to just flip it from ```yes``` to ```no``` or if you have to add the whole line? I generally ignore ```avahi-daemon``` on desktop systems and it's one of the 1st things I remove on headless systems. The only functionality you might miss if you even use it at all is "hostname.local" resolution vs having to use the pi's ip address to ssh into it. But for me it's not a big deal because I always set all my pi's to have static ip's anyway.

[The man page](https://linux.die.net/man/5/avahi-daemon.conf#:~:text=disallow%2Dother%2Dstacks%3D%20Takes,is%20responsible%20for%20mDNS%20traffic.) explains what the option does.

[The arch wiki](https://wiki.archlinux.org/index.php/avahi) is also an awesome source of info for all things Linux.

JasonLG1979, 2020-10-31:
I use raspotify on a Pi Zero + a HiFiBerry DAC+ Zero as basically a Spotify only appliance so I rip out just about everything not needed for that end. I also tweak what's left over.

[image]

farhandsome, 2020-10-31:
@JasonLG1979 , Got it! I changed the avahi-daemon and haven't had any issues. One other thing though... how do I go about adding a default mic to /etc/asound.conf? I think there's an error in that it's trying to use my USB speaker as the mic. When I go into alsamixer, it thinks the USB speaker has capture controls. See below (this device is my PLAYBACK device):

[image]

This device is my card 0, but my mic is card 1.

farhandsome, 2020-10-31:
Here's the major error I'm getting when I'm running some applications:

ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

farhandsome, 2020-10-31:
@JasonLG1979, do I need PulseAudio? Looks like it's not installed on my RPi.

JasonLG1979, 2020-10-31:
Yep no problem we can use a combination of the ```plug``` (horrible name it's a sample rate converter), ```asym``` and ```dsnoop``` (basically dmix in reverse) plugins to create a default virtual device that's a combination of the 2 cards.

1. Do you know what format (bit depth) and sampling rate Alexa expects or if it even matters?

2. Do you know if Alexa expects the volume control for the mic to have a specific name? Most of the time a mics volume control is called "Capture" or "Mic" or something like that. ```alsamixer -c1``` will open the mic's mixers so we can see what they're called if they even exist?

2. We'll need to check to see what the available formats and sample rates are for the mic similar to how we did with the sound card.

```arecord -d 1 -f cd -t wav -Dhw:1,0 test1.wav --dump-hw-params```
```arecord -d 1 -f cd -t wav -Dhw:1,1 test2.wav --dump-hw-params```

You can delete the wav's when you're done. We don't really care about them.

3. When you figure out your hw-params you're going to need to test out your mic to make sure you get a decent level if not we can use software volume to give the mic a boast.

> do I need PulseAudio? Looks like it's not installed on my RPi.

I wouldn't.

JasonLG1979, 2020-10-31:
> defaults.bluealsa.device

Are you trying to transmit or receive bluetooth audio?

farhandsome, 2020-10-31:
@JasonLG1979 

1) Just from browsing the internet, it looks like Alexa AVS may be expected 16-bit 16Khz sampling rate.

2) 
[image]

Not sure if Alexa is expecting a specific name, but it doesn't look like mine is called anything by default.

3. Maybe here's the issue? When I run the "arecord", it gives me this:
`arecord: main:828: audio open error: Device or resource busy`

4. Originally, I was using BT since I couldn't run Raspotify. But now, I can disable it if I need to. All I did was turn it off on the pi GUI. I didn't disable the BT or anything yet.

JasonLG1979, 2020-10-31:
If you're trying to receive bluetooth audio with ```bluealsa``` you can set that up by following [this howto I wrote](https://github.com/JasonLG1979/PiZero-Bluetooth-Audio-Receiver) it's written with a Pi Zero in mind but will work just fine on any pi provided it has bluetooth ofc. Stop at the audio set up though we'll need to do that custom. Otherwise if you don't need bluetooth uninstall ```bluealsa```.

> Not sure if Alexa is expecting a specific name, but it doesn't look like mine is called anything by default.

Hit F4 to switch to capture.

JasonLG1979, 2020-10-31:
> arecord: main:828: audio open error: Device or resource busy

Something is Bogarting the mic. Shutdown/disable anything that's using the mic.

JasonLG1979, 2020-10-31:
> Just from browsing the internet, it looks like Alexa AVS may be expected 16-bit 16Khz sampling rate

Cool. If your mic supports that we can just give that to Alexa without having to resample but downsampling is easier on the CPU anyway since throwing stuff away is much easier then making stuff up.

JasonLG1979, 2020-10-31:
I just dug a craptacular old webcam out of my junk drawer and this is what I got from ```arecord -d 1 -f cd -t wav -Dhw:1,0 test1.wav --dump-hw-params```

```
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: [8000 48000]
PERIOD_TIME: [1000 32768000]
PERIOD_SIZE: [32 262144]
PERIOD_BYTES: [64 524288]
PERIODS: [2 1024]
BUFFER_TIME: (1333 65536000]
BUFFER_SIZE: [64 524288]
BUFFER_BYTES: [128 1048576]
TICK_TIME: ALL
```

farhandsome, 2020-10-31:
@JasonLG1979,

Ok, killed everything on the mic. Here's mine:

```
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: [16000 48000]
PERIOD_TIME: [1000 16384000]
PERIOD_SIZE: [32 262144]
PERIOD_BYTES: [64 524288]
PERIODS: [2 1024]
BUFFER_TIME: (1333 32768000]
BUFFER_SIZE: [64 524288]
BUFFER_BYTES: [128 1048576]
TICK_TIME: ALL
```

Old reliable Logitech C270 lol.

JasonLG1979, 2020-10-31:
Looks like  it will do S16_LE 16k mono:

```arecord -c1 -d60 -fS16_LE -r16000 -twav -Dhw:1,0 test3.wav --dump-hw-params```

Try that yourself to see if your mic works with it. If it does record yourself from different places in the room and make sure the levels are decent. I had to give the webcam a 10dB boast to be able to hear myself from across my living room speaking in a normal voice.

JasonLG1979, 2020-10-31:
That will record for 60 sec.

JasonLG1979, 2020-10-31:
> Old reliable Logitech C270 lol.

Mine's some Chinese knock off,lol!!!

farhandsome, 2020-11-01:
@JasonLG1979 how can I see what apps are using the mic? I've turned off the audio reactive python code, but it still seems to be hogging the mic for some reason.

JasonLG1979, 2020-11-01:
With a single mic though you're going to have trouble with Alexa hearing you when you speak when there's music playing or any other background noise. You might want to look into an array mic.

> how can I see what apps are using the mic? I've turned off the audio reactive python code, but it still seems to be hogging the mic for some reason.

```sudo apt install lsof```
```sudo lsof /dev/snd/pcm*```

farhandsome, 2020-11-01:
@JasonLG1979, yeah I've realized that limitation.. I'll probably upgrade eventually to a good mic. Here's what lsof tells me:

```
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
COMMAND  PID USER   FD   TYPE DEVICE SIZE/OFF  NODE NAME
python3 1006 root  mem    CHR 116,56          11155 /dev/snd/pcmC1D0c
python3 1006 root    5u   CHR 116,56      0t0 11155 /dev/snd/pcmC1D0c
```
Not sure what these are...

[25 more replies]