dtcooper/raspotify #315: Remove volume control

hbsagen, 2020-10-26:
Some implementations of Spotify Connect does not have volume control. I would prefer a setting that enables this, now iOS is confused and sometimes maximize the volume on the phone while using Spotify. My other Spotify Connect players don’t induce this behavior in iOS.

JasonLG1979, 2020-10-28:
An update is in the works that updates raspotify to the latest librespot version. When that hits you'll be able to completely disable volume control if you like or currently you can make raspotify remember it's previous volume by enabling caching in ```/etc/default/raspotify``` with:

```CACHE_ARGS="--cache /var/cache/raspotify" --disable-audio-cache```

Then remove:
```--initial-volume=100``` from the ```VOLUME_ARGS``` section.

It still won't read the volume from your phone though it will only remember it's own volume from session to session.

JasonLG1979, 2020-10-28:
It would really be handy for librespot to query the volume of the controlling device before it connected now that you mention it. That sounds like a great [feature request](https://github.com/librespot-org/librespot/issues/new).

closed 2021-11-24 by github-actions[bot]