dtcooper/raspotify #316: Conflict between Raspotify and ir-ctl?

sspearing, 2020-10-28:
I am using Raspotify on a Pi Zero configured to play audio via PWM on BCM pins 13 and 14 and an appropriate device tree overlay.  I have found that while Raspotify is playing, I cannot reliably transmit IR codes using the ir-ctl command.  I have found the problem to be somewhat inconsistent however once the problem is present, it is repeatable.  With Raspotify playing, ir-ctl reports success but the IR signal is not successfully received by the external device.  If playback is stopped, the same ir-ctl command will succeed.  Commands are being issued at the command prompt.  IR commands issued from the original remote will always succeed.

It's possible this is a conflict between PWM audio playback and ir-ctl in which case it would be outside of Raspotify scope but if this is the case, any suggestions would be appreciated.

JasonLG1979, 2020-10-30:
Raspotify doesn't have anything directly to do the PWM. It just sends PCM audio to ALSA just like any other audio player.

sspearing, 2020-11-02:
After googling into a bunch of rabbit holes I came across a recent update to the gpio-ir-tx module to improve tx reliability. As one might expect IR transmission is made up of pulses and spaces.   If the space was longer than 10 us,the module was releasing its hardware locks and if the system was under high load, it would not get it back in time.  I was using 5.4.51. An apt upgrade -all brought me up to 5.4.71 which seems to have solved the issue. Sorry for the distraction.

closed 2020-11-02 by sspearing