dtcooper/raspotify #318: Connection error without HDMI

a1ex90, 2020-10-30:
I'm running raspotify on a fresh install raspberry pi b with raspbian lite. If I've connected a monitor during boot with HDMI everything is running fine and music playing through the audio jack. I can switch songs, volume, etc. without any issues seamlessly. But once I boot without any Monitor plugged nothing is working. I see my pi in Spotify and can connect, but no music is playing through the audio jack. Furthermore, and this is where things confuse me if I try to switch a song or change the volume it disconnects from Spotify. Which doesn't happen when I booted with HDMI connected. Oddly enough if I boot with HDMI and remove it after booting is completed everything just works fine.

JasonLG1979, 2020-10-31:
Raspberry Pi 4?

[This thread comment](https://www.raspberrypi.org/forums/viewtopic.php?t=263942#p1606122) may help? It appears as though the 2 HDMI ports and the analog output are the same card as far as the Pi 4 is concerned just separate devices with the analog output having several subdevices for whatever reason.

The comment mentions adding this to ```/boot/cmdline.txt``` to "split" the cards apart:

```
snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_compat_alsa=0
```

All that being said the analog audio output is complete and absolute garbage by any subjective and objective measure on all Raspberry Pi's. It's made for beeps and boops and that's about it. A $5 generic USB DAC would be a vast improvement and well worth the money and more than likely less hassle. For like $25 you can a HiFiBerry DAC+ Light DAC HAT that would be exponentially better than the analog output and guaranteed to work perfectly.

a1ex90, 2020-11-02:
I'm using a Raspberry Pi 1 Model B, so I guess it's another issue. I might give a USB DAC a try at some point and already have a HiFiBerry at my second raspofity setup on a Pi 3. Still would love to find the source of this issue with the audio jack.

JasonLG1979, 2020-11-02:
```snd_bcm2835``` is used by all raspberry pi versions. You can check to see if the module is loaded with:

```lsmod | grep snd_bcm2835```

a1ex90, 2020-11-03:
The module is loaded and the sound output of speaker-test per example works. It seems to be an issue with raspotify.

a1ex90, 2020-11-03:
Ok I found the solution: adding OPTIONS="--device hw:0" solved the issue. Somehow when no HDMI is plugged in aplay treats the audio-jack as device number 0, while in alsamixer and alsa.conf it is device number 1.

closed 2020-11-03 by a1ex90

JasonLG1979, 2020-11-04:
> Ok I found the solution: adding OPTIONS="--device hw:0" solved the issue. Somehow when no HDMI is plugged in aplay treats the audio-jack as device number 0, while in alsamixer and alsa.conf it is device number 1.

```hw:0``` is shorthand for ```hw:0,0,-1``` which is shorthand for card 0, device, 0, subdevice -1.
https://www.alsa-project.org/alsa-doc/alsa-lib/pcm_plugins.html#pcm_plugins_hw

By device do you mean card or do you actually mean device? Card, device and subdevice numbers don't change after boot. Cards associated to the card numbers can disappear if things are physically unplugged or their drivers are forcibly unloaded but they're not going to shuffle.

You want to just use ```aplay -l``` (lowercase L) and actually set the device number to the analog output. Because what you're doing now is basically saying "OK, I'll take the 1st device on card 0, I'm not sure what that is but I'll take it."