dtcooper/raspotify #319: Raspotify stopped working after a year / SYSTEM reinstall didn't help (Vero 4k+, OSMC)

Madvih, 2020-11-03:
Any chance to help me solve this issue?

I'm running raspotify on OSMC (VERO 4K+) with pulseaudio. 
It was working fine for a year. Last week or two, when I connect to raspotify, it connects OK.

service.status here

`raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-11-03 14:07:25 CET; 11s ago
  Process: 1623 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1625 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1627 (librespot)
   Memory: 6.7M
   CGroup: /system.slice/raspotify.service
           `-1627 /usr/bin/librespot --name raspotify (madosmc) --backend alsa --bitrate 320 --cache /var/cache/raspotify --enable-volume-normalisation --linear-volume --initial-volume=

Nov 03 14:07:25 madosmc systemd[1]: Starting Raspotify...
Nov 03 14:07:25 madosmc systemd[1]: Started Raspotify.
Nov 03 14:07:25 madosmc librespot[1627]: [2020-11-03T13:07:25Z INFO  librespot] librespot (raspotify v0.14.0) 6aac8a6 (2020-07-24). Built on 2020-07-26. Build ID: wPNyqn0m
Nov 03 14:07:25 madosmc librespot[1627]: [2020-11-03T13:07:25Z WARN  libmdns] Failed to register IPv6 receiver: Os { code: 19, kind: Other, message: "No such device" }
Nov 03 14:07:25 madosmc librespot[1627]: [2020-11-03T13:07:25Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-b-82wn.ap.spotify.com:4070"
Nov 03 14:07:26 madosmc librespot[1627]: [2020-11-03T13:07:26Z INFO  librespot_core::session] Authenticated as "11171064570" !
Nov 03 14:07:26 madosmc librespot[1627]: [2020-11-03T13:07:26Z INFO  librespot_playback::audio_backend::alsa] Using alsa sink
Nov 03 14:07:26 madosmc librespot[1627]: [2020-11-03T13:07:26Z INFO  librespot_core::session] Country: "CZ"
Nov 03 14:07:27 madosmc librespot[1627]: [2020-11-03T13:07:27Z INFO  librespot_playback::player] Loading <Loud> with Spotify URI <spotify:track:2RwGFM7YrPq53XYSEk9eoW>
Nov 03 14:07:28 madosmc librespot[1627]: [2020-11-03T13:07:28Z INFO  librespot_playback::player] <Loud> (232026 ms) loaded
`

BUT WHEN I START PLAYING SONG... no sound at all(tried all devices) and in service status is this...

[image]

service.status
`osmc@madosmc:~$ sudo systemctl status raspotify
* raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-11-03 14:07:25 CET; 3min 12s ago
  Process: 1623 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1625 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1627 (librespot)
   Memory: 18.4M
   CGroup: /system.slice/raspotify.service
           `-1627 /usr/bin/librespot --name raspotify (madosmc) --backend alsa --bitrate 320 --cache /var/cache/raspotify --enable-volume-normalisation --linear-volume --initial-volume=

Nov 03 14:10:35 madosmc librespot[1627]:   17: 0xab001d24 - std::thread::local::LocalKey<T>::with::h2213c9258ce9c67a
Nov 03 14:10:35 madosmc librespot[1627]:   18: 0xab008658 - futures::future::Future::wait::h648a3718536bde53
Nov 03 14:10:35 madosmc librespot[1627]:   19: 0xab001858 - std::sys_common::backtrace::__rust_begin_short_backtrace::h34ff2c5cefca6e3d
Nov 03 14:10:35 madosmc librespot[1627]:   20: 0xab01ded8 - core::ops::function::FnOnce::call_once{{vtable.shim}}::h581f1baab70f4017
Nov 03 14:10:35 madosmc librespot[1627]:   21: 0xab4348c8 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::hb42eae37b0651b8e
Nov 03 14:10:35 madosmc librespot[1627]:                        at /rustc/5c1f21c3b82297671ad3ae1e8c942d2ca92e84f2/src/liballoc/boxed.rs:1076
Nov 03 14:10:35 madosmc librespot[1627]:   22: 0xab4348c8 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::h51216037af35b9c0
Nov 03 14:10:35 madosmc librespot[1627]:                        at /rustc/5c1f21c3b82297671ad3ae1e8c942d2ca92e84f2/src/liballoc/boxed.rs:1076
Nov 03 14:10:35 madosmc librespot[1627]:   23: 0xab4348c8 - std::sys::unix::thread::Thread::new::thread_start::hcd96612ae799c481
Nov 03 14:10:35 madosmc librespot[1627]:                        at src/libstd/sys/unix/thread.rs:87
`

I already reinstalled everything, today even whole system, but no change at all...

Thanks for you time!

JasonLG1979, 2020-11-04:
`systemctl status raspotify` (you don't need sudo for that) is not very useful for that last bit as it cut off the 1st part of the error. I'm also kinda curious as to why there are emojis in there?

Anyway, use `journalctl -u raspotify -b` to get all of the logs for raspotify for the current boot.

Madvih, 2020-11-05:
Thank you for this cmd. I made a progress but still, can't run raspotify.

`osmc@madosmc:~$ journalctl -u raspotify -b
-- Logs begin at Wed 2020-11-04 21:06:08 CET, end at Thu 2020-11-05 02:30:11 CET. --
Nov 04 21:06:11 madosmc systemd[1]: Starting Raspotify...
Nov 04 21:06:11 madosmc systemd[1]: Started Raspotify.
Nov 04 21:06:11 madosmc librespot[415]: [2020-11-04T20:06:11Z INFO  librespot] librespot (raspotify v0.14.0) 6aac8a6 (2020-07-24). Built on 2020-07-26. Build ID
Nov 04 21:06:12 madosmc librespot[415]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 19, kind: Other, message: "No such de
Nov 04 21:06:12 madosmc librespot[415]: stack backtrace:
Nov 04 21:06:12 madosmc librespot[415]:    0: 0xab1ec584 - backtrace::backtrace::libunwind::trace::had1ce870e7642f03
Nov 04 21:06:12 madosmc librespot[415]:                        at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/libunwind.rs:86
Nov 04 21:06:12 madosmc librespot[415]:    1: 0xab1ec584 - backtrace::backtrace::trace_unsynchronized::he46c884781dd51a8
Nov 04 21:06:12 madosmc librespot[415]:                        at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/mod.rs:66
Nov 04 21:06:12 madosmc librespot[415]:    2: 0xab1ec584 - std::sys_common::backtrace::_print_fmt::h621902e5e498623c
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libstd/sys_common/backtrace.rs:78
Nov 04 21:06:12 madosmc librespot[415]:    3: 0xab1ec584 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h39c43d9eea8febb9
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libstd/sys_common/backtrace.rs:59
Nov 04 21:06:12 madosmc librespot[415]:    4: 0xab213e88 - core::fmt::write::h1faa4c62c7847860
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libcore/fmt/mod.rs:1076
Nov 04 21:06:12 madosmc librespot[415]:    5: 0xab1e5cd4 - std::io::Write::write_fmt::h1d02362bb0653180
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libstd/io/mod.rs:1537
Nov 04 21:06:12 madosmc librespot[415]:    6: 0xab1eee68 - std::sys_common::backtrace::_print::hb6e6bf916a89634f
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libstd/sys_common/backtrace.rs:62
Nov 04 21:06:12 madosmc librespot[415]:    7: 0xab1eee68 - std::sys_common::backtrace::print::h70b0c25b24fd7afb
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libstd/sys_common/backtrace.rs:49
Nov 04 21:06:12 madosmc librespot[415]:    8: 0xab1eee68 - std::panicking::default_hook::{{closure}}::hc261daa3190f9e66
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libstd/panicking.rs:198
Nov 04 21:06:12 madosmc librespot[415]:    9: 0xab1eeb1c - std::panicking::default_hook::h5ce743d47ceb319b
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libstd/panicking.rs:218
Nov 04 21:06:12 madosmc librespot[415]:   10: 0xab1ef520 - std::panicking::rust_panic_with_hook::hb609da6fd4918712
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libstd/panicking.rs:486
Nov 04 21:06:12 madosmc librespot[415]:   11: 0xab1ef13c - rust_begin_unwind
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libstd/panicking.rs:388
Nov 04 21:06:12 madosmc librespot[415]:   12: 0xab211560 - core::panicking::panic_fmt::h48562d60f069801a
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libcore/panicking.rs:101
Nov 04 21:06:12 madosmc librespot[415]:   13: 0xab2113b8 - core::option::expect_none_failed::hde7f522962a61940
Nov 04 21:06:12 madosmc librespot[415]:                        at src/libcore/option.rs:1272
Nov 04 21:06:12 madosmc librespot[415]:   14: 0xaaddd3a0 - librespot::main::h6424af6f8ac98335
`

After this I run `librespot --name "test" --bitrate 320` 
and this create spotify-connect and works fine. I just need to make raspotify working, don't understand why raspotify doesn't work and this "librespot --name "test" --bitrate 320" command working fine?!...

JasonLG1979, 2020-11-05:
Restart raspotify with `sudo systemctl restart raspotify` and then try to connect and see it works.

JasonLG1979, 2020-11-05:
If it doesn't show me the logs from the restart on only.

JasonLG1979, 2020-11-05:
And use triple triple backticks ``` for terminal output so it doesn't line wrap. It makes it easier to read.

Madvih, 2020-11-05:
```osmc@madosmc:~$ sudo systemctl restart raspotify
Warning: The unit file, source configuration file or drop-ins of raspotify.service changed on disk. Run 'systemctl daemon-reload' to reload units.
osmc@madosmc:~$ sudo systemctl daemon-reload
osmc@madosmc:~$ sudo systemctl restart raspotify```

after this it restarted with no output

here's more info about service
```osmc@madosmc:~$ systemctl show raspotify
Type=simple
Restart=always
NotifyAccess=none
RestartUSec=10s
TimeoutStartUSec=1min 30s
TimeoutStopUSec=1min 30s
RuntimeMaxUSec=infinity
WatchdogUSec=0
WatchdogTimestampMonotonic=0
RootDirectoryStartOnly=no
RemainAfterExit=no
GuessMainPID=yes
MainPID=13368
ControlPID=0
FileDescriptorStoreMax=0
NFileDescriptorStore=0
StatusErrno=0
Result=success
UID=110
GID=114
NRestarts=0
ExecMainStartTimestamp=Thu 2020-11-05 10:32:20 CET
ExecMainStartTimestampMonotonic=48374622067
ExecMainExitTimestampMonotonic=0
ExecMainPID=13368
ExecMainCode=0
ExecMainStatus=0
ExecStartPre={ path=/bin/mkdir ; argv[]=/bin/mkdir -m 0755 -p /var/cache/raspotify ; ignore_errors=no ; start_time=[Thu 2020-11-05 10:32:20 CET] ; stop_tim
ExecStartPre={ path=/bin/chown ; argv[]=/bin/chown raspotify:raspotify /var/cache/raspotify ; ignore_errors=no ; start_time=[Thu 2020-11-05 10:32:20 CET] ;
ExecStart={ path=/usr/bin/librespot ; argv[]=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS
Slice=system.slice
ControlGroup=/system.slice/raspotify.service
MemoryCurrent=10747904
CPUUsageNSec=[not set]
TasksCurrent=[not set]
IPIngressBytes=18446744073709551615
IPIngressPackets=18446744073709551615
IPEgressBytes=18446744073709551615
IPEgressPackets=18446744073709551615
Delegate=no
CPUAccounting=no
CPUWeight=[not set]
StartupCPUWeight=[not set]
CPUShares=[not set]
StartupCPUShares=[not set]
CPUQuotaPerSecUSec=infinity
IOAccounting=no
IOWeight=[not set]
StartupIOWeight=[not set]
BlockIOAccounting=no
BlockIOWeight=[not set]
StartupBlockIOWeight=[not set]
MemoryAccounting=yes
MemoryMin=0
MemoryLow=0
MemoryHigh=infinity
MemoryMax=infinity
MemorySwapMax=infinity
MemoryLimit=infinity
DevicePolicy=auto
TasksAccounting=yes
[... 133 lines cut ...]
DefaultDependencies=yes
OnFailureJobMode=replace
IgnoreOnIsolate=no
NeedDaemonReload=no
JobTimeoutUSec=infinity
JobRunningTimeoutUSec=infinity
JobTimeoutAction=none
ConditionResult=yes
AssertResult=yes
ConditionTimestamp=Thu 2020-11-05 10:32:20 CET
ConditionTimestampMonotonic=48374319175
AssertTimestamp=Thu 2020-11-05 10:32:20 CET
AssertTimestampMonotonic=48374319200
Transient=no
Perpetual=no
StartLimitIntervalUSec=10s
StartLimitBurst=5
StartLimitAction=none
FailureAction=none
FailureActionExitStatus=-1
SuccessAction=none
SuccessActionExitStatus=-1
InvocationID=16e042c2d0714117a53d4f9305ff01e7
CollectMode=inactive
``

JasonLG1979, 2020-11-05:
That's not really useful.

Did you install raspotify from this repo?

Have you modified the unit file? If so what have you changed?

JasonLG1979, 2020-11-05:
What does your `/etc/default/raspotify` look like because I noticed the args in your previous `systemctl status raspotify` contains:
``` --initial-volume=```

Was that just cut off when you pasted it?

Madvih, 2020-11-05:
yeah, I was trying everything since, reinstalled raspotify, (from this repo), played with settings, reinstalled everything actually.

my `/etc/default/raspotify` it's in default settings

```# /etc/default/raspotify -- Arguments/configuration for librespot

# Device name on Spotify Connect
#DEVICE_NAME="raspotify"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
#BITRATE="320"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info. Make sure whatever arguments you specify
# aren't already covered by other variables in this file. (See the daemon's
# config at `/lib/systemd/system/raspotify.service` for more technical details.)
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
#OPTIONS="--username <USERNAME> --password <PASSWORD>"

# Uncomment to use a cache for downloaded audio files. Cache is disabled by
# default. It's best to leave this as-is if you want to use it, since
# permissions are properly set on the directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

# By default, the volume normalization is enabled, add alternative volume
# arguments here if you'd like, but these should be fine.
#VOLUME_ARGS="--enable-volume-normalisation --linear-volume --initial-volume=100"

# Backend could be set to pipe here, but it's for very advanced use cases of
# librespot, so you shouldn't need to change this under normal circumstances.
#BACKEND_ARGS="--backend alsa"
```

Madvih, 2020-11-05:
but as I siad, after your help yesterday with `journalctl -u raspotify -b` I google some outputs and tried `librespot --name "test" --bitrate 320` what starts spotify-connect spot and works fine.

JasonLG1979, 2020-11-05:
All those errors from `journalctl -u raspotify -b` tell me that raspotify is trying to start at boot before the network is fully up. It's set to start after the `network.target` but that just means after all network related services have started it doesn't mean the network is actually completely up. I get the same thing but raspotify restarts itself after the timeout set in the service unit file.

Let's see if adding a delay to the start of the raspotify service helps.

1st show me your service file, post the result of this:
```
cat /etc/systemd/system/raspotify.service
```

Madvih, 2020-11-10:
Ok, thank you. I don't know if network is the issue here, system running still, and even if I restart the service it doesn't run propely. Maybe permission problem? Raspotify doesn't have right permission for librespot or some needed actions.
```
osmc@madosmc:~$ sudo service raspotify status
* raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-11-08 21:11:07 CET; 1 day 12h ago
  Process: 29436 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 29438 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 29440 (librespot)
   Memory: 444.0K
   CGroup: /system.slice/raspotify.service
           `-29440 /usr/bin/librespot --name raspotify (madosmc) --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100

Nov 09 15:54:56 madosmc librespot[29440]: [2020-11-09T14:54:56Z WARN  libmdns::fsm] error sending packet Os { code: 1, kind: PermissionDenied, message: "Operation not permitted" }
Nov 09 15:56:56 madosmc librespot[29440]: [2020-11-09T14:56:56Z WARN  libmdns::fsm] error sending packet Os { code: 1, kind: PermissionDenied, message: "Operation not permitted" }
Nov 09 16:10:56 madosmc librespot[29440]: [2020-11-09T15:10:56Z WARN  libmdns::fsm] error sending packet Os { code: 1, kind: PermissionDenied, message: "Operation not permitted" }
```

This `cat /etc/systemd/system/raspotify.service` doesn't exist on my system, but I found it under /etc/systemd/system/multi-user.target.wants/raspotify.service

```
osmc@madosmc:~$ cat /etc/systemd/system/multi-user.target.wants/raspotify.service
[Unit]
Description=Raspotify
After=network.target

[Service]
User=raspotify
Group=raspotify
Restart=always
RestartSec=10
PermissionsStartOnly=true
ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify ; /bin/chown raspotify:raspotify /var/cache/raspotify
Environment="DEVICE_NAME=raspotify (%H)"
Environment="BITRATE=320"
Environment="CACHE_ARGS=--disable-audio-cache"
Environment="VOLUME_ARGS=--enable-volume-normalisation --linear-volume --initial-volume=100"
Environment="BACKEND_ARGS=--backend alsa"
EnvironmentFile=-/etc/default/raspotify
ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS

[Install]
WantedBy=multi-user.target
```

Thank you sir!

JasonLG1979, 2020-11-10:
Those are libmdns errors. Are you behind a firewall?

rcmoliveira, 2020-11-12:
Try to change User and Group in raspotify.service to the user you ran librespot with.

Madvih, 2020-11-13:
> Those are libmdns errors. Are you behind a firewall?

No, I'm not. Librespot works fine, it's just raspotify what's not working. But thx for hint!

Madvih, 2020-11-13:
> Try to change User and Group in raspotify.service to the user you ran librespot with.

I already did that before I started this topic, sadly it didn't help.

JasonLG1979, 2020-11-13:
> No, I'm not. Librespot works fine, it's just raspotify what's not working. But thx for hint!

They are the same thing. Raspotify is basically just a systemd unit file that makes librespot run as a service.

Madvih, 2020-11-13:
> > No, I'm not. Librespot works fine, it's just raspotify what's not working. But thx for hint!
> 
> They are the same thing. Raspotify is basically just a systemd unit file that makes librespot run as a service.

Service is running fine, I can even connect to the speaker through spotify, but when I press play, it start's playing with no sound, service check status, output is with what I posted up. If I change volume or try something with spotify, it automatically drops the speaker from spotify. But as I said before, when I use librespot creation of the speaker in console, it works fine.

JasonLG1979, 2020-11-14:
Sounds like it might be a sound configuration issue then maybe?

Have you created a `~/.asoundrc`? ("`~/`" means the current user's home directory) 

I ask because `~/.asoundrc` is user specific. Any sound configuration in it would not affect raspotify. It's possible with a  `~/.asoundrc` for sound to work for just that particular user but not work for system level services because they don't see that config.

So it's possible for librespot to work just fine for you but for it to fail as a service.

JasonLG1979, 2020-11-14:
If it's an audio config issue you're going to need to talk to the OSMC folks. As anything I'd tell you is not going to be with OSMC in mind because I have no idea what their audio is configured like. I could very well help you get raspotify up and running and break everything else along the way.

closed 2021-11-24 by github-actions[bot]