dtcooper/raspotify #320: First track corrupted and very fast

ReaLArize, 2020-11-03:
Hey Guys,
i got a really strange problem here...
When the first track is playing, the sound is way too fast and cracks for a few minutes.
After some time, the song seems to manage itself to the right speed, plays at normal speed and everything is fine.
I would guess this is a buffer or samplerate issue but idk.

JasonLG1979, 2020-11-04:
Sounds like a severe case of buffer underruns. There are several different things that can cause that. 

1. Hardware details:
    Pi version?
    Sound card/DAC details?

2. Software details:
    Headless, desktop version?
    What other audio services/applications are you running?
    Are there any CPU hungry apps/services running?
 
3. Is your network otherwise stable and fast enough to stream audio?

4. What does your `/etc/default/raspotify` look like?

5. What does `systemctl status raspotify` say right after it happens?

ReaLArize, 2020-11-04:
## Harware Details
- Pi 4
- USB Audio Device for output
## Software
- Desktop version
- 15% CPU load max
- Airplay and Bluetooth audio services are running, but the error only happens with raspotify
```
/etc/asound.conf
pcm.!default {
    type plug
    slave.pcm {
        type multi
        slaves {
            a { channels 2 pcm "hw:2,0" }  # the real device
            b { channels 2 pcm "hw:Loopback,0" }  # the loopback driver
        }
        bindings {
            0 { slave a channel 0 }
            1 { slave a channel 1 }
            2 { slave b channel 0 }
            3 { slave b channel 1 }
        }
    }
    ttable [
        [ 1 0 1 0 ]   # left  -> a.left,  b.left
        [ 0 1 0 1 ]   # right -> a.right, b.right
    ]
}
```

- Network is stable and fast enough. It seems like the audio is playing way too fast, not too slow
```
DEVICE_NAME="Raspberry Pi"
BITRATE="320"
VOLUME_ARGS="--enable-volume-normalisation --linear-volume --initial-volume=100"
```
```
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-11-03 15:56:29 GMT; 17h ago
  Process: 609 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 615 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 616 (librespot)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/raspotify.service
           └─616 /usr/bin/librespot --name Raspberry Pi --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --linear-volume --initial-volume=100

Nov 03 16:40:08 raspberrypi librespot[616]: [2020-11-03T16:40:08Z WARN  librespot_core::session] Connection to server closed.
Nov 03 16:40:08 raspberrypi librespot[616]: [2020-11-03T16:40:08Z WARN  librespot] Spirc shut down unexpectedly
Nov 03 16:40:08 raspberrypi librespot[616]: [2020-11-03T16:40:08Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-b-bvn9.ap.spotify.com:4070"
Nov 03 16:40:09 raspberrypi librespot[616]: [2020-11-03T16:40:09Z INFO  librespot_core::session] Authenticated as "mlqyianzgt7ev9h4p04ke8riu" !
Nov 03 16:40:09 raspberrypi librespot[616]: [2020-11-03T16:40:09Z INFO  librespot_playback::audio_backend::alsa] Using alsa sink
Nov 03 16:40:09 raspberrypi librespot[616]: [2020-11-03T16:40:09Z INFO  librespot_core::session] Country: "DE"
Nov 04 09:05:46 raspberrypi librespot[616]: [2020-11-04T09:05:46Z INFO  librespot_playback::player] Loading <Thousand> with Spotify URI <spotify:track:7l246AlpGtFORaXX5vWbal>
Nov 04 09:05:46 raspberrypi librespot[616]: [2020-11-04T09:05:46Z INFO  librespot_playback::player] <Thousand> (183275 ms) loaded
```

JasonLG1979, 2020-11-04:
What's listening to the loopback?

ReaLArize, 2020-11-04:
My music to led sync service.
But that makes no difference whether it is on or off

JasonLG1979, 2020-11-04:
> My music to led sync service.
> But that makes no difference whether it is on or off

OK, I was just wondering if maybe you weren't trying to maybe mirror your audio via bluetooth or something like that?

If I had to guess it's a buffering issue inside of librespot. The current version of librespot packaged in raspotify is out of date. There's an update in the works that greatly improves buffer performance. You can grab a wip version of that coming update here [raspotify_0.15.0.librespot.v0.1.3_armhf.deb](https://github.com/dtcooper/raspotify/releases/download/0.15.0/raspotify_0.15.0.librespot.v0.1.3_armhf.deb) And see if that helps. Otherwise you can route `slave a` though a plug -> dmix to set your own buffer setting overrides.

On a totally unrelated note:

I've been trying to help @farhandsome with their audio setup and they were wanting to control LED's with music/audio. I don't have the hardware so I'm of very limited use when it comes to that. Maybe you could help them. Here's that thread if you're interested in lending a hand: https://github.com/dtcooper/raspotify/issues/314

JasonLG1979, 2020-11-04:
I run raspotify and bluealsa on a headless Pi Zero with a HiFiBerry DAC+ Zero hat as a Spotify Connect device/Bluetooth receiver. So plug > dmix and a singular audio config is a must. Otherwise you're at the mercy of the different services. Granted plug > dmix isn't the "audiophile" way since resampling can occur but for lossy streaming audio that bridge is already burned. For me a 1 sec buffer seems to work best and prevents underrunns, but being as though a Pi4 is exponentially more powerful than a Pi Zero you could get a away smaller. Not that it really matters. Latency caused by a bigger buffer is really only a big deal when recording.

ReaLArize, 2020-11-04:
The update to 0.15.0 fixed the problem. Many Thanks :D

I run raspotify and i think its bluealsa too.
The asound.conf should route the sound to my usb soundcard and to the loopback device.
Now i got a pyaudio python script, that records 512 frames of that loopback, does a numpy FFT and writes the array to the standard output.
There listens my .net core webservice with proccesses the array and sends instructions (via UDP) to an ESP-32 and this one is connected to the LEDs.

With this setup i get to send an instruction about every 16ms, which is about 60FPS.

I'm not really familiar with python or linux at all. Of course I would like to help @farhandsome but I have too little experience for that.

closed 2020-11-04 by ReaLArize

farhandsome, 2020-11-05:
Hi @ReaLArize, really cool stuff though! Could you explain a little bit more about your music to LED service?