dtcooper/raspotify #321: Discussion: Adding option to run as different user

joshuajaharwood, 2020-11-04:
## Problem
By default on Raspbian, running an audio application alongside Raspotify causes the applications to compete for device access.

## Elaboration
Since it's possible that a user will be running more than one audio application alongside raspotify, a user is reasonably likely to run Pulseaudio since it ships by default with Raspbian. Doing so is far less complex than having to configure an ALSA `dmix`.

Pulseaudio is run by default per-user (under `pi` by default), whereas raspotify runs as the `raspotify` user. Raspotify sends its audio to the default ALSA device, but this can't be redirected to Pulse by ALSA as it doesn't have access to Pulse. This will either cause raspotify to compete for audio devices, or crash (based on my testing).

## Solutions
My suggestion would be to allow the user to run raspotify as the default user, rather than its own. Doing so now requires some non-user-friendly config changes to the systemd service.

My understanding of ALSA/Pulse is rudimentary, so graciously looking for feedback on this idea.

@dtcooper 
@JasonLG1979

JasonLG1979, 2020-11-04:
> Since it's possible that a user will be running more than one audio application alongside raspotify, a user is reasonably likely to run Pulseaudio since it ships by default with Raspbian. Doing so is far less complex than having to configure an ALSA `dmix`.

AFAIK PulseAudio does not come installed by default in the desktop version of Raspberry Pi OS and I know for a fact that it's not installed by default on the Lite version. It's possible though something else you installed pulled PulseAudio in as a dependency. 

The default ALSA input is dmix'd. You don't have to roll your own dmixer if you don't want to. The most basic config just involves making sure you're telling ALSA which card you want to use as default.

> Pulseaudio is run by default per-user (under pi by default), whereas raspotify runs as the raspotify user. Raspotify sends its audio to the default ALSA device, but this can't be redirected to Pulse [...]

PulseAudio is not designed to run system wide (it bad enough that it runs as the privileged `pi` user), that's intentional, it's the same reason raspotify runs as the unprivileged `raspotify` system user. For security reasons. 

Don't get me wrong PulseAudio is great for doing what it's supposed to do most of the time but this is not what it was designed to do.

If you're having issues configuring ALSA properly follow the steps in this `/etc/asound.conf` to walk you through creating a basic working system wide config.

```
# /etc/asound.conf

###############################################################################

# Find your sound card:
# aplay -l

# Find your card's supported format(s) and sampling rate(s) While no other audio is being played:
# aplay -Dhw:<card #>,<device #> --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav
# For example:
# aplay -Dhw:0,0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav

# The output should look something like this if the card and device combination is correct:

# ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
# FORMAT:  S16_LE S24_LE S32_LE
# SUBFORMAT:  STD
# SAMPLE_BITS: [16 32]
# FRAME_BITS: [32 64]
# CHANNELS: 2
# RATE: [8000 192000]
# ...

# The above card supports formats S16_LE, S24_LE, and S32_LE, at up to a sampling rate of 192000.

# Uncomment and replace with the sampling rate of your choice.
# defaults.pcm.dmix.rate 48000

# Uncomment and replace with the format of your choice.
# defaults.pcm.dmix.format S16_LE 

# Change to the card number or name that you want to be the default control card.
# Default: 0
defaults.ctl.card 0

# Change to the card number or name that you want to be the default playback card.
# It should usually be the same as defaults.ctl.card.
# Default: 0
defaults.pcm.card 0

# Change to the device number that you want to be the default device on the default card.
# 0 or 1 is usually the correct device number.
# Default: 0
defaults.pcm.device 0

# Change to the subdevice number that you want to be the default subdevice on the default device.
# Should rarely need to be changed.
# Default: -1
defaults.pcm.subdevice -1

# To install high quality samplerate converters on Debian based systems:
# sudo apt install -y --no-install-recommends libasound2-plugins 

# To list available rate_converter's:
# echo "$(ls /usr/lib/*/alsa-lib | grep "libasound_module_rate_")" | sed -e "s/^libasound_module_rate_//" -e "s/.so$//"

# Uncomment and replace speexrate_medium with the rate_converter of your choice.
# defaults.pcm.rate_converter speexrate_medium

# Once everything is configure close and save this file and test with:
# speaker-test -c2 -l1

# While audio is playing you can verify your settings with:
# cat /proc/asound/card*/pcm*p/sub*/hw_params
```

I more complex `/etc/asound.conf` can be found [here](https://gist.github.com/JasonLG1979/f29cb0f14218a3b20db5d32c2cfae5ed).

JasonLG1979, 2021-11-17:
Raspotify is primarily intended to be used in a headless enviroment (Raspberry Pi OS lite) For desktop versions of Raspberry Pi OS [spotifyd](https://spotifyd.github.io/spotifyd/installation/Raspberry-Pi.html) offers similar functionality, much better PulseAudio compatibility and is a better choice.

closed 2021-11-17 by JasonLG1979