dtcooper/raspotify #323: No audio output via jack 3.5mm

dmdt, 2020-11-08:
I have installed raspotify. It start and I can connect to it by Spotify connect but I have no sound.
My setup:

- Raspberry Pi 4
- Display connected via hdmi1
- Speakers connected to the jack 3.5mm port

Other audio play normal. I tried to change '--device' in config file this do nothing. I tried to set 'hw:0,1', 'hw:0,2', 'hw:2,0', 'hw:2,1', 'hw:1,0' with no effect. (I restarted service after each change in config).

```systemctl status raspotify.service
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-11-08 12:44:42 +03; 2s ago
  Process: 4340 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 4342 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 4344 (librespot)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/raspotify.service
           └─4344 /usr/bin/librespot --name Media Center --backend alsa --bitrate 160 --disable-audio-cache --enable-volume-normalisation --linear-volume --in

Nov 08 12:44:42 RPi4 systemd[1]: Starting Raspotify...
Nov 08 12:44:42 RPi4 systemd[1]: Started Raspotify.
Nov 08 12:44:42 RPi4 librespot[4344]: [2020-11-08T09:44:42Z INFO  librespot] librespot (raspotify v0.14.0) 6aac8a6 (2020-07-24). Built on 2020-07-26. Build ID```

JasonLG1979, 2020-11-08:
Headless or desktop version?

dmdt, 2020-11-08:
> Headless or desktop version?

Desktop

JasonLG1979, 2020-11-08:
Do you have pulseaudio installed?

JasonLG1979, 2020-11-08:
Raspotify is not going to play well with pulseaudio if you have it installed for a few reasons.

1. The version of librespot that's compiled and packaged in raspotify isn't compiled with the pulseaudio backend.

2. Raspotify is a system level service. Pulseaudio is a user level service. Meaning even if librespot were to be compiled with the pulseaudio backend you'd have to run pulseaudio as a system level service which is a PITA and might break other stuff.

3. Pulseaudio takes an exclusively locks your sound card. When pulseaudio is running nothing else can make noise unless it goes though pulseaudio.

All that being said if you don't have pulseaudio installed you can use this `/etc/asound.conf` to walk you through creating a very basic working system wide config.  `/etc/asound.conf` does not exist by default you will need to work though the steps and then create the file and paste the contents.

```
# /etc/asound.conf

###############################################################################

# Find your sound card:
# aplay -l

# Find your card's supported format(s) and sampling rate(s) While no other audio is being played:
# aplay -Dhw:<card #>,<device #> --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav
# For example:
# aplay -Dhw:0,0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav

# The output should look something like this if the card and device combination is correct:

# ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
# FORMAT:  S16_LE S24_LE S32_LE
# SUBFORMAT:  STD
# SAMPLE_BITS: [16 32]
# FRAME_BITS: [32 64]
# CHANNELS: 2
# RATE: [8000 192000]
# ...

# The above card supports formats S16_LE, S24_LE, and S32_LE, at up to a sampling rate of 192000.

# Uncomment and replace with the sampling rate of your choice.
# defaults.pcm.dmix.rate 48000

# Uncomment and replace with the format of your choice.
# defaults.pcm.dmix.format S16_LE 

# Change to the card number or name that you want to be the default control card.
# Default: 0
defaults.ctl.card 0

# Change to the card number or name that you want to be the default playback card.
# It should usually be the same as defaults.ctl.card.
# Default: 0
defaults.pcm.card 0

# Change to the device number that you want to be the default device on the default card.
# 0 or 1 is usually the correct device number.
# Default: 0
defaults.pcm.device 0

# Change to the subdevice number that you want to be the default subdevice on the default device.
# Should rarely need to be changed.
# Default: -1
defaults.pcm.subdevice -1

# To install high quality samplerate converters on Debian based systems:
# sudo apt install -y --no-install-recommends libasound2-plugins 

# To list available rate_converter's:
# echo "$(ls /usr/lib/*/alsa-lib | grep "libasound_module_rate_")" | sed -e "s/^libasound_module_rate_//" -e "s/.so$//"

# Uncomment and replace speexrate_medium with the rate_converter of your choice.
# defaults.pcm.rate_converter speexrate_medium

# Once everything is configure close and save this file and test with:
# speaker-test -c2 -l1

# While audio is playing you can verify your settings with:
# cat /proc/asound/card*/pcm*p/sub*/hw_params
```

dmdt, 2020-11-08:
Thanks. After adding asound.conf and restarting service raspotify sound appeared. In config file I stated card number as 2, device number as 0.

JasonLG1979, 2020-11-08:
@dtcooper please enable a wiki. This is like the 10th audio config problem/question I've help resolve, lol!!! I could write a fairly comprehensive configuration guide.

dmdt, 2020-12-05:
With new system update raspberry is bundled with pulseaudio. So is there any solution to use raspotify on the latest OS versions?

JasonLG1979, 2020-12-05:
> With new system update raspberry is bundled with pulseaudio. So is there any solution to use raspotify on the latest OS versions?

Uninstall pulseaudio.

dtcooper, 2020-12-05:
@JasonLG1979 What about submitting a PR against the README?

JasonLG1979, 2020-12-05:
> What about submitting a PR against the README?

I can do that, but my concern is that the README is kinda a mess already. A well organized wiki is a better choice IMHO.

HansAnebreid, 2020-12-07:
Thanks! Solved my issue! //Hans

dmdt, 2020-12-09:
> > With new system update raspberry is bundled with pulseaudio. So is there any solution to use raspotify on the latest OS versions?
> 
> Uninstall pulseaudio.

Is there other solution? After I tried to delete or disable pulseaudio I got absence of sound in my system and should to reinstall it. Now on clear image I have sound from raspotify but when I mute system sound - raspotify still playing.

JasonLG1979, 2020-12-09:
> Is there other solution? After I tried to delete or disable pulseaudio I got absence of sound in my system and should to reinstall it. Now on clear image I have sound from raspotify but when I mute system sound - raspotify still playing.

As I've said pulseaudio isn't going to play well with raspotify. The Raspberry Pi OS devs really threw a wrench in the works with the addition of pulseaudio for no valid reason really except maybe easier Bluetooth audio (even then there's not really anything that bluealsa can't do that pulseaudio can when properly configured). IMHO raspotify is best suited for headless systems especially now. If you're not using your Pi as a desktop system you'd be better of going headless and installing the light version. It does not include pulseaudio.

Otherwise you might look at the ALSA/dmix without grabbing hardware device section in this wiki:

https://wiki.archlinux.org/index.php/PulseAudio

digixmax, 2021-03-21:
I currently set the OPTION parameter --device hw:X,0 in /etc/default/raspotify to hardcode output playing on hw device X, otherwise there is no sound from raspotify.  This behavior is the same on both Raspian Buster and Lubuntu 20.04.2.

Rather than hardcoding the selection, how can I set up raspotify so that it would play out on whichever device the system is currently selects to use as output device (HDMI, Built-in Headphone, USB Audio)?

closed 2021-11-17 by JasonLG1979

cfkmo, 2022-05-01:
Sorry for warming up a 2 year old thread, but I didn't want to create a new for that again. Is this issue realted to pulseaudio still exsiting or do I simply need to change the "audio backend" configuration to pulse (instead of alsa)?

JasonLG1979, 2022-05-01:
The wiki had about all you'd need to know.

cfkmo, 2022-05-02:
sorry. I have searched for the information in "Configuration" & "Troubleshooting", but didn't find that section.
Thanks for your reply ...