dtcooper/raspotify #324: Raspotify steal spotify focus after few seconds

Son-kun, 2020-11-09:
Hi, after start a any music from my playlist on PC/Android my Raspotify on RPi "steal" active device, but also stop play music anywhere. But if I start play music on Raspotify and then switch device to PC/Android then all works fine.

My journal when Raspotify try to "steal" focus:
```
Nov 09 23:26:47 raspberrypi librespot[32482]: [2020-11-09T22:26:47Z INFO  librespot_playback::player] Loading <Scar/let> with Spotify URI <spotify:track:4aQfnCt8IGSf2Hwh8GSBU2>
Nov 09 23:26:47 raspberrypi librespot[32482]: [2020-11-09T22:26:47Z INFO  librespot_connect::spirc] No more tracks left in queue
Nov 09 23:26:47 raspberrypi librespot[32482]: [2020-11-09T22:26:47Z INFO  librespot_playback::player] <Scar/let> (257786 ms) loaded
```

JasonLG1979, 2020-11-11:
The "active" device is defined by whatever device last set to be active by the user. If raspotify was the last active player and you open a desktop or mobile app and raspotify is the still the active player it didn't steal focus because it never lost it. All other Spotify apps (mobile, web, desktop) work the same way.

The other clients give up their active status when they quit. Raspotify being a service never quits. I think that's where confusion may lay.

JasonLG1979, 2020-11-11:
You can test this by playing something in any Spotify app and then opening another. You may or may not get asked if you want to continue to listen on the 1st instance or if you'd like to listen on the 2nd instance, but 1st instance will always attempt to keep it's status as the active player.

Son-kun, 2020-11-12:
Yes Jason, but that isn't my issue. I understand how spotify switch device in normal circumstances.
In my case, I have manualy selected device on PC (I can also switch few times device before start play and that don't change anythink). When start play music, after few seconds device changes to raspotify, and music is stoped.
What is important, it's no normal stop, from PC client I can't push play button (it's gray) I can only start new "playlist" to start play music and it's start on Raspotify.
Issue don't occur when I have other devices like PC, android, PS4 (without active raspotify).

JasonLG1979, 2020-11-12:
Does the playlist that you're trying to play contain local files? If so where are the local files located?

JasonLG1979, 2020-11-12:
It sounds like the PC (Windows?) app is bugging out and handing over playback to raspotify. Maybe it was the last active player previous to the desktop app? Raspotify/librespot does not in any way try to assert itself as the active player that's up to Spotify. It's passive. All it does it is advertise itself as a client. It doesn't initiate connections.

Son-kun, 2020-11-12:
Most of my playlist don't have any local files, also issue occure when I have PC (yes, Windows 10) off, and only Android app and Raspotify are online.

JasonLG1979, 2020-11-12:
> Most of my playlist don't have any local files, also issue occure when I have PC (yes, Windows 10) off, and only Android app and Raspotify are online.

I just checked and I can reproduce on Android and the Spotify Linux desktop app. Players will always prefer raspotify (librespot) when they are started. In my case it's as soon as the player is opens with delay. I don't remember this being the case in the recent past. The Android and desktop apps have been updated lately (raspotify has not) maybe they changed the behavior? Or maybe they changed the API? I go over to the [librespot repo](https://github.com/librespot-org/librespot) after all raspotify is basically just a systemd service file that starts librespot. Librespot is the actual binary.

closed 2021-11-24 by github-actions[bot]