dtcooper/raspotify #330: Stops playing music after changing track

johnyKast, 2020-11-30:
Hello everyone! It's my first attempt on using the Raspotify and despite that I am thrilled 😁 , I am currently facing the following issue:
I m connecting to the Raspotify,playing the song that i've chosen,however if I try to change the song then the music stops! In the spotify app, i can see that the song is still playing,however I can not listen to it.
In order to make it work again, i have to reset the Raspotify or reboot the system. Since I do that,I am able to connect again and listen to the chosen song.If i try to pause and play again or change song ,it does the same...
It's my 1st project on RPi3 mod B and unfortunately I don't have any experience on it.I would appreciate a detailed description of a solution for that issue...
Thanks in advance,

johnyKast, 2020-11-30:
Forgot to attach the below:

Dec 01 00:11:28 raspberrypi librespot[7920]: [2020-11-30T22:11:28Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-b-scr8.ap.spotify.com:4070"
Dec 01 00:11:50 raspberrypi librespot[7920]: [2020-11-30T22:11:50Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-b-l4jk.ap.spotify.com:4070"
Dec 01 00:11:51 raspberrypi librespot[7920]: [2020-11-30T22:11:51Z INFO  librespot_core::session] Authenticated as "11164106629" !
Dec 01 00:11:51 raspberrypi librespot[7920]: [2020-11-30T22:11:51Z INFO  librespot_playback::audio_backend::alsa] Using alsa sink
Dec 01 00:11:51 raspberrypi librespot[7920]: [2020-11-30T22:11:51Z INFO  librespot_core::session] Country: "BE"
Dec 01 00:11:51 raspberrypi librespot[7920]: [2020-11-30T22:11:51Z INFO  librespot_playback::player] Loading <Breathe> with Spotify URI <spotify:track:5oPUBUzrAnwvlmMzl1VW7Y>
Dec 01 00:11:52 raspberrypi librespot[7920]: [2020-11-30T22:11:52Z INFO  librespot_playback::player] <Breathe> (334800 ms) loaded
Dec 01 00:12:04 raspberrypi rtkit-daemon[894]: Supervising 0 threads of 0 processes of 1 users.
Dec 01 00:12:04 raspberrypi rtkit-daemon[894]: Supervising 0 threads of 0 processes of 1 users.
Dec 01 00:12:04 raspberrypi rtkit-daemon[894]: Supervising 0 threads of 0 processes of 1 users.
Dec 01 00:12:04 raspberrypi rtkit-daemon[894]: Supervising 0 threads of 0 processes of 1 users.
Dec 01 00:12:04 raspberrypi rtkit-daemon[894]: Supervising 0 threads of 0 processes of 1 users.
Dec 01 00:12:04 raspberrypi dbus[478]: [system] Rejected send message, 2 matched rules; type="method_call", sender=":1.59" (uid=113 pid=8050 comm="/usr/bin/pulseaudio --start --log-target=syslog ") interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" error name="(unset)" requested_reply="0" destination="org.bluez" (uid=0 pid=563 comm="/usr/lib/bluetooth/bluetoothd ")
Dec 01 00:12:04 raspberrypi dbus[478]: [system] Rejected send message, 2 matched rules; type="method_call", sender=":1.59" (uid=113 pid=8050 comm="/usr/bin/pulseaudio --start --log-target=syslog ") interface="org.bluez.Manager" member="GetProperties" error name="(unset)" requested_reply="0" destination="org.bluez" (uid=0 pid=563 comm="/usr/lib/bluetooth/bluetoothd ")
Dec 01 00:12:04 raspberrypi pulseaudio[8050]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Dec 01 00:12:04 raspberrypi pulseaudio[8050]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Dec 01 00:12:04 raspberrypi pulseaudio[8050]: [pulseaudio] bluez5-util.c: GetManagedObjects() failed: org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 2 matched rules; type="method_call", sender=":1.59" (uid=113 pid=8050 comm="/usr/bin/pulseaudio --start --log-target=syslog ") interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" error name="(unset)" requested_reply="0" destination="org.bluez" (uid=0 pid=563 comm="/usr/lib/bluetooth/bluetoothd ")
Dec 01 00:12:04 raspberrypi pulseaudio[8050]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 2 matched rules; type="method_call", sender=":1.59" (uid=113 pid=8050 comm="/usr/bin/pulseaudio --start --log-target=syslog ") interface="org.bluez.Manager" member="GetProperties" error name="(unset)" requested_reply="0" destination="org.bluez" (uid=0 pid=563 comm="/usr/lib/bluetooth/bluetoothd ")
Dec 01 00:12:18 raspberrypi kernel: w1_master_driver w1_bus_master1: Family 0 for 00.560000000000.06 is not registered.
Dec 01 00:12:59 raspberrypi librespot[7920]: [2020-11-30T22:12:59Z INFO  librespot_playback::player] Loading <Omen> with Spotify URI <spotify:track:144adL7pGHEWRwute2wxzZ>
Dec 01 00:12:59 raspberrypi librespot[7920]: [2020-11-30T22:12:59Z INFO  librespot_playback::player] <Omen> (216026 ms) loaded
Dec 01 00:13:09 raspberrypi kernel: w1_master_driver w1_bus_master1: Family 0 for 00.d60000000000.8a is not registered.
Dec 01 00:14:15 raspberrypi kernel: w1_master_driver w1_bus_master1: Family 0 for 00.360000000000.63 is not registered.

JasonLG1979, 2020-12-01:
There's a lot of chatter about pluseaudio and bluetooth in the logs. What's you're setup?

johnyKast, 2020-12-01:
> There's a lot of chatter about pluseaudio and bluetooth in the logs. What's you're setup?

Actually the issue was solved since i installed the  the latest .deb package .Or at least, that was the last part that i did..
Thanks anyway for the reply Jason!

SaulGoodman1337, 2020-12-03:
same issue here .... :(
i'm using pi-btaudio and have no errors in my logfiles at all... 
can anyone help?

SaulGoodman1337, 2020-12-03:
interesting. if i take the HDMI interface as device it works.
pi-btaudio is probably the guilty one here...

SaulGoodman1337, 2020-12-03:
Ahhhhh... after skipping the track i have to wait 1 minute or 2. then a error is shown in my logfile.

`thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: Error("snd_pcm_recover", Sys(ENODEV))', playback/src/audio_backend/alsa.rs:108:25
stack backtrace:
   0:   0x896584 - backtrace::backtrace::libunwind::trace::had1ce870e7642f03
                       at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/libunwind.rs:86
   1:   0x896584 - backtrace::backtrace::trace_unsynchronized::he46c884781dd51a8
                       at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.46/src/backtrace/mod.rs:66
   2:   0x896584 - std::sys_common::backtrace::_print_fmt::h621902e5e498623c
                       at src/libstd/sys_common/backtrace.rs:78
   3:   0x896584 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h39c43d9eea8febb9
                       at src/libstd/sys_common/backtrace.rs:59
   4:   0x8bde88 - core::fmt::write::h1faa4c62c7847860
                       at src/libcore/fmt/mod.rs:1076
   5:   0x88fcd4 - std::io::Write::write_fmt::h1d02362bb0653180
                       at src/libstd/io/mod.rs:1537
   6:   0x898e68 - std::sys_common::backtrace::_print::hb6e6bf916a89634f
                       at src/libstd/sys_common/backtrace.rs:62
   7:   0x898e68 - std::sys_common::backtrace::print::h70b0c25b24fd7afb
                       at src/libstd/sys_common/backtrace.rs:49
   8:   0x898e68 - std::panicking::default_hook::{{closure}}::hc261daa3190f9e66
                       at src/libstd/panicking.rs:198
   9:   0x898b1c - std::panicking::default_hook::h5ce743d47ceb319b
                       at src/libstd/panicking.rs:218
  10:   0x899520 - std::panicking::rust_panic_with_hook::hb609da6fd4918712
                       at src/libstd/panicking.rs:486
  11:   0x89913c - rust_begin_unwind
                       at src/libstd/panicking.rs:388
  12:   0x8bb560 - core::panicking::panic_fmt::h48562d60f069801a
                       at src/libcore/panicking.rs:101
  13:   0x8bb3b8 - core::option::expect_none_failed::hde7f522962a61940
                       at src/libcore/option.rs:1272
  14:   0x51fea4 - <librespot_playback::audio_backend::alsa::AlsaSink as librespot_playback::audio_backend::Sink>::write::hfd03664996f46394
  15:   0x5108e0 - <librespot_playback::player::PlayerInternal as futures::future::Future>::poll::h42ccf25cc1b87898
  16:   0x48b088 - futures::task_impl::std::set::h9bf896dc3d2ccfae
  17:   0x46cd24 - std::thread::local::LocalKey<T>::with::h2213c9258ce9c67a
  18:   0x473658 - futures::future::Future::wait::h648a3718536bde53
  19:   0x46c858 - std::sys_common::backtrace::__rust_begin_short_backtrace::h34ff2c5cefca6e3d
  20:   0x488ed8 - core::ops::function::FnOnce::call_once{{vtable.shim}}::h581f1baab70f4017
  21:   0x89f8c8 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::hb42eae37b0651b8e
                       at /rustc/5c1f21c3b82297671ad3ae1e8c942d2ca92e84f2/src/liballoc/boxed.rs:1076
  22:   0x89f8c8 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::h51216037af35b9c0
                       at /rustc/5c1f21c3b82297671ad3ae1e8c942d2ca92e84f2/src/liballoc/boxed.rs:1076
  23:   0x89f8c8 - std::sys::unix::thread::Thread::new::thread_start::hcd96612ae799c481
                       at src/libstd/sys/unix/thread.rs:87
[2020-12-03T13:24:17Z DEBUG librespot_playback::player] drop PlayerInternal[0]
[2020-12-03T13:24:25Z WARN  libmdns::fsm] couldn't parse packet from V4(192.168.178.70:5353): type 47 is invalid`

the strange thing is that it works the first time... only when i change the track i have no sound anymore and this error appears on the logfile...

SaulGoodman1337, 2020-12-03:
i have found a similar problem here (https://github.com/dtcooper/raspotify/issues/227)... 

then i built bluez-alsa myself and .... it works !

The original Raspian version seems to cause problems. should this be included in the readme?

stefano-lupo, 2021-03-07:
I was having the exact same issue here where things would work just fine if I let a playlist play. However, when I tried to go to the next track, the audio would stop coming through, even though the spotify track appeared to continue playing.

[Here](https://gist.github.com/stefano-lupo/fd3bc31e57c572b8bb0020c0f106414a) is a gist of my sys log with some annotations I added explaining what I did to trigger each line. Theres not much to go by here and I honestly haven't looked into any of this at all. This required a hard reset of raspotify with via systemctl.

I noticed that a lot of the people talking about this issue were using some sort of bluetooth setup. I had never used bluetooth with my pi from what I remember. However this pi is pretty old and I honestly can't remember what audio configuation I had done to it. In any case, I was trying to use the 3.5mm jack on the PI and wasn't trying to use bluetooth so I wasn't sure if my problem was the same as these ones.

In any case, nuking pulse audio from my pi seemed to do the trick and now I can fully control spotify as expected!
```
 sudo apt --purge remove pulseaudio*
```

closed 2021-11-17 by JasonLG1979