dtcooper/raspotify #331: Cant get output on USB device?

nebdy, 2020-11-30:
Getting confused cant get the sound out of my USB DAC

When i list the hardware devices I get

pi@music:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 3/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
**card 2: CODEC [USB Audio CODEC], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0**

And i want it out on the 

**card 2: CODEC [USB Audio CODEC], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0**

So i have edited /etc/default/raspotify

to include the line

OPTIONS="--device hw:2,0"

But nope it does not come out on the USB

Any thoughts please

Ta

JasonLG1979, 2020-12-01:
If I had to guess, your USB DAC doesn't support a sampling rate of 44.1 kHz. If that's the case you'll have to resample to something your DAC does support, probably 48 kHz.

 `aplay -Dhw:2,0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav` should tell us what your DAC supports.

nebdy, 2020-12-01:
Ah cool I think you are onto something here

Playing WAVE '/usr/share/sounds/alsa/Front_Right.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
HW Params of device "hw:2,0":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S8 U8 S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: [8 16]
FRAME_BITS: [8 32]
CHANNELS: [1 2]
RATE: [32000 48000]
PERIOD_TIME: [1000 16384000]
PERIOD_SIZE: [32 524288]
PERIOD_BYTES: [64 524288]
PERIODS: [2 1024]
BUFFER_TIME: (1333 32768000]
BUFFER_SIZE: [64 1048576]
BUFFER_BYTES: [64 1048576]
TICK_TIME: ALL
--------------------

I get 48kHz not 44.1 kHz

Now how do I change the sample rate to? 48kHz?

JasonLG1979, 2020-12-01:
> Now how do I change the sample rate to? 48kHz?

This assumes you want to use `CODEC` as your default sound card.

1. install higher quality resampling libs, because if you must resample at least make it sound as good as you can:
`sudo apt install -y --no-install-recommends libasound2-plugins`

2. Copy and paste this into `/etc/asound.conf` (the file does not exist by default):
```
# /etc/asound.conf

pcm.!default {
    type plug
    # lavrate_higher and speexrate_medium offer the best bang for your buck,
    # CPU usage vs quality.
    rate_converter lavrate_higher
    slave.pcm {
        type dmix
        ipc_perm 0660
        ipc_key 1024
        ipc_gid audio
        slave {
            pcm {
                type hw
                card CODEC
            }
            channels 2
            rate 48000
            format S16_LE
            periods 0
            period_time 0
            buffer_time 0
            period_size 6000
            buffer_size 24000
        }
        bindings {
            0 0
            1 1
        }
    }
}

ctl.!default {
    type hw
    card CODEC
}
```

3. Remove `--device hw:2,0` from your raspotify config and restart raspotify.

closed 2021-11-17 by JasonLG1979