dtcooper/raspotify #332: Raspotify device shown on iOS, but not on macOS …

fooness, 2020-12-03:
Hello everyone. I installed `raspotify` two days ago on my Raspberry Pi 3 running Raspbian buster.

`Dec 03 09:50:35 raspberrypi librespot[2898]: [2020-12-03T08:50:35Z INFO  librespot] librespot (raspotify v0.14.0) 6aac8a6 (2020-07-24). Built on 2020-07-26. Build ID: wPNyqn0m`

For about an hour, I was trying to figure out why my Pi is no longer shown in the devices menu in Spotify on macOS. In the end I figured out, that when starting the playback on my Pi via Spotify on iOS, then afterwards the Pi is also shown in and can be controlled from macOS.

I guess this is very weird behavior? macOS is in the same WiFi as the Pi, there’s no firewall issues, … why do I need to initially “connect” to the device from iOS?

grescher, 2020-12-14:
Hello @fooness 
I encountered the same issue. Not only _Spotify@macOS_ but the _spotify-qt@Pi_ didn't find _raspotify@Pi_ until I started playing from my mobile. I was advised to write my login and password in the raspotify [configuration file](https://github.com/dtcooper/raspotify/blob/master/README.md#configuration). And it worked for me.

fooness, 2020-12-14:
> I was advised to write my login and password in the raspotify [configuration file](https://github.com/dtcooper/raspotify/blob/master/README.md#configuration). And it worked for me.

Wouldn’t my Spotify Connect be exposed to the web, then?

RudiKleinert, 2021-01-13:
> For about an hour, I was trying to figure out why my Pi is no longer shown in the devices menu in Spotify on macOS. In the end I figured out, that when starting the playback on my Pi via Spotify on iOS, then afterwards the Pi is also shown in and can be controlled from macOS.

Hi all, I have discovered the same behavior but with Windows instead of MacOS and Android instead of iOS. So this seems to be a general problem?

I used raspotify on my Pi3 on and off throughout a few years now and it kinda worked some of the time and sometimes it didn't really. At least stability always seemed to be an issue and playback would suddenly stop or the device wouldn't respond to the remote by my smartphone anymore. Anyhow since I installed the latest version again a few days ago it seems to run pretty smoothly and there were no hickups in quite some hours of playback since then. But the issue described above seems to be the regular case. So the raspotify-client isn't visible in the spotify web interface until I select it via the android app (where it is visible as soon as the pi boots up).
I have also put my accound log-in data into the config but that didn't change anything for me. BTW: Is it correct to put the login into the config file in clear text? Not that anyone could easily access my Pi, but still wondering...

Elalith, 2021-01-13:
I've seen the same issue here. The Windows client has issues finding the Pi, but my Android phone and iPad always find it instantly.  But... after I've connected via my phone or tablet, The Windows client instantly shows a connection.

BTW.. i've never entered my credentials in the property file and it works fine. Funny detail. When I'm nowhere near my Pi or home, my phone still shows the Pi and can even cast music to it.

JasonLG1979, 2021-11-20:
The new release is live. Please update and see if that solves your issue.

JasonLG1979, 2021-11-24:
This is an upstream issue. If after updating to the latest version of Raspotify the issue persists please file a bug with `librespot`.

closed 2021-11-24 by JasonLG1979