dtcooper/raspotify #335: automatically clear cache

ondrejtichacek, 2020-12-19:
Once every while raspotify fills up the cache folder `/var/cache/raspotify/` and starts behaving strangely (obviously, since also the system partition is full at that point in my case).

I set up a cron job to delete the cache every month, but this is not an ideal solution.

The best would be to automatically delete old cached files if the folder reaches preconfigured size. Or something in that fashion.

Knowing how much job the ideal solution would be, raspotify should at least provide the automatic cache deletion service out of the box. It took me a while to figure out what is going on.

Thanks! :-)

shawaj, 2021-01-04:
I added a similar request at librespot https://github.com/librespot-org/librespot/issues/558

shawaj, 2021-01-04:
Duplicate of https://github.com/dtcooper/raspotify/issues/144

JasonLG1979, 2021-11-17:
Already implemented up stream.

closed 2021-11-17 by JasonLG1979