dtcooper/raspotify #337: Cannot play through headphone jack

tomdlmurphy, 2020-12-20:
Hi!

I'm not able to force the sound to come out of the headphone jack; it just stays on the HDMI output. I've using `raspi-config` to set the sound to headphone jack. I've also tried using just the desktop sound icon. I know it's something wrong with raspotify because I can get sound through the headphone jack with other applications (youtube on chromium). I've taken advice from [this post](https://github.com/dtcooper/raspotify/issues/22), but when when I add

`OPTIONS="--onstart 'sudo amixer cset numid=3 1' --onstop 'sudo amixer cset numid=3 0'"`

to the conf file in /etc/default the service no longer works. Here is the output from sudo service raspotify status:

`● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2020-12-20 09:28:05 EST; 8s ago
  Process: 5554 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 5555 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 5556 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS (code=exited, status=1
 Main PID: 5556 (code=exited, status=1/FAILURE)

Dec 20 09:28:05 raspberrypi librespot[5556]:                         Play all tracks at the same volume
Dec 20 09:28:05 raspberrypi librespot[5556]:         --normalisation-pregain PREGAIN
Dec 20 09:28:05 raspberrypi librespot[5556]:                         Pregain (dB) applied by volume normalisation
Dec 20 09:28:05 raspberrypi librespot[5556]:         --linear-volume
Dec 20 09:28:05 raspberrypi librespot[5556]:                         increase volume linear instead of logarithmic.
Dec 20 09:28:05 raspberrypi librespot[5556]:         --autoplay      autoplay similar songs when your music ends.
Dec 20 09:28:05 raspberrypi librespot[5556]:         --disable-gapless
Dec 20 09:28:05 raspberrypi librespot[5556]:                         disable gapless playback.
Dec 20 09:28:05 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Dec 20 09:28:05 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
`

Here is the content of my `raspotify` file in /etc/default:

`
# /etc/default/raspotify -- Arguments/configuration for librespot

# Device name on Spotify Connect
#DEVICE_NAME="raspotify"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
#BITRATE="160"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info. Make sure whatever arguments you specify
# aren't already covered by other variables in this file. (See the daemon's
# config at `/lib/systemd/system/raspotify.service` for more technical details.)
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
OPTIONS="--device hw:1,1"

OPTIONS="--username tomdlmurphy@gmail.com --password FUCKINGsaudisTRIEDtoHACKme9348"

OPTIONS="--onstart 'sudo amixer cset numid=3 1' --onstop 'sudo amixer cset numid=3 0'"

# Uncomment to use a cache for downloaded audio files. Cache is disabled by
# default. It's best to leave this as-is if you want to use it, since
# permissions are properly set on the directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

# By default, the volume normalization is enabled, add alternative volume
# arguments here if you'd like, but these should be fine.
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
OPTIONS="--device hw:1,1"

OPTIONS="--username tomdlmurphy@gmail.com --password FUCKINGsaudisTRIEDtoHACKme9348"

OPTIONS="--onstart 'sudo amixer cset numid=3 1' --onstop 'sudo amixer cset numid=3 0'"

# Uncomment to use a cache for downloaded audio files. Cache is disabled by
# default. It's best to leave this as-is if you want to use it, since
# permissions are properly set on the directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

# By default, the volume normalization is enabled, add alternative volume
# arguments here if you'd like, but these should be fine.
#VOLUME_ARGS="--enable-volume-normalisation --linear-volume --initial-volume=100"

# Backend could be set to pipe here, but it's for very advanced use cases of
# librespot, so you shouldn't need to change this under normal circumstances.
#BACKEND_ARGS="--backend alsa"`

Any help would be much appreciated! Super excited to use this! Thanks @dtcooper

tomdlmurphy, 2020-12-20:
Just figure it out! I was redefining my options variable so it wasn't taking the device argument. Haha embarassing. For those not clear the line used to get audio from the headphone jack would be:

`OPTIONS="--username myuser --password mypass --device hw:1,0"`

The number after `hw` may vary. You need to run `aplay -l` in the terminal to see what number correspond to your headphone jack. The first number represents the card number, the second number represents the subdevice number.

dtcooper, 2020-12-30:
Awesome! Closing, however please feel free to add the hot tip in the README or wiki. :)

closed 2020-12-30 by dtcooper