dtcooper/raspotify #339: ARM 64 support

dusatvoj, 2020-12-21:
Hi, there are a lot of ARM64 OSes for RPi. RaspberryPi OS will be 64bit in a while too.
Is there a possibility to build ARM64bit images? :thinking:
Thank you

thorstenhaga-tradedoubler, 2020-12-21:
[Spocon](https://github.com/spocon/spocon) has arm64 already and others too

 spocon_1.5.2_amd64.deb (11.6 MiB)
spocon_1.5.2_arm64.deb (11.6 MiB)
spocon_1.5.2_armhf.deb (11.6 MiB)
spocon_1.5.2_i386.deb (11.6 MiB)
spocon_1.5.2_ppc64el.deb (11.6 MiB)
spocon_1.5.2_s390x.deb (11.6 MiB

spocon, 2020-12-21:
https://github.com/spocon/spocon . Basically the same but more package support :

 spocon_1.5.2_amd64.deb (11.6 MiB)
spocon_1.5.2_arm64.deb (11.6 MiB)
spocon_1.5.2_armhf.deb (11.6 MiB)
spocon_1.5.2_i386.deb (11.6 MiB)
spocon_1.5.2_ppc64el.deb (11.6 MiB)
spocon_1.5.2_s390x.deb (11.6 MiB)

dtcooper, 2020-12-30:
Not for this latest release, but yes. I'll put that on the roadmap. 🥇

insign, 2021-02-01:
Just here to cry.

pghpete, 2021-02-20:
It is a bit hackish, but this may be worth a try in the mean time... [](https://thewayeye.net/posts/installing-raspotify-arm64/)

zhuykovkb, 2021-06-17:
> It is a bit hackish, but this may be worth a try in the mean time... https://thewayeye.net/posts/installing-raspotify-arm64/
> 
> I am running Ubuntu 20.10 for the raspberry pi which is also arm64.

awesome trick!

JasonLG1979, 2021-11-17:
PR's welcome!!!

JasonLG1979, 2021-11-17:
Duplicate of https://github.com/dtcooper/raspotify/issues/263

closed 2021-11-17 by JasonLG1979