dtcooper/raspotify #341: Can't output sound through bluetooth device

Atla5De5troyer, 2020-12-26:
Hi, I recently got a raspberry pi 400 and I want to stream music through raspotify using my bluetooth headphones.  For some reason, it's not outputting sound through them. This is my only way to listen to music on my raspberry pi as it doesn't have an audio jack and my monitor doesn't have speakers, so it would be a great help if someone could help me fix this. I've followed lots of tutorials and forums online but none are yet to work.

konkit, 2020-12-27:
This helped me, I just got it working myself:
https://github.com/dtcooper/raspotify/issues/78#issuecomment-640061060

Atla5De5troyer, 2020-12-27:
> This helped me, I just got it working myself:
> https://github.com/dtcooper/raspotify/issues/78#issuecomment-640061060

So what steps did you actually take? What did you do to make it work?

konkit, 2020-12-27:
0. Install raspotify with `curl -sL https://dtcooper.github.io/raspotify/install.sh | sh`
1. I created a file /home/pi/.config/systemd/user/raspotify.service and filled it with: 
```
[Unit]                                                                                                                                                                                                                                                                                                                     
Description=Raspotify                                                                                                                                                                                                                                                                                                      
After=network.target                                                                                                                                                                                                                                                                                                       
                                                                                                                                                                                                                                                                                                                            
[Service]                                                                                                                                                                                                                                                                                                                  
Restart=always                                                                                                                                                                                                                                                                                                             
RestartSec=10                                                                                                                                                                                                                                                                                                              
Environment="DEVICE_NAME=raspotify (%H)"                                                                                                                                                                                                                                                                                   
Environment="BITRATE=160"                                                                                                                                                                                                                                                                                                  
Environment="CACHE_ARGS=--disable-audio-cache"                                                                                                                                                                                                                                                                             
Environment="VOLUME_ARGS=--enable-volume-normalisation --linear-volume --initial-volume=100"                                                                                                                                                                                                                               
Environment="BACKEND_ARGS=--backend alsa"                                                                                                                                                                                                                                                                                  
EnvironmentFile=-/etc/default/raspotify                                                                                                                                                                                                                                                                                    
ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS                                                                                                                                                                                                    
                                                                                                                                                                                                                                                                                                                             [Install]                                                                                                                                                                                                                                                                                                                  
WantedBy=default.target
```
2. Call `systemctl --user enable /home/pi/.config/systemd/user/raspotify.service` - that `--user` part is important.

and it worked - at least in my case. After that, I had two devices (one installed by the raspotify itself, one by the above script), so I removed the first one by `sudo systemctl disable raspotify.service`.

Atla5De5troyer, 2020-12-27:
> 0. Install raspotify with `curl -sL https://dtcooper.github.io/raspotify/install.sh | sh`
> 1. I created a file /home/pi/.config/systemd/user/raspotify.service and filled it with:
> ``` [...]

Ok, I'll try that out when I get the chance and let you know if it works

Atla5De5troyer, 2020-12-28:
@konkit I tried what you suggested but I got stuck on creating the file with it saying 'Error writing /home/pi/.config/systemd/user/raspotify.service: no such file or directory'. I assume you meant using sudo nano, which is what I used to try and create the file. Do you know how to fix this issue?

konkit, 2020-12-28:
Try running `mkdir -p /home/pi/.config/systemd/user/` first, the folder needs to exist before you create a file there. Also, you shouldn't have to run nano with sudo, since it's your home directory, just nano should be enough

Atla5De5troyer, 2020-12-28:
> Try running `mkdir -p /home/pi/.config/systemd/user/` first, the folder needs to exist before you create a file there. Also, you shouldn't have to run nano with sudo, since it's your home directory, just nano should be enough

Ok thanks, I'll try that now

Atla5De5troyer, 2020-12-28:
@konkit So I'm guessing you use _sudo systemctl start raspotify_ to start it?

konkit, 2020-12-28:
I went to folder `/home/pi/.config/systemd/user/` and called `systemctl --user start raspotify.service`

Atla5De5troyer, 2020-12-28:
> I went to folder `/home/pi/.config/systemd/user/` and called `systemctl --user start raspotify.service`

OMG Yes!!! Thank you so much!!! It's working now! Also, quick question did you ever fix the widevine issue on raspberry pi's?

konkit, 2020-12-28:
Sorry, I can't help you with that, because I don't even know what that widevine problem is and I control Spotify from my smartphone, from Spotify app.

Atla5De5troyer, 2020-12-28:
> Sorry, I can't help you with that, because I don't even know what that widevine problem is and I control Spotify from my smartphone, from Spotify app.

Ok, that's fine anyway. Thanks for the help!

closed 2020-12-28 by Atla5De5troyer

kennethshsu, 2024-09-05:
I don't think this works in 2024 now? Anyone having any luck? Here are the steps I followed:

1. **Install Raspotify**

2. **Create a service file:**
   - Navigate to the directory and create a Raspotify service file:
     ```bash
     mkdir -p /home/pi/.config/systemd/user/
     nano /home/pi/.config/systemd/user/raspotify.service
     ```
   - Paste the following configuration into the service file:
     ```ini
     [Unit]
     Description=Raspotify
     After=network.target

     [Service]
     Restart=always
     RestartSec=10
     Environment="DEVICE_NAME=raspotify (%H)"
     Environment="BITRATE=160"
     Environment="CACHE_ARGS=--disable-audio-cache"
     Environment="VOLUME_ARGS=--enable-volume-normalisation --linear-volume --initial-volume=100"
     Environment="BACKEND_ARGS=--backend alsa"
     EnvironmentFile=-/etc/default/raspotify
     ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS

     [Install]
     WantedBy=default.target
     ```

3. **Enable the service:**
   - Enable the newly created service:
     ```bash
     systemctl --user enable /home/pi/.config/systemd/user/raspotify.service
     ```

4. **Start the service:**
   - Start Raspotify by running:
     ```bash
     systemctl --user start raspotify.service
     ```
From the above, I don't see Raspotify running; however, if I do `sudo systemctl start raspotify`, Raspotify will show up in Spotify, but of course bluetooth audio does not work.

JWesorick, 2025-05-04:
When I run `systemctl --user status raspotify.service` I see errors with `librespot` args. Trying to fix those now.

JWesorick, 2025-05-04:
Mine ended up working with this:

```
[Unit]
Description=Raspotify (Spotify Connect Client)
Documentation=https://github.com/dtcooper/raspotify
Documentation=https://github.com/librespot-org/librespot
Documentation=https://github.com/dtcooper/raspotify/wiki
Documentation=https://github.com/librespot-org/librespot/wiki/Options
Wants=network.target sound.target
After=network.target sound.target

OnFailure=raspotify-crash-report-generator.service

[Service]
DynamicUser=no
# SupplementaryGroups=audio

# /usr/bin/raspotify-crash-report-generator.sh run by
# raspotify-crash-report-generator.service handles
# restart logic.
 
#Restart=on-failure
#RestartSec=10

# Some of these may be implied by DynamicUser=yes
RestrictRealtime=true
RestrictSUIDSGID=true
RestrictNamespaces=true
RestrictAddressFamilies=AF_UNIX AF_INET AF_INET6 AF_NETLINK

ProtectHostname=true
ProtectControlGroups=true
ProtectKernelLogs=true
ProtectKernelModules=true
ProtectKernelTunables=true
ProtectProc=invisible
# ProtectHome=true
ProtectHome=no
# ProtectSystem=strict
ProtectSystem=full
ProtectClock=yes

DevicePolicy=strict
DeviceAllow=char-alsa rw
DeviceAllow=/dev/null rw
DeviceAllow=/dev/random r
DeviceAllow=/dev/urandom r

UMask=077
CacheDirectoryMode=0700
CacheDirectory=%N
StateDirectoryMode=0700
StateDirectory=%N
ConfigurationDirectoryMode=0600
# ConfigurationDirectory=%N
RuntimeDirectory=%N

# PrivateTmp=true
PrivateTmp=no
PrivateUsers=true

ProcSubset=pid
NoNewPrivileges=true
PermissionsStartOnly=true
LockPersonality=true
MemoryDenyWriteExecute=true
RemoveIPC=true

CapabilityBoundingSet=

SystemCallArchitectures=native
SystemCallFilter=@system-service
SystemCallFilter=~@privileged @resources
SystemCallErrorNumber=EPERM

Environment=LIBRESPOT_NAME="%N (%H)"
Environment=LIBRESPOT_BACKEND="alsa"
# Environment=LIBRESPOT_CACHE=%C/%N
Environment=LIBRESPOT_CACHE=%h/.cache/raspotify
# Environment=LIBRESPOT_SYSTEM_CACHE=%S/%N

# This Moves librespot's /tmp to RAM
# It is overridden in the config.
# See the config for details.
# Environment=TMPDIR=%t/%N
Environment=TMPDIR=%t
EnvironmentFile=-%E/%N/conf

ExecStart=/usr/bin/librespot

[Install]
WantedBy=multi-user.target
```