dtcooper/raspotify #343: new installation (of .deb) fails because the preinst script tries to change /etc/default/raspotify which does not exist at that point

ernstkl, 2020-12-29:
the preinst script tries to change /etc/default/raspotify and but for a new installation (first time installation) that file is not present at that point. For that reason installing the deb package with dpkg fails: 
``` new raspotify package pre-installation script subprocess returned error exit status 2```

I could circumvent the problem by doing
sudo touch /etc/default/raspotify
before installing the package, which supports my analysis of the error

The solution, I guess, is to just do the sed operation in the postinst script instead.

https://github.com/dtcooper/raspotify/blob/a991c2d2f8e94367f8f5c347450712e9d18abfde/raspotify/DEBIAN/preinst#L5

dtcooper, 2020-12-30:
Hi @ernstkl what version are you using? I am going to withdraw the most recent tag's `deb` file now because of this. (I never deployed the `.deb` to the gh-pages reprepro for this reason so people using apt to download the package won't have seem this bug.)

ernstkl, 2020-12-30:
Hi! I  built the package myself two days ago, and the file name looks like this:
`raspotify_0.15.0~librespot.v0.1.3-37-g0adb851_armhf.deb`

dtcooper, 2020-12-30:
Yeah. That's broken. I'm about to push a fix, can you rebuild and test with me? I've been putting off upgrading the package (and librespot is wayyyyy out of date) to avoid breaking people's installations. But now there's major missing features!

dtcooper, 2020-12-30:
And by "about to" give me 20 minutes! :)

dtcooper, 2020-12-30:
(Also what version of raspbian are you using, @ernstkl: `lsb_release -a`)

ernstkl, 2020-12-30:
yes, I would be happy to test something for you

*$* lsb_release -a

dtcooper, 2020-12-30:
Thanks! I mean could you execute `lsb_release -a` in your shell and paste the output. 😛

ernstkl, 2020-12-30:
sorry, the output was in my reply (via mail) but got cut off:

Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

dtcooper, 2020-12-30:
If possible, can you try this build I've just made: [raspotify_0.15.0-3-g3bcb36d-dirty~librespot.v0.1.3-37-g0adb851_armhf.tar.gz](https://github.com/dtcooper/raspotify/files/5755737/raspotify_0.15.0-3-g3bcb36d-dirty.librespot.v0.1.3-37-g0adb851_armhf.tar.gz)?

ernstkl, 2020-12-30:
installation went fine, I answered the question whether to install the new config file with "no"
and playing spotify works as well

ernstkl, 2020-12-30:
I just noticed the new config file (stored for me as raspotify.dpkg-dist) has no newline at end of file. Which is not problem (I tried it) but a bit strange maybe.

ernstkl, 2020-12-30:
first-time installation (after a dpkg --purge raspotify) works fine as well, as expected

dtcooper, 2020-12-30:
Wonderful! I added a new line (and one more substitution based on a corner case if you edited the systemd unit file directly), and made it so I rearranged/edited less of the `/etc/default/respotify` file, so the diff looks better.

Here's the actual release (not just a dirty local copy: https://github.com/dtcooper/raspotify/releases/download/0.16.0/raspotify_0.16.0.librespot.v0.1.3-37-g0adb851_armhf.deb

Might you test that for posterity too? If it works, I'll push it to the repo and all our users can enjoy! Thanks so much for your help, @ernstkl.

ernstkl, 2020-12-30:
sorry for not replying earlier (the latest github notifications went to the spam folder for some reason).

The https://github.com/dtcooper/raspotify/releases/download/0.16.0/raspotify_0.16.0.librespot.v0.1.3-37-g0adb851_armhf.deb
installs just fine, with and without a purge of the old package.

And it plays the music. That is all I can say.

Thanks for providing this nice wrapper around librespot!

dtcooper, 2020-12-30:
Great. Thanks for the help. Closing this out.

closed 2020-12-30 by dtcooper

dtcooper, 2020-12-30:
> And it plays the music. That is all I can say.

Pretty sure that mean's it's working!