dtcooper/raspotify #344: It works on the orange pi zero LTS

febs, 2021-01-03:
This is actually not an issue. I wanted to report that raspotify is working perfectly fine on the Single Board Computer "Orange Pi Zero LTS", using Armbian 20.11.3.

You might want to include that on the README file, eventually. 
Thanks,

dtcooper, 2021-01-04:
Feel free to submit a PR with your additions to the README. Thanks!

closed 2021-01-04 by dtcooper

thanhtantran, 2024-05-27:
Not working man. I am using Armbian_22.05.3_Orangepizero_focal_current_5.15.48

when run the automate installer `sudo apt-get -y install curl && curl -sL https://dtcooper.github.io/raspotify/install.sh | sh`

it stops and return

```
Unmet minimum required package version(s):

coreutils >= 8.32 is required but 8.30-3ubuntu2 is installed.
systemd >= 247.3 is required but 245.4-4ubuntu3.17 is installed.
libasound2 >= 1.2.4 is required but 1.2.2-2.1ubuntu2.5 is installed.
alsa-utils >= 1.2.4 is required but 1.2.2-1ubuntu2.1 is installed.
```