dtcooper/raspotify #346: Click noise when skipping songs

febs, 2021-01-04:
I am experiencing the opposite issue as this: https://github.com/dtcooper/raspotify/issues/231

This is happening on a raspberry Pi zero with a DAC hat board.
raspotify version 0.16.0~librespot.v0.1.3-37-g0adb851 armhf

If I allow the music to flow from a tune to another everything works flawlessly. 
If I skip manually to another tune, then I hear a popping sound similar to a clipping one. 

The logs don't report anything problematic.

jackobtone, 2021-01-08:
In my case, when I skip songs it sounds like spotify is replaying 1-2sec of last song.

oskarnilsson, 2021-03-30:
I have the exact same issue. A loud crackling sound when skipping or changing song. Using a Pi Zero and playing through the USB which is plugged into my dac/amp.

JasonLG1979, 2021-11-20:
The new release is live. Please update and see if that solves your issue.

JasonLG1979, 2021-11-24:
This is an upstream issue. If after updating to the latest version of Raspotify the issue persists please file a bug with librespot.

closed 2021-11-24 by JasonLG1979