dtcooper/raspotify #347: 3.5mm output works only when booted with display

amarchand91, 2021-01-04:
Hi all,

Happy to be a new member of this community! I have recently acquired a Raspberry Pi 3 Model B with the intention of using it solely as a Spotify Connect device via Raspotify.

After installing Raspotify, I had issues getting Raspotify to output to the 3.5mm jack, even whilst other sources such as the browser had no issue doing so. This was solved by changing the defaults.ctl.card and defaults.pcm card values from 0 to 1 in /usr/share/alsa/alsa.conf

3.5mm output now works with Raspotify, but only when Raspberry Pi is booted with a display connected. If booted without a display, there is no audio output. My goal is to have this Raspberry Pi as a headless playback device so this is less than ideal. I read in another thread that pulseaudio has caused a lot of issues with 3.5mm audio so I uninstalled pulseaudio; nevertheless this issue persists.

Any/all feedback is highly appreciated.

Best,
Alden

dirkk0, 2021-01-18:
You may have to change this back to zero via ssh from another computer again.

To me it was the other way around. Once I connected the display, the audio stopped working.
I tried editing the config.txt as suggested [here](https://github.com/dtcooper/raspotify/issues/22#issuecomment-377174304), but that didn't help.

ConorShore, 2021-04-16:
Checkout this pull request I just made (https://github.com/dtcooper/raspotify/pull/386). Had the same problem and made a fix to the docs. Basically for me when you plugged in a display the hw device number changed, so rather than selected a device by number, do it by name

JasonLG1979, 2021-11-17:
```
# Available devices can be listed with the following command:
# librespot --name hello --device ? | grep plughw -v | grep hwdep -v | grep hw
# This lists available devices (note b1=HDMI1, b2=HDMI2 etc.). 
# If you can’t find your device with that command, try running:
# librespot --name hello --device ?
# This gives a more verbose output, which can help you identify the target
# sound card if it has a weird name
# Say you want Spotify to play out of HDMI1, you run that command, and copy the 
# entire line b1 then you add '--device ' with your preferred device to 
# the OPTIONS field below. For example, you want to set Spotify to output out of 
# HDMI1 you run the command, copy the line 'hw:CARD=b1,DEV=0' from the output and 
# add '--device hw:CARD=b1,DEV=0' to options. Another example using a hardware 
# DAC would be '--device hw:CARD=IQaudIODAC,DEV=0'
# This method is useful, as the name wont change depending on which devices are
# connected to the Pi, as was with the old way (i.e. --device hw:0,0)
# This might not work if you're using a USB sound card, and you'll be stuck to
# specifying devices like --device hw:0,0
```

closed 2021-11-17 by JasonLG1979