dtcooper/raspotify #350: stops playing after some songs

wdklotz, 2021-01-09:
I run latest version od raspbian: 
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
.....
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-01-09 16:42:43 CET; 20min ago
  Process: 1740 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 1741 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 1742 (librespot)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/raspotify.service
           └─1742 /usr/bin/librespot -u wdklotz@gmail.com -p xxxxxxx --device hw:1,0 --name raspotify (raspberrypi) --device-type speaker --backend alsa --bitrate 160 -c /var/cache/raspotify --enable-volume-normalisation --volume-ctrl linear --initial-volume 100

**Jan 09 17:03:00 raspberrypi librespot[1742]: [2021-01-09T16:03:00Z ERROR librespot_core::session] Os { code: 104, kind: ConnectionReset, message: "Connection reset by peer" }
Jan 09 17:03:00 raspberrypi librespot[1742]: [2021-01-09T16:03:00Z WARN  librespot] Spirc shut down unexpectedly**
Jan 09 17:03:00 raspberrypi librespot[1742]: [2021-01-09T16:03:00Z WARN  librespot_audio::fetch] Error from channel for data receiver for range 5503572 (+21076).
Jan 09 17:03:00 raspberrypi librespot[1742]: [2021-01-09T16:03:00Z WARN  librespot_audio::fetch] Error from channel for data receiver for range 5487188 (+16384).
Jan 09 17:03:00 raspberrypi librespot[1742]: [2021-01-09T16:03:00Z WARN  librespot_audio::fetch] Error from channel for data receiver for range 5409364 (+32768).
Jan 09 17:03:00 raspberrypi librespot[1742]: [2021-01-09T16:03:00Z WARN  librespot_audio::fetch] Error from channel for data receiver for range 5442132 (+45056).
Jan 09 17:03:00 raspberrypi librespot[1742]: [2021-01-09T16:03:00Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-b-f6mh.ap.spotify.com:4070"
Jan 09 17:03:00 raspberrypi librespot[1742]: [2021-01-09T16:03:00Z INFO  librespot_core::session] Authenticated as "wdklotz" !
Jan 09 17:03:00 raspberrypi librespot[1742]: [2021-01-09T16:03:00Z INFO  librespot_playback::audio_backend::alsa] Using alsa sink
Jan 09 17:03:00 raspberrypi librespot[1742]: [2021-01-09T16:03:00Z INFO  librespot_core::session] Country: "FR"

**### What causes this connection reset? How can it be corrected?**

pbrownpb18, 2021-01-25:
I have found the following:
This happens to me when the track to be played does not have a Title show up in the queue.
It clears the entire Play Queue.
It only happens when playing on Raspotify.
Log messages captured as follows:
Jan 25 09:36:15 raspberrypi librespot[16994]: [2021-01-25T16:36:15Z INFO  librespot_playback::player] Loading <> with Spotify URI <spotify:track:0nbDRhm5RCixXYioMBt0sx>
Jan 25 09:36:15 raspberrypi librespot[16994]: [2021-01-25T16:36:15Z WARN  librespot_playback::player] <spotify:track:0nbDRhm5RCixXYioMBt0sx> is not available
Jan 25 09:36:15 raspberrypi librespot[16994]: [2021-01-25T16:36:15Z WARN  librespot_connect::spirc] The player has stopped unexpectedly.

JasonLG1979, 2021-11-17:
It's a network issue.

closed 2021-11-17 by JasonLG1979