dtcooper/raspotify #355: Songs played too fast + pitch

marcianito, 2021-01-17:
Hey all,

I installed raspotify together with snapserver on my Pi4 (raspbian). It worked mostly out of the box. I only added the appropiate pipe in the raspotify config file:
``BACKEND_ARGS="--backend pipe --device /tmp/snapfifo" ``
My Spotify Connect on cellphone recognized the device and music was played and outputtet. So far, so good.
The only (annoying) issue is that all songs are outputted a little to fast and with a minimal pitch.
I could not find anything while searching the Web and also no equivalent possible entries in config files to "fix" that.
Does anyone else has experienced this before?
Hope you can help! This is an awesome project and super cool for home audio systems and I would love to use it!
Thanks!!

thiesschneider, 2021-01-19:
I have the same problem but unfortunately no solution until now

thiesschneider, 2021-01-22:
Hey @marcianito ,

You have to add to your snapserver.conf the stream:

```
[stream]
stream = pipe:///tmp/snapcast/snapfifo?name=Snapserver&sampleformat=44100:16:2
```

If you change the sampleformat it runs in the normal speed

Hope this helps!

closed 2021-11-17 by JasonLG1979