dtcooper/raspotify #365: I just have a question: does it use/need a gui? or can it run on a cli setup?

Simao-Santos, 2021-02-11:


Elalith, 2021-02-11:
It doesn't even have a GUI ;) I'm running Raspotify on my headless Raspberry Pi Zero W, which is connected to my 80's audioreceiver. I'm "casting" songs through the Spotify desktop app and the mobile app.

Simao-Santos, 2021-02-11:
@Elalith are you using a hifiberry to connect to the 80s receiver?

Elalith, 2021-02-11:
Nope, I'm using a [Pimoroni PHAT DAC](https://shop.pimoroni.com/products/phat-dac), but it should work with any DAC.

Simao-Santos, 2021-02-11:
Nice I was looking for something like that!
Did you have to solder anything? Do you recommend it?

Elalith, 2021-02-11:
I had to solder the female header on the DAC, but that's an easy thing to do.

Simao-Santos, 2021-02-11:
The only problem is I don't have a solder 😓
Can you recommend a very simple cheap USB solder or something?
Thanks for the help eheh

Elalith, 2021-02-12:
any cheap 12/25watt soldering iron will do ;) and it helps if you get tin with a flux core. Personally I use Griffon 99/1 (tin/copper).

But finding a HiFiBerry DAC+ Zero is propably easier ;) They have an presoldered header.

Elalith, 2021-02-12:
If you lived anywhere near where I live, I would solder it for you ;)

Elalith, 2021-02-12:
This is a very usefull tutorial: https://youtu.be/8Z-2wPWGnqE

closed 2021-04-09 by Simao-Santos