dtcooper/raspotify #366: Combined Cache of several instances?

JustMyGithub, 2021-02-12:
I use several Instances of raspotify for independent audio sinks on the same system. Is it possible to use a combined cache avoiding the same song being cached in each cache?

5AMsan, 2021-02-12:
According to this line https://github.com/dtcooper/raspotify/blob/c14c4c5f1d69d8abcb6e0d62e2e39d221ef64777/raspotify/etc/default/raspotify#L31, you should be able to use a network location.

JustMyGithub, 2021-02-13:
Yes, I can configure the same cache path, but the question is whether this works out. Simple example: If the filename of the cached files was hostname--songid this would not work even if all instances have the same path in the config file, because the different instances would have different hostnames and thereby not find the cached file and re-download.

JasonLG1979, 2021-11-17:
That's a pretty unique use case. I contribute upstream to librespot but I haven't really dug into that part of the code. I'd jump on their gitter and ask.

https://gitter.im/librespot-org/spotify-connect-resources

closed 2021-11-17 by JasonLG1979

JustMyGithub, 2021-11-19:
Thanks for pointing me there @JasonLG1979 - but can you elaborate on the structure of the librespot community? I don't see a bug/issue/fr-Tracker there, only a chat. If I post to that chat, the request will be forgotten pretty fast I guess.

JasonLG1979, 2021-11-19:
Yours is a question. It's not a bug or a feature request. The chat is the best place to ask. If you'll get an answer or not I can't say. The most reliable route would be to read the code or test it out yourself and see if multiple files are generated when multiple users play the same song.