dtcooper/raspotify #368: selective issue of no-sound

jjuergensen, 2021-02-20:
I'm running raspotify on a pi 4, and have an odd problem with not getting sound.  As a running process, it looks like this:

`raspoti+   826  0.0  0.0  12756  3876 ?        Ssl  20:47   0:00 /usr/bin/librespot --name raspotify (pimusic) --device-type speaker --device hw:2,0 --backend alsa --bitrate 160 --disable-audio-cache --enable-volume-normalisation --volume-ctrl linear --initial-volume 100`

I'm trying to run it through a USB-connected Musical Fidelity V90 DAC, which is listed via aplay as card 2, with only one subdevice, hence hw:2,0.   No dice, no sound.  If I switch to headphones (hw:1,0) and restart the service, I can stream to it perfectly fine.

When I run `speaker-test -c2 -twav -D plughw:2,0` that also works fine through the DAC.

I'm also running shairport sync, and it will happily stream anything through the DAC.  I've even tried disabling the shairport service and rebooting, in case there was some contention, but that didn't help.

I'm at a complete loss for what to try; any help would be appreciated.  (yes, I can stream spotify via airplay, but it would be preferable to do it within the spotify app)

kavsingh, 2021-08-31:
@jjuergensen i also ran into this, fixed it similar to https://github.com/dtcooper/raspotify/issues/363#issuecomment-774573316

basically, went with troubleshooting advice: https://github.com/dtcooper/raspotify#troubleshooting

- set `defaults.ctl.card`, `defaults.pcm.card` and `defaults.pcm.device` in `/usr/share/alsa/alsa.conf` to match USB device. in your case, as per aplay
  ```
  defaults.ctl.card 2
  defaults.pcm.card 2
  defaults.pcm.device 0
  ```
- remove `--device hw:2,0` from service startup 

this worked for me, hope it does for you

JasonLG1979, 2021-11-19:
Troubleshooting has been moved to:

https://github.com/dtcooper/raspotify/wiki/Troubleshooting

It's also not advisable to edit `/usr/share/alsa/alsa.conf` as it can be overwritten by system updates.

As explained in the wiki, the proper file to edit is `/etc/asound.conf` (it does not exist by default)  `/etc/asound.conf` is specifically for overriding `/usr/share/alsa/alsa.conf`.

closed 2021-11-24 by JasonLG1979