dtcooper/raspotify #369: raspotify instance becomes undiscoverable after a few hours of not being used

viktak, 2021-02-25:
Hi,
I have the latest version running mostly OK on an old Raspberry Pi Model B Rev 2. It works perfectly during the day, but overnight, when not used for several hours, something happens to it, because in the morning it is not discoverable any more (by my Windows PCs).

`sudo systemctl status raspotify` shows that the system is running, but I need to issue `sudo systemctl restart raspotify` to make it discoverable again.

The system, otherwise, seems to be doing fine, I can `ssh` in any time with no issues.

twisst, 2021-03-03:
Same here. Restart did not help for me.

twisst, 2021-03-05:
Turns out I had to connect to the network with an Ethernet cable. With the cable, I could connect to Raspotify again, and after that also over wifi again. I don't really understand why it works, but it does :-)

Chreece, 2021-03-23:
I have the same issue and did a crontab service restart every morning to resolve this

Simao-Santos, 2021-04-09:
> I have the same issue and did a crontab service restart every morning to resolve this

You mean restart the raspberry pi itself?

JasonLG1979, 2021-11-17:
This a network issue not a Raspotify/librespot issue. Many routers do not play well with mDNS. I actually replaced a router for this reason.

closed 2021-11-17 by JasonLG1979

viktak, 2021-11-18:
> Many routers do not play well with mDNS

Would you mind elaborating a little on this?

JasonLG1979, 2021-11-18:
https://superuser.com/questions/730288/why-do-some-wifi-routers-block-multicast-packets-going-from-wired-to-wireless