dtcooper/raspotify #373: Hangs when connection lost

giggywithit, 2021-03-10:
It seems that if the internet part of the connection is broken while playing back, that the device no longer appears as a connect device and is stuck hanging. The audio device is also stuck in an in-use state while there is no audio actually being played back.

`raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-01-31 09:57:53 EST; 1 months 7 days ago
  Process: 632 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 633 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 634 (librespot)
    Tasks: 3 (limit: 3839)
   CGroup: /system.slice/raspotify.service
           └─634 /usr/bin/librespot --name CASPER --device-type speaker --backend alsa --bitrate 320 --disable-audio-cache --volume-ctrl=linear --initial-volume=65 --mixer=alsa --mixer-linear-volume --

Mar 09 14:42:12 GHOSTM3U librespot[634]: [2021-03-09T19:42:12Z INFO  librespot_playback::player] Loading <Thrill Minute> with Spotify URI <spotify:track:1FUPzWVtZDyrHEDHZH6OaZ>
Mar 09 14:42:12 GHOSTM3U librespot[634]: [2021-03-09T19:42:12Z INFO  librespot_playback::player] <Thrill Minute> (271586 ms) loaded
Mar 09 14:46:43 GHOSTM3U librespot[634]: [2021-03-09T19:46:43Z INFO  librespot_playback::player] Loading <Another Exit> with Spotify URI <spotify:track:2P7QCrj7UvspZA78ySXB2f>
Mar 09 14:46:43 GHOSTM3U librespot[634]: [2021-03-09T19:46:43Z INFO  librespot_playback::player] <Another Exit> (332933 ms) loaded
Mar 09 14:52:16 GHOSTM3U librespot[634]: [2021-03-09T19:52:16Z INFO  librespot_playback::player] Loading <Quintana Roo> with Spotify URI <spotify:track:5ecg3Tj9W0fXyEoAbGHfgZ>
Mar 09 14:52:16 GHOSTM3U librespot[634]: [2021-03-09T19:52:16Z INFO  librespot_playback::player] <Quintana Roo> (249066 ms) loaded
Mar 09 14:56:30 GHOSTM3U librespot[634]: [2021-03-09T19:56:30Z INFO  librespot_playback::player] Loading <Luna Nueva (Only a Dream)> with Spotify URI <spotify:track:3umZAedKUa42pU1c8ee85N>
Mar 09 14:58:51 GHOSTM3U librespot[634]: [2021-03-09T19:58:51Z INFO  librespot_playback::player] <Luna Nueva (Only a Dream)> (270706 ms) loaded
Mar 09 15:15:39 GHOSTM3U librespot[634]: [2021-03-09T20:15:39Z ERROR librespot_core::session] Os { code: 110, kind: TimedOut, message: "Connection timed out" }
Mar 09 15:15:39 GHOSTM3U librespot[634]: [2021-03-09T20:15:39Z WARN  librespot] Spirc shut down unexpectedly
`

JasonLG1979, 2021-11-20:
The new release is live. Please update and see if that solves your issue.

JasonLG1979, 2021-11-24:
This is an upstream issue. If after updating to the latest version of Raspotify the issue persists please file a bug with librespot.

closed 2021-11-24 by JasonLG1979