dtcooper/raspotify #380: Drop raspotify

HITMIK, 2021-03-27:
Problem when I play spotify on raspberry pi 4 with newest fw, kernel etc.

pi@raspberrypi:~ $ sudo systemctl status raspotify
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-03-27 15:45:56 CET; 58s ago
  Process: 16088 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 16089 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 16090 (librespot)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/raspotify.service
           └─16090 /usr/bin/librespot --name kodi --device-type avr --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --volume-ctrl linea

Mar 27 15:46:44 raspberrypi librespot[16090]:   17:   0x5b3304 - <librespot_playback::player::PlayerInternal as futures::future::Future>::poll::he6e04a5bccf94e34
Mar 27 15:46:44 raspberrypi librespot[16090]:   18:   0x517300 - futures::task_impl::std::set::hab1c1811473e3998
Mar 27 15:46:44 raspberrypi librespot[16090]:   19:   0x50d838 - std::sys_common::backtrace::__rust_begin_short_backtrace::h8286c5a0fd04a387
Mar 27 15:46:44 raspberrypi librespot[16090]:   20:   0x5284f4 - core::ops::function::FnOnce::call_once{{vtable.shim}}::h76e6c7604d7daced
Mar 27 15:46:44 raspberrypi librespot[16090]:   21:   0x954940 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::h6b7ad05f87ff90ae
Mar 27 15:46:44 raspberrypi librespot[16090]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/alloc/src/boxed.rs:1042
Mar 27 15:46:44 raspberrypi librespot[16090]:   22:   0x954940 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::he9130e675b8d97ee
Mar 27 15:46:44 raspberrypi librespot[16090]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/alloc/src/boxed.rs:1042
Mar 27 15:46:44 raspberrypi librespot[16090]:   23:   0x954940 - std::sys::unix::thread::Thread::new::thread_start::h7a3f6590d8e03e95
Mar 27 15:46:44 raspberrypi librespot[16090]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/sys/unix/thread.rs:87

HITMIK, 2021-04-01:
pi@raspberrypi:~ $ sudo systemctl status raspotify
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-04-01 15:14:28 CEST; 6min ago
  Process: 632 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 642 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 652 (librespot)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/raspotify.service
           └─652 /usr/bin/librespot --name kodi --device-type avr --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --volume-ctrl linear

Apr 01 15:20:58 raspberrypi librespot[652]:   17:   0x543304 - <librespot_playback::player::PlayerInternal as futures::future::Future>::poll::he6e04a5bccf94e34
Apr 01 15:20:58 raspberrypi librespot[652]:   18:   0x4a7300 - futures::task_impl::std::set::hab1c1811473e3998
Apr 01 15:20:58 raspberrypi librespot[652]:   19:   0x49d838 - std::sys_common::backtrace::__rust_begin_short_backtrace::h8286c5a0fd04a387
Apr 01 15:20:58 raspberrypi librespot[652]:   20:   0x4b84f4 - core::ops::function::FnOnce::call_once{{vtable.shim}}::h76e6c7604d7daced
Apr 01 15:20:58 raspberrypi librespot[652]:   21:   0x8e4940 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::h6b7ad05f87ff90ae
Apr 01 15:20:58 raspberrypi librespot[652]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/alloc/src/boxed.rs:1042
Apr 01 15:20:58 raspberrypi librespot[652]:   22:   0x8e4940 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::he9130e675b8d97ee
Apr 01 15:20:58 raspberrypi librespot[652]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/alloc/src/boxed.rs:1042
Apr 01 15:20:58 raspberrypi librespot[652]:   23:   0x8e4940 - std::sys::unix::thread::Thread::new::thread_start::h7a3f6590d8e03e95
Apr 01 15:20:58 raspberrypi librespot[652]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/sys/unix/thread.rs:87

pi@raspberrypi:~ $ sudo systemctl status raspotify
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-04-01 15:14:28 CEST; 6min ago
  Process: 632 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 642 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 652 (librespot)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/raspotify.service
           └─652 /usr/bin/librespot --name kodi --device-type avr --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --volume-ctrl linear

Apr 01 15:20:58 raspberrypi librespot[652]:   22:   0x8e4940 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::he9130e675b8d97ee
Apr 01 15:20:58 raspberrypi librespot[652]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/alloc/src/boxed.rs:1042
Apr 01 15:20:58 raspberrypi librespot[652]:   23:   0x8e4940 - std::sys::unix::thread::Thread::new::thread_start::h7a3f6590d8e03e95
Apr 01 15:20:58 raspberrypi librespot[652]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/sys/unix/thread.rs:87
Apr 01 15:21:20 raspberrypi librespot[652]: [2021-04-01T13:21:20Z WARN  libmdns::fsm] couldn't parse packet from 192.168.100.228:5353: type 47 is invalid
Apr 01 15:21:20 raspberrypi librespot[652]: [2021-04-01T13:21:20Z WARN  libmdns::fsm] couldn't parse packet from 172.30.32.1:5353: type 47 is invalid
Apr 01 15:21:20 raspberrypi librespot[652]: [2021-04-01T13:21:20Z WARN  libmdns::fsm] couldn't parse packet from 192.168.100.228:5353: type 47 is invalid
Apr 01 15:21:20 raspberrypi librespot[652]: [2021-04-01T13:21:20Z WARN  libmdns::fsm] couldn't parse packet from 172.30.32.1:5353: type 47 is invalid
Apr 01 15:21:20 raspberrypi librespot[652]: [2021-04-01T13:21:20Z WARN  libmdns::fsm] couldn't parse packet from 192.168.100.228:5353: type 47 is invalid
Apr 01 15:21:20 raspberrypi librespot[652]: [2021-04-01T13:21:20Z WARN  libmdns::fsm] couldn't parse packet from 172.30.32.1:5353: type 47 is invalid

tobischenk, 2021-04-02:
> Problem when I play spotify on raspberry pi 4 with newest fw, kernel etc.
> 
> pi@raspberrypi:~ $ sudo systemctl status raspotify
> ● raspotify.service - Raspotify
> Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
> Active: active (running) since Sat 2021-03-27 15:45:56 CET; 58s ago
> Process: 16088 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
> Process: 16089 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
> Main PID: 16090 (librespot)
> Tasks: 2 (limit: 4915)
> CGroup: /system.slice/raspotify.service
> └─16090 /usr/bin/librespot --name kodi --device-type avr --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --volume-ctrl linea
> 
> Mar 27 15:46:44 raspberrypi librespot[16090]: 17: 0x5b3304 - <librespot_playback::player::PlayerInternal as futures::future::Future>::poll::he6e04a5bccf94e34
> Mar 27 15:46:44 raspberrypi librespot[16090]: 18: 0x517300 - futures::task_impl::std::set::hab1c1811473e3998
> Mar 27 15:46:44 raspberrypi librespot[16090]: 19: 0x50d838 - std::sys_common::backtrace::__rust_begin_short_backtrace::h8286c5a0fd04a387
> Mar 27 15:46:44 raspberrypi librespot[16090]: 20: 0x5284f4 - core::ops::function::FnOnce::call_once{{vtable.shim}}::h76e6c7604d7daced
> Mar 27 15:46:44 raspberrypi librespot[16090]: 21: 0x954940 - <alloc::boxed::Box as core::ops::function::FnOnce>::call_once::h6b7ad05f87ff90ae
> Mar 27 15:46:44 raspberrypi librespot[16090]: at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/alloc/src/boxed.rs:1042
> Mar 27 15:46:44 raspberrypi librespot[16090]: 22: 0x954940 - <alloc::boxed::Box as core::ops::function::FnOnce>::call_once::he9130e675b8d97ee
> Mar 27 15:46:44 raspberrypi librespot[16090]: at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/alloc/src/boxed.rs:1042
> Mar 27 15:46:44 raspberrypi librespot[16090]: 23: 0x954940 - std::sys::unix:🧵:Thread:🆕:thread_start::h7a3f6590d8e03e95
> Mar 27 15:46:44 raspberrypi librespot[16090]: at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/sys/unix/thread.rs:87

I get the same errors on Pi4 with new kernel fw etc, when I use a seperate Audio Device. If i set the Flash --device hw:0,0 it works just fine.

tobischenk, 2021-04-02:
Moreover I wan able to fix the Errors, when using --device plughw:1,0 instead of hw:1,0 in /etc/default/raspotify. Maybe you should set your audio device using plughw.

HITMIK, 2021-04-02:
I try:  --device plughw:1,0 and --device hw:1,0 but nothing work.

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 1: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry DAC+ HiFi pcm512x-hifi-0 [HiFiBerry DAC+ HiFi pcm512x-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

With old fw was OK.

tobischenk, 2021-04-02:
have you tried to use the bcm2835 audio output of the pi onboard Audio? Just to check if that would work.

HITMIK, 2021-04-02:
Yes, with bcm2835 is OK. However, I need hifiberry dac+ output.

HITMIK, 2021-04-02:
I SOLVED.

I use on rpi4 Home Assistant in docker. HA maybe use hifiberry as audio output. I install addon Alsa & PulseAudio in HA a fix this.

Lipown, 2021-07-17:
Same issue as you except the Alsa and PulseAudio addon not working on this issue unfortunatelly.

closed 2021-11-17 by JasonLG1979