dtcooper/raspotify #382: raspotify crashes when another app has used audio

RikWesselink, 2021-04-07:
I'm working on a music player that plays both Spotify and radio streams. **When I switch from a radio stream to Spotify, raspotify becomes unresponsive.** The same thing happens if I open a browser and play a YouTube video. If I wait for about 7 seconds after I stopped the radio stream (or YouTube video), there is no problem.

I'm using MPlayer for playing the radio streams. For MPlayer, it's not an issue if YouTube is also playing, I'll just hear both sounds mixed. 

For audio out, I'm using a USB soundcard. I've added `--device hw:2.0 `to the raspotify config. This USB soundcard is set as default output on my Raspberry Pi.

When I check the status of raspotify at the time of the problem this is what I get:
```
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset
   Active: active (running) since Wed 2021-04-07 16:54:20 CEST; 3min 23s ago
  Process: 11038 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=e
  Process: 11039 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotif
 Main PID: 11040 (librespot)
    Tasks: 2 (limit: 3860)
   CGroup: /system.slice/raspotify.service
           └─11040 /usr/bin/librespot --name raspotify (raspberrypi) --device-ty

Apr 07 16:57:39 raspberrypi librespot[11040]:   17:   0x56b304 - <librespot_play
Apr 07 16:57:39 raspberrypi librespot[11040]:   18:   0x4cf300 - futures::task_i
Apr 07 16:57:39 raspberrypi librespot[11040]:   19:   0x4c5838 - std::sys_common
Apr 07 16:57:39 raspberrypi librespot[11040]:   20:   0x4e04f4 - core::ops::func
Apr 07 16:57:39 raspberrypi librespot[11040]:   21:   0x90c940 - <alloc::boxed::
Apr 07 16:57:39 raspberrypi librespot[11040]:                        at /rustc/7
Apr 07 16:57:39 raspberrypi librespot[11040]:   22:   0x90c940 - <alloc::boxed::
Apr 07 16:57:39 raspberrypi librespot[11040]:                        at /rustc/7
Apr 07 16:57:39 raspberrypi librespot[11040]:   23:   0x90c940 - std::sys::unix:
Apr 07 16:57:39 raspberrypi librespot[11040]:                        at /rustc/7
```

RikWesselink, 2021-04-09:
Okay, I found something, but I'm not sure on how to apply this ([link](https://superuser.com/questions/1157370/how-to-configure-pulseaudio-to-input-output-via-alsa)). It seems that _"...PulseAudio always uses ALSA as backend, and on startup opens all ALSA devices, and since most ALSA devices can't be opened multiple times, this will cause all ALSA applications that try to use an ALSA device directly when PulseAudio is running to fail."_

That seems to be the reason why Raspotify wont work together with other applications that use the default PulseAudio.

RikWesselink, 2021-04-09:
I thought that maybe a solution would be to configure Raspotify as a user service, as discussed in the Read Me (for Pi Desktop). It should now use the default device from PulseAudio, but it doesn't.

JasonLG1979, 2021-05-23:
With pulseaudio installed you are correct, since pulseaudio is a per user service you would need to configure librespot to also be a user level service. The issue you're having is because you've told librespot to bypass pulseaudio and try to open the device directly. As pulseaudio already has it open that will fail. Remove `--device hw:2.0` from your config. On systems with pulseaudio running the "default" ALSA device should already be pulseaudio. Librespot should play out of whatever device you have configured as an output in your sound settings.

closed 2021-11-17 by JasonLG1979