dtcooper/raspotify #384: Device not showing up on local wifi

assfartman, 2021-04-11:
So I installed this earlier and it has been working perfectly after I added my account details into the config file however only I seem to be able to see the device and I was wondering if I've messed up here or it's just part of the software. 
Thanks in advance

Elalith, 2021-04-12:
Just checking, but... where are you missing your device?

The reason for asking is: My device never initially shows up on my Spotify Windows client, but does appear on my android and iOS/iPadOs clients.

If you're missing the device in your network, then there is a whole other problem. Have you tried ping'ing it or used an app like Fing to look for your device? It is possible it changed it's IP-adress after a reboot.

BTW... my device is configured like this:
(Yes, mine is called Framboosje, which is Little Raspberry in dutch)

As you can see, I didn't need to enter my credentials and yet.. my device is casteable within and outside my network. I just needed to cast locally before I could reach it outside.

```
# /etc/default/raspotify -- Arguments/configuration for librespot

# Device name on Spotify Connect
DEVICE_NAME="Framboosje"

# The displayed device type in Spotify clients.
# Can be "unknown", "computer", "tablet", "smartphone", "speaker", "tv",
# "avr" (Audio/Video Receiver), "stb" (Set-Top Box), and "audiodongle".
DEVICE_TYPE="avr"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
BITRATE="320"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info. Make sure whatever arguments you specify
# aren't already covered by other variables in this file. (See the daemon's
# config at `/lib/systemd/system/raspotify.service` for more technical details.)
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
#OPTIONS="--username <USERNAME> --password <PASSWORD>"

# Uncomment to use a cache for downloaded audio files. Cache is disabled by
# default. It's best to leave this as-is if you want to use it, since
# permissions are properly set on the directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

# By default, the volume normalization is enabled, add alternative volume
# arguments here if you'd like, but these should be fine.
VOLUME_ARGS="--enable-volume-normalisation --volume-ctrl linear --initial-volum$

# Backend could be set to pipe here, but it's for very advanced use cases of
# librespot, so you shouldn't need to change this under normal circumstances.
#BACKEND_ARGS="--backend alsa"
```

JasonLG1979, 2021-11-20:
The new release is live. Please update and see if that solves your issue.

JasonLG1979, 2021-12-03:
Closing due to inactivity.

closed 2021-12-03 by JasonLG1979