dtcooper/raspotify #385: Unavailable to control / can't play this right now

SirCameron, 2021-04-14:
I'm having issues with the device being avaialable to connect to.
Changing the name or type of device and then restarting raspotify will update in Spotify on my computer and phone, but both remain greyed out with the above message.

Anyone else had this issue and know how to solve it?

github42069420, 2021-04-15:
currently have this issue.

KSuondProject, 2021-04-15:
Same here.
From fresh install no way to use raspotify. 
No error displayed on the systemctl status but when connected the track doesn't starts and after 2 secs spotify chose another device to stream.

ConorShore, 2021-04-16:
I had this problem when the device you're trying to connect to doesn't exist. This happened for me when i unplugged the display from my Pi, moving my DAC from hw:3 to hw:2. Raspotify would appear, I'd try to play a song and it'd look like it was working for a few seconds then just disconnect.  Try following the instructions in my pull request, and let me know if it works, or you have any comments! (https://github.com/dtcooper/raspotify/pull/386). The instructions are in the config file

dmtroyer, 2021-04-27:
I had this issue on multiple devices even after restarts. It came back after starting and stopping a couple of songs in Spotify.

luthor112, 2021-05-30:
I have the same problem.

luthor112, 2021-05-30:
I might have found the cause (in someone else's comment on another website): This only works for Spotify Premium users.
I bought Premium and now it works.

iorippi, 2021-08-31:
@luthor112 I'm Premium account user so it can't be the reason.

glensc, 2021-11-17:
It worked yesterday for me with same docker image, the only thing that may have changed meanwhile is that I played around with `amixer cset` settings.

I can hear sound fine when I play the sound in docker container with aplay, but starting the librespot ends up with new device appearing, but grayed out with status "Unable to control".

```
aplay -D hw:CARD=Headphones,DEV=0 /usr/share/sounds/alsa/Front_Center.wav
librespot --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --name hello6 --device hw:CARD=Headphones,DEV=0
```

JasonLG1979, 2021-11-17:
> I can hear sound fine when I play the sound in docker container with aplay, but starting the librespot ends up with new device appearing, but grayed out with status "Unable to control".

@glensc That could be a few different things. Most of which have nothing to do with Raspotify/librespot.

The 1st and easiest possibility is that maybe you don't have a Spotify Premium account? A Spotify Premium account is required.

The 2nd is that it's a network configuration issue. Running Raspotify in a docker container is not really a supported use case, that is to say I will not offer help in regards to how to configure your docker container's network. Generally speaking I don't offer much networking help at all even on native Raspotify installs. This isn't a network help forum, sorry.

Raspotify is a packaged version of librespot. Aside from basic alsa configuration if it's not a packaging issue it's not a Raspotify problem.

I can't tell you what's going on though  in your case because you have not provided any logs so I can only guess at this time.

glensc, 2021-11-17:
For bizarre reason, can't pick speaker if I have podcast playing or nothing playing. putting a song to play the devices magically appeared!

> **SOLVED**:
> 
> For some reason, Spotify Connect does not work when playing podcasts! When switching to an actual song, the devices were available again! :laughing:
> 

- https://community.volumio.org/t/solved-spotify-connect-cant-play-this-right-now/6663/2

JasonLG1979, 2021-11-17:
> For bizarre reason, can't pick speaker if I have podcast playing or nothing playing. putting a song to play the device magically appeared!

That is a known librespot some (possibly most or all?) podcasts are in mp3 format. Librespot only supports vorbis at this time.

glensc, 2021-11-17:
duplicate: https://github.com/dtcooper/raspotify/issues/253

glensc, 2021-11-17:
This switch to a song workaround needs to be put to wiki or readme.

JasonLG1979, 2021-11-17:
> This switch to a song workaround needs to be put to wiki or readme.

What do you mean. Do you mean that you can play a podcast after you initially play a song?

glensc, 2021-11-17:
audio codec can't be an issue, as I can:
1. put a song to play
2. connect to librespot device
3. switch to podcast

but this won't work:
1. put podcast to play
2. connect to librespot device

glensc, 2021-11-17:
media I tested with:
- https://open.spotify.com/episode/25hQtnUyDNkxZWjPYCnSk8?si=feU8EvkdSnmK_M7DrQsbAw
- https://open.spotify.com/track/62QohZz1aZNS0ZyAv1v6E4?si=4a324293850741c1

glensc, 2021-11-17:
as I see it, it affects only the first song played once librespot starts up. I can play, pause whatever I want once I've connected once.

but my experience is limited to under 24 hours of use of this software ;)

closed 2021-11-17 by JasonLG1979

glensc, 2021-11-19:
@JasonLG1979 
1. i am using premium
2. there's no logs to provide. as nothing is happening if it's not unavailable to control state
3. the software is working perfectly in the docker container. no support needed

in any case, putting a song to play workaround just works.

after restart (or start) of the program, spotify is in "unavailable to control" state if:
1. i am playing a podcast
2. nothing is playing

if it's common problem, and has a very simple fix, it should end up in a wiki troubleshoot section.

JasonLG1979, 2021-11-19:
I have released a new version.  It will be pushed to the deb repo soon.

https://github.com/dtcooper/raspotify/releases/tag/0.31.1