dtcooper/raspotify #387: Won't connect to RPi0W

fruitofgeorge, 2021-04-18:
I'm using a raspberry pi 0 W running the latest buster with an allo miniboss DAC. Sound outputs fine with speaker-test. The connect service shows up in other players but when I click on it, it never connects and just continues playing on the current device.

ConorShore, 2021-04-28:
Can you try running `aplay -l` and `librespot --name hello --device ?` and put the results here

fruitofgeorge, 2021-04-29:
I already specified the proper card in the config file so that wasn't the issue.
Looks like there were two problems working in conjunction. The instructions I had for turning off power management on the wifi don't actually stick after a reboot and apparently librespot isn't actually available for several minutes after starting the service. It looks to be working now that power management is really off and waiting 10 minutes before trying to connect.

fruitofgeorge, 2021-05-01:
I take it back. This morning it played one song and then just stopped and wouldn't reconnect. Wifi power management says it's off. No clue what the problem is. The librespot service status shows these warnings:
```
librespot[454]: [2021-04-30T10:16:10Z WARN  librespot_audio::fetch] Error from channel for data receiver for range 680700 (+16384).
librespot[454]: [2021-04-30T10:16:10Z WARN  librespot_audio::fetch] Error from channel for data receiver for range 643836 (+36864).
```

JasonLG1979, 2021-11-17:
This is a networking issue unrelated to Raspotify. Your network is unstable and/or your Pi doesn't have a good wifi signal.

closed 2021-11-17 by JasonLG1979