dtcooper/raspotify #389: Script to determine whether audio plays - systemwide to switch on/off speakers

microfx, 2021-04-22:
Hey!

Not understanding where a good location is to write files to with a script that switches on my speakers over webhooks in Homeassistant (smarthome). My idea is to "touch /etc/default/speaker_is_on" and check whether the file exists and then use that information to let the script decide what to do. Problem is: the system service is in an environment – so where is a good place to write such a file so also other services / scripts can check if that file exists (spoiler: /etc/default/ is not a good place).

I want to determine whether the speakers are already turned on or off / being used then not switch them off, if already on do nothing (if that makes sense) – want to use this information also for Airplay (shairport) – they shutoff speakers on their own  even if the speakers are used and I want to handle that. 

I am using this kind of script: 

'''
#!/bin/bash
if [ "$PLAYER_EVENT" = "start" ]; then
  curl -d "" http://raspi.local:8123/api/webhook/p1_speaker_on 
elif [ "$PLAYER_EVENT" = "stop" ]; then
  curl -d "" http://raspi.local:8123/api/webhook/p1_speaker_off
else
  sudo touch /home/pi/touchfile
fi
'''

Would be happy if someone could clarify a bit how this can work out!

giggywithit, 2021-06-01:
I know the following link isn't integrated with this particular project, but it may be of some use..

https://github.com/01BTC10/snd-aloop-gpio-control

JasonLG1979, 2021-11-17:
duplicate of https://github.com/dtcooper/raspotify/issues/93

closed 2021-11-17 by JasonLG1979