dtcooper/raspotify #390: Main process exits - error 101

mhfarahmandm, 2021-04-26:
I installed the package on Ubuntu 20.10 desktop 64 bit
but it returns 101 error and exits
Any rules I should follow?

dtcooper, 2021-04-27:
Expected behaviour. The package is specifically compiled for raspberry pi. Try installing librespot.

closed 2021-04-27 by dtcooper

mhfarahmandm, 2021-04-28:
> Expected behaviour. The package is specifically compiled for raspberry pi. Try installing librespot.

Actually it's ubuntu on raspberry pi [(RPI version of ubuntu desktop)](https://ubuntu.com/tutorials/how-to-install-ubuntu-desktop-on-raspberry-pi-4)

----
> but it returns 101 error and exits

also when main process exits.  it restart in a few mins but it stops in connecting step: `[2021-04-28T04:27:16Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-e-1tk7.ap.spotify.com:4070"`

while it is in connecting state it look like this screenshot on spotify mobile
[image]

dtcooper, 2021-04-28:
Okay, I thought when you said 64 bit you meant amd64. Anyway, do you have / are you using a spotify premium account?